snabbco/snabb #864: update the wiki

lperkov, 2016-04-08:
Can somebody update the wiki so we know which servers are working and which ones are not. I'm talking about this wiki page:

https://github.com/snabbco/snabb/wiki/Snabb-Lab

Thanks!

eugeneia, 2016-04-12:
@lukego @domenkozar Might be a good time to update/remove obsolete lab information that is scattered around.

lukego, 2016-04-13:
Thanks for the poke @lperkov. I have updated the wiki page. Is it telling you what you need now?

@domenkozar I would like to delete this wiki page but I think we need two things: some kind of dashboard showing people what machines are available and what PCI devices they have and documentation in the snabb repo for using the lab (getting an account, using `lock`, etc). Something for you?

domenkozar, 2016-04-14:
@lukego I'll write a draft and post it here.

lperkov, 2016-04-14:
It would be nice if there were instructions how to run docker images... We have some features we need to develop on top of snabb swtich and were planing on using docker which we could run on these machines.

What do you think if we did some recording using asciinema and posted it for everybody to see? Something like this:

https://asciinema.org/a/05cdmz78fhcl5jeo4xyiqqr33

plajjan, 2016-04-15:
@lperkov do you mean how to package Snabb into a docker container? Running one should be rather similar to running any container. 

BTW, is this for packet forwarding or mgmt side integration work? If the latter, why bother with Snabb lab? Just run it on your local laptop :)

lukego, 2016-04-15:
My experience has been that Snabb "just works" inside Docker provided that you use `docker run --privileged` to get root access.

eugeneia, 2016-04-15:
@lperkov There is [src/scripts/dock.sh](https://github.com/snabbco/snabb/blob/master/src/scripts/dock.sh) which is the entry point the CI uses to run tests within docker.

I usually do this to test within docker:

```
cd snabb/src
# To run e.g. the Snabb NFV selftest:
SNABB_PCI0=0000:01:00.0 scripts/dock.sh make program/snabbnfv/selftest.sh
# Or to get an interactive shell, to use e.g. snabb top to interact
# with a proces running in the container.
DOCKERFLAGS=-t SNABB_PCI0=0000:01:00.0 scripts/dock.sh bash
```

PR #874 (2016-05-12): Import snabblab docs from the wiki into the manual
Fixes #864

I've modified it a bit and added minimal `Using the lab` and `Questions` sections.

After this is merged, we should update the wiki to point to the manual.

cc @lukego @eugeneia
closed 2016-05-12 by eugeneia