snabbco/snabb #865: Describe testing framework for developes

plajjan, 2016-04-11:
Opening issue instead of discussing on Slack. While https://github.com/snabbco/snabb/blob/master/src/doc/testing.md mentions how to run tests in various ways it doesn't really cover anything on the topic of *how to write* a test, a test suite or where to place / structure it.

In my head there are generally two types of tests;
 * functional tests that make sure that the app does what it's supposed to, like drop a packet based on an ACL
 * performance testing, i.e. how fast it does something

Both of these can be run in the context of a single app or in a larger context, like with OpenStack, where it becomes more of integration testing. Also note how the tests don't necessarily need to be different, the same test case can be used for functional testing and performance testing, however the environment around it needs to be much more closely controlled for performance testing and return values need to look different.

Functional testing is fairly straight forward to test. We already have selftests and if they return 0 all is good and if they return 1 something went bad (although not all the selftest() functions use os.exit(1) on failure). Many of the apps only have a single test case defined. I haven't looked closely enough to know if this is because the app is so simple that a single test case is enough (I doubt it) or if what could have been multiple tests have been flattened because of lack of precedent on how to write many tests (more likely IMHO). Python unittests have the concept of test case (smallest unit - a single test) grouped into test suites. I think we should have something similar for Snabb so we have some more granularity in testing.

For performance we obviously need to measure pps/bps and not just a binary return value so either the test function (selftest() ?) need to be modified to return something else or we always run these from snabbmark? In addition the result need to be compared with something - possibly a previous commit, i.e. everything is baselined from an earlier version. We could define threshold values such that a perf test fails if the performance drops with more than X% (long discussion to follow this I'm guessing ;)

Maybe the whole test fixture is rigged for perf tests, i.e. it's not the selftest() function that sets up the app network but something else.

I find the structure of snabbmark and packetblaster to be a bit weird. Overall it's tricky with Snabb since Snabb is somewhat akin to a library / framework. Usually I write my own program and just import the libraries I need but the suggested development model for Snabb seems to be to put your own program into the directory structure of Snabb. When I fork Snabb to create SnabbDDoS and people go to my git repo they won't see my readme file first - they'll see that of Snabb and programs are being upstreamed into a single repo which I'm not convinced is the right model. Anyway, my point is that if this is as popular as any other library then we'll get a complete overload of applications like lwAFTR, SnabbWall or similar. This repo will explode and the tests have to be structured. The tests in https://github.com/snabbco/snabb/blob/master/src/program/snabbmark/snabbmark.lua cover very different areas - how will it look with 100 different applications?

Maybe this issue needs to be broken down into smaller issues. I've just dumped a few thoughts from the top of my head :)

eugeneia, 2016-04-12:
> For performance we obviously need to measure pps/bps and not just a binary return value so either the test function (selftest() ?) need to be modified to return something else or we always run these [...]

SnabbBot is doing most of what you are describing already, see the green ticks next to commits on PRs. The interface it uses for that is `make benchmarks`, and the individual benchmark entry points are in `src/bench/`. Right now there are too little benchmarks enabled, everybody is welcome to populate `src/bench/`!

Snabbmark and `make benchmarks` are not properly documented because we are pretty sure that they are not the end of the story. We needed *some* place to put micro benchmarks and *some* interface for CI so that's where we are at now. I expect at least `snabbmark` to be replaced by a better solution akin to #688 in the near future.

> I find the structure of snabbmark and packetblaster to be a bit weird. [...] how will it look with 100 different applications?

One reason for why it is convenient to have a bunch of real world applications and various benchmarks in `snabbmark` is so that we have a good use-case coverage while developing the Snabb core framework. If I decide its a good idea to replace core API X with core API Y I can see what impact that change has on the bulk of the code base. If I decide to “optimize” core API Z I can use `snabbmark` to confirm the performance impact with more use cases than I had in mind. In my opinion having 100 “snabbmarks” would be a good thing.

The reason we have multiple Snabb applications in the upstream repository is that we find it is the most effective way of collaborating and benefiting from changes. We do believe that this workflow is “Linux scale”. See #746. Nothing should prevent you from changing the README in your fork though.

plajjan, 2016-04-14:
Of course it's convenient to have some real programs to test with but that doesn't necessarily mean it should be included in the same git repo. To draw on your kernel analogies I bet it's important to test various kernel syscalls with actual user space apps yet the kernel git repo doesn't include user space applications.

I know Snabb has adopted the Linux merge model and I have nothing to say about that in itself, but you cannot expect everything else to follow that model. I think it's awkward to ship my app hidden deep inside the source tree of snabb. Is this now the main repo of my app? Is this where users should open issues? With the changes you just asked me to do on lwAFTR to get the VLAN app accepted in here I have made changes to lwAFTR as if this was the master for lwAFTR where in reality I think igalia/snabb is the master. Are you going to accept or reject my changes to the app (SnabbDDoS) that I've written? I've received feedback that people don't understand where the source to my program is and it certainly isn't very obvious - you have to be familiar with the snabb eco system to know where to look. I can rewrite the top level README but that will likely give me a headache every time I pull in the latest master.

I don't have an answer to where developers of programs using Snabb should place their code or how the overall structure should be. I just feel that the current solution is suboptimal.

plajjan, 2016-04-14:
@eugeneia are you happy with functional testing as is? i.e. a selftest() in each module that does varying things? Thoughts on using proper unit test framework with concepts of test units / cases?

lukego, 2016-04-14:
Background: There is an incoming PR (#835) with [documentation about the current maintenance model](https://github.com/lukego/snabb/blob/doc-git-workflow-redux/src/doc/git-workflow.md#becoming-a-maintainer). This suggests maintaining applications within a unified source tree (and git repo) with clearly established maintainer roles (e.g. @plajjan decides what changes to merge in `program/ddos`).

This is indeed tighter coupling than many other projects use. This is bound to create some tension but hopefully also cohesion. I think this model is the Linux kernel one fairly purely. Snabb applications are like major Linux features such as bridge, router, iptables, etc that are all maintained in one unified tree.

So @plajjan here is what I would suggest for SnabbDDoS:

1. We create a `ddos` branch on this repo that only you can push to. You can push the latest development version of SnabbDDoS here.
2. People can open Pull Requests for SnabbDDoS here. You are the "upstream" (Assignee) for those and take the lead on review and decide when to merge. (You can even ask people to rebase if you like, you are the upstream :-)).
3. You can "release" SnabbDDoS by opening a Pull Request from `ddos` to a [next-hop branch](https://github.com/lukego/snabb/blob/doc-git-workflow-redux/src/doc/git-workflow.md#becoming-a-maintainer) where it will swiftly land on master and be included in the monthly releases.

Then SnabbDDoS is being regularly released together with Snabb. You can still decide whether to make your own releases separately from other branches (e.g. if you want to have an LTS release, etc, and we haven't worked out those details on the upstream repo yet).

This way the `snabbco/snabb` repository acts as a central point for maintaining all of the applications together. This is like the Linux Kernel Mailing List (LKML) in the earlier days i.e. a central place to send patches where the appropriate maintainer will process them.

This model can be generalized and excepted in the future:

- If you prefer for Issues and PRs to be submitted to a separate fork e.g. `plajjan/snabb` then we can inform people that this is the process. This may not be necessary in the beginning but it is your call. The LKML has divided up into [many topic-specific mailing lists](http://vger.kernel.org/vger-lists.html) over time and Snabb likely will too. (This seems to be how lwaftr is operating for example.)
- If you prefer to have a separate Git repository e.g. `plajjan/ddos` then we could invent a simple mechanism to link Snabb as a Git submodule or subtree. This would provide you with more separation between your application and the Snabb upstream community if you prefer that.

I will continue to push for the kernel model - unified git repo, well-defined maintainer structure, applications developed in-tree - as the operating model for the `snabbco/snabb` repo and try to help people climb onboard and become maintainers of applications/modules/documents/aspects. I like the social side of this and find it practical even though it is not without its frustrations. This is not the only game in town though and application developers absolutely get to opt-in and opt-out whenever they like.

lukego, 2016-04-14:
Coming back to testing:

If we have 100 programs in the tree then they should each have their own test cases developed under `program/foo`. The CI is flexible and will try to accommodate those tests independently of how they are written. Today if you would just put a `selftest.sh` function in a directory it will automatically be executed by CI and can do anything that it wants. Over time we can teach the CI more fancy tricks e.g. if a performance test returns a CSV file then we could generate a comparison between releases.

Snabbmark is not a central point that needs to test all applications. Rather it is one convenient place to put reasonably generic benchmarks to make them easy to run.

If people would like to make the test framework more sophisticated then a good start would be to try a new way locally for an application, e.g. under `program/ddos/`, and if it works well then suggest that other people adopt it too. This way everybody can try their own styles and the convenient once can catch on. (The simple `selftest` model we have now is borrowed from Sun Openboot / Openfirmware btw.)

lukego, 2016-04-14:
@plajjan I added you to the @snabbco/maintainers group on Github. If you want you can push your DDoS code upstream like this:

```
git remote add snabbco https://github.com/snabbco/snabb
git push snabbco ddos
```

and then SnabbDDoS will appear on the [branches list](https://github.com/snabbco/snabb/branches/all) and I can mark that branch as protected so that you own it.

lukego, 2016-04-14:
> I can rewrite the top level README but that will likely give me a headache every time I pull in the latest master.

Good point!

How about if the first section of the README would be links to the subdirectories of the in-tree applications? (ddos, packetblaster, nfv, lwaftr, etc.) This is much more important than the *blah blah blah* parts for people looking for the code to some application.

plajjan, 2016-04-14:
I think the kernel analogy is halting. When I start the kernel.. I start the kernel. I then go and configure it in various ways, maybe load some modules or set some sysctls, but there is one "entry point". Similarly, I can start a Cisco IOS router and then go and configure various features of it. Snabb on the other hand doesn't work like that. We have multiple entry points, namely the "programs", thus it doesn't look like one cohesive system. It is a library/framework that people use to write their own programs. It's more similar to the Python Twisted library/framework on top of which you can base your program - like https://github.com/scrapy/scrapy (but that lives in separate git repos).

Now, there is nothing stopping Snabb from becoming a cohesive system. VPP already is. You start the thing, you get a CLI and it does basic packet forwarding things out of the box, like it has an IPv4 RIB, IPV6, MPLS, does VRFs and so forth. You can write plugins for it, and if SnabbDDoS would be written for VPP I suppose it would be a plugin that would be inserted into the node graph between two other "default" nodes.

There are of course benefits and drawbacks to this approach. One drawback is that you kind of end up in a bit of the same situation as with kernel forwarding - one of the reasons it is slow is because there's so much features that have to be checked for every packet. With Snabb you can write something that is super-duper optimised for a specific use case and you don't have to waste a single CPU cycle on checking something you know you don't need.

So, I think to me, upstreaming all apps makes sense if Snabb acts as one system. Right now it acts like a library / framework and thus it doesn't make sense.

If I'm in charge of ddos then noone else should be allowed to make changes, which is in direct violation of what @eugeneia asked me to do for the VLAN PR. How are the Igalia folks supposed to handle that I've changed their code without them approving or it landing on their repo? (This is why I wanted to do a two-stage rocket out of that and not squeeze it into one PR). 

Anyway, this has consequences to end users. I think the Snabb eco system is built for developers and that is great... for developers. But if programs are the end product then they will be shipped as such, they will be documented as such and so forth. A network engineer wants to deploy an lwAFTR so he googles for "open source lwAFTR" (go ahead, try it!), finds igalia, ends up with the Snabb top level README and has no idea how to proceed (okay, granted this is also lacking in other areas but it is confusing). If he just wants an AFTR why isn't it packaged like that? Let's say I want to build a docker container out of SnabbDDoS so I setup a Dockerfile in the git repo root, oh wait.. ;)

PS. I don't have a complete suggestion on how I want things to look, I know what parts I find weird today but what might come across as suggestions should be regarded as input to a discussion :)

lukego, 2016-04-14:
@plajjan Yes code ownership becomes tricky :).

The VLAN code you changed was in `apps/` which is a directory tree for shared code that is supposed to be collaboratively developed and extended and adopted by more applications. The LWAFTR developers have hopefully chosen to put that app here to benefit from community work exactly like what you have done. If the intention were for this to be a private "hands off" app then the implementation could be in `program/lwaftr/` instead.

It is also possible that this change will break something for them. The solution I see is that they will ship the lwaftr to users from their own branch(es) for production use. The `master` branch is then more of a synchronization point with other developers. They can avoid breakages in merges from master by reviewing the changes they merge and by contributing tests to the upstream CI.

This is still the kernel model. Redhat don't ship Linus's tree to their customers verbatim but they do use it for synchronizing their kernels with development from everybody else in the Linux universe.

You are right though that we are still working out the details for all these issues. I quite like the idea of people distributing binaries like `snabb-ddos` and `snabb-lwaftr` that they have built from their own production/release branches. Our CI could also build releases of individual applications from their own release branches rather than from master. We will have to see what works well in practice.

plajjan, 2016-04-14:
> How about if the first section of the README would be links to the subdirectories of the in-tree applications? (ddos, packetblaster, nfv, lwaftr, etc.) This is much more important than the blah blah blah parts for people looking for the code to some application.

For a developer it is important to get some introduction to Snabb - the framework, while an end user probably cares only for the programs.

Will a long list of programs, that target end users, scale? Maybe I'm getting ahead of myself. It's not a very long list right now, so perhaps it's fine ;)

plajjan, 2016-04-14:
@lukego have a look at the PR #863 again. @eugeneia asked me to update the current uses of the VLAN app so I've changed programs/lwaftr/setup.lua to use apps.vlan.vlan instead of apps.lwaftr.vlan

lukego, 2016-04-14:
I saw that. This seems like a small and helpful thing to do i.e. updating the users of an app/library after changing its interface. I see being able to do this as a big benefit to having all the code in one repository with common CI coverage. Can certainly be worth @mentioning the lwaftr team so that it doesn't surprise them though.

If it were a more substantial change then I would expect to get their sign-off.

This kind of issue is what I am responsible for considering when @eugeneia sends your VLAN change upstream to me in a Pull Request to `next`. The code passes a few steps on the way to master and everybody uses their best judgement along the way and hopefully we don't screw up too often.

kbara, 2016-04-14:
I see Snabb as using a hybrid between the Linux model and the huge Google monorepo model at this point, to get the kinds of advantages http://danluu.com/monorepo/ mentions.

lukego, 2016-04-20:
Great reference to the Google monorepo! That does resonate with me and seems like basically what I am trying to describe. Thanks for posting that link!

I do also see this as a continuum with Snabb perhaps somewhere between the kernel and the Google/Facebook repos. Over time it will be interesting to see the evoluation of what functionality we want to have in-tree and what we want to interface externally to.

eugeneia, 2016-04-25:
> @eugeneia are you happy with functional testing as is? i.e. a selftest() in each module that does varying things? Thoughts on using proper unit test framework with concepts of test units / cases?

@plajjan I personally like our simple selftest convention, and have had mostly negative experiences with unit test frameworks (imho little value, and usually you spend lots of time working around their restrictions, ultimately the tests become very artificial). We have functional unit tests as well as minor and major integration tests accessible in a uniform way (make), and I think pure unit tests are rather the exception than the norm (currently), most selftests perform some kind of integration testing (using NIC, OS capabilities, ...). So from my point of view a unit test framework would be overkill for the few tests it would apply well to.