snabbco/snabb #871: Standardise port names for NIC type apps

plajjan, 2016-04-14:
I think it would make sense to standardise the port names for NIC type apps. Intel app calls it rx/tx where Tap uses input/output. In general I think it would be useful if we could adhere to a uniform standard for simple apps (where there is only input/output).

Personally I would prefer to rename the Intel ports to input/output. This was one of the first things I ran into with Snabb when I started playing with it 2 years ago. Somewhat embarrassingly I put myself in the same position just last week and it took me way to long to notice that I was using the wrong port names.

Having the same names would make it easier to write apps that can easily switch between multiple types of NICs.

lukego, 2016-04-14:
Yes! Consistent naming has been on my list forever too. I agree that `input` and `output` are better names than `rx` and `tx` too.

The only alternative I see worth considering is anonymous ports accessed by index. Then the name is irrelevant. This is already implemented but I know that you recently noticed problems with the implementation (can't distinguish between the first port or a port named `1`) so I would be okay with removing that anonymous-ports feature and always using names instead.

eugeneia, 2016-09-21:
I want to make this change because it enables the “common I/O” API we need. So, are we set on input/output over rx/tx? The statistics framework currently uses `{rx,tx}{bytes,packets,errors,...}` as standard names, should these be changed to `{input,ouput}{bytes,packets,errors,...}` accordingly?

petebristow, 2016-09-21:
I would certainly prefer `input` / `output`.
Yes I think all references to rx/tx should become input / output.

eugeneia, 2016-09-23:
I opened https://github.com/snabbco/snabb/pull/1025 to resolve this issue. Help is welcome!