snabbco/snabb #891: Test compilation on all distros

lukego, 2016-04-22:
I propose that we add CI coverage to ensure that Snabb builds successfully on every major Linux distribution's userland for all versions that are not end-of-life.

This would basically require running `make` in containers or VMs running all supported versions of Redhat, Debian, Ubuntu, etc.

Easy or hard, @domenkozar?

lukego, 2016-04-22:
... even better would be to use VMs and actually run a part of the test suite that includes PCI device access. Then we would get compatibility testing with diverse important kernel versions.

domenkozar, 2016-04-22:
Relatively easy. Depending on what distros to support. We have most `deb` and `rpm` distros already prepared and installed inside the VM: https://github.com/NixOS/nixpkgs/blob/master/pkgs/build-support/vm/default.nix#L753

Nix itself is tested that way so it installs: http://hydra.nixos.org/eval/1251528

lukego, 2016-04-24:
Nice!

I had a play around with this and I see that it works, I can reproduce the problem from #867 (failure to build on CentOS 6 due to older GCC):

```nix
with import <nixpkgs> {};
with vmTools;

let snabb = stdenv.mkDerivation rec {
  name = "snabb-${version}";
  version = "2016.04";
  src = fetchurl {
    url = "https://github.com/SnabbCo/snabbswitch/archive/v${version}.tar.gz";
    sha256 = "1agrcd19jf28cmar8452qm42f53hxzbvrjzq1a4rm5gv1rhpb3yd";
  };

}; in

runInLinuxImage (snabb // { diskImage = diskImages.centos65x86_64; })
```

That boots a CentOS 6.5 VM and gets a compilation failure:

```
Formatting '/tmp/nix-build-snabb-2016.04.drv-0/disk-image.qcow2', fmt=qcow2 size=4294967296 backing_file=/nix/store/vlsa5ikv51cc2qbmpr6vpcz68qabfs2w-centos-6.5-x86_64/disk-image.qcow2 encryp
tion=off cluster_size=65536 lazy_refcounts=off refcount_bits=16
loading kernel modules...
[    0.360209] EXT4-fs (vda): couldn't mount as ext3 due to feature incompatibilities
mounting Nix store...
mounting host's temporary directory...
starting stage 2 (/nix/store/y64r5c1m6khhy042qm3jmlv12ykq5871-vm-run-stage2)
unpacking sources
unpacking source archive /nix/store/qziji771avy740jpg9w4lin27sy78nrq-v2016.04.tar.gz
source root is snabb-2016.04

...

C(AVX2)   obj/arch/avx2_c.o
cc1: error: unrecognized command line option "-mavx2"
make[1]: *** [obj/arch/avx2_c.o] Error 1
make[1]: Leaving directory `/tmp/snabb-2016.04/src'
make: *** [all] Error 2
[   20.505023] reboot: Power down
```

This should make it possible for us to test Snabb with both the userspace and kernel of all the major distros including relatively archaic versions. This would be quite awesome. Since Snabb is a very low-level program it is extremely interesting to test with diverse host kernels and that is not something we get with e.g. Docker containers.

Seems like your Nix code for our OpenStack test suite has the right ingredient for passing in PCI devices: https://github.com/domenkozar/snabb-openstack-testing/blob/master/tests.nix#L28-L30. How would I add that to the script above?

domenkozar, 2016-04-24:
You should be able to set attribute `QEMU_OPTS` on your `mkDerivation`.

You'll also need to run qemu as sudo, which is a bit tricky. Replace:

`with import <nixpkgs> {};` to

```
with import <nixpkgs> {
  config.packageOverrides = pkgs: {
    vmTools.qemuProg = "/var/setuid-wrappers/sudo ${pkgs.qemu}/bin/qemu-kvm";
  };
};
```

That's not enough, the whole build is done in chroot, so you have to escape it (I just got an idea how to do this better using https://github.com/snabblab/snabblab-nixos/issues/33).

```
lib.overrideDerivation (runInLinuxImage (snabb // { diskImage = diskImages.centos65x86_64; })) (attrs:
  { __noChroot = true; requiredSystemFeatures = [ "performance" ] }
);
```

domenkozar, 2016-04-24:
We should make this easier eventually once https://github.com/snabblab/snabblab-nixos/issues/33 is implemented.

lukego, 2016-04-24:
Seems that if I use `runInLinuxImage` to `runInLinuxVM` then it will still run inside a VM with the distro of my choice but it will build the software using Nix instead of the host tools?

This would seem like a nice way to separately test running a standard binary vs one built on the distro under test.

domenkozar, 2016-04-25:
`runInLinuxVM` doesn't respect `diskImage`, so you're using plain NixOS in that case.

lukego, 2016-04-25:
Is Hydra active at the moment?

I defined a Hydra job to build on all known Linux distros - knowing that most will fail - but it does not seem to be executing. https://hydra.snabb.co/jobset/snabb/compile-distros

lukego, 2016-04-25:
oh, actually now it did execute, but seemingly half an hour after I requested the build. and seems to have stopped on the first error while I was hoping it would separately test each platform. Any ideas?

domenkozar, 2016-04-25:
Looking at https://github.com/NixOS/nix/blob/master/release.nix#L284, there's a bit more setup involved.

domenkozar, 2016-04-27:
Scratch that, that's only needed if you want to build `.deb` or `.rpm` packages. You very close, see https://hydra.snabb.co/eval/478

domenkozar, 2016-04-29:
@lukego this is ready for fixing in snabb :)

Fails on 
(Ubuntu 15.10, Fedora 23, Debian 8.4):

```
HOSTLINK  host/minilua
host/minilua.o: In function `luaS_newlstr':
minilua.c:(.text+0x3e5b): undefined reference to `memcpy@GLIBC_2.2.5'
host/minilua.o: In function `luaV_concat':
minilua.c:(.text+0x4ffa): undefined reference to `memcpy@GLIBC_2.2.5'
host/minilua.o: In function `luaL_addvalue':
minilua.c:(.text+0x930f): undefined reference to `memcpy@GLIBC_2.2.5'
collect2: error: ld returned 1 exit status
```

Fails on CentOS 6.5:

```
cc1: error: unrecognized command line option "-mavx2"
```

domenkozar, 2016-08-03:
For anyone interested, we're now doing building Snabb on a few different platforms. See https://hydra.snabb.co/eval/2832