snabbco/snabb #915: Snabb bot coverage for i350

petebristow, 2016-05-10:
#897 adds enhances the i350 support. It would be great to get snabb bot test cover running for this. This issue is to track the logistics of getting a card installed in the correct machine. @lukego

lukego, 2016-05-10:
@eugeneia How about if I put an I350 in davos?

eugeneia, 2016-05-10:
@lukego In that case it would be trivial to run the I350 selftest, do we want to run the SnabbNFV integration tests using the I350 as well though?

lukego, 2016-05-10:
Just for now I think that only selftest is realistic/supported.

lukego, 2016-05-16:
I installed a dual-port I350 NIC into `davos` and cabled the ports together.

eugeneia, 2016-05-16:
Which one is it though?

```
max@davos:~/snabbswitch/src$ lspci | grep I350
01:00.0 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
01:00.1 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
05:00.0 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
05:00.1 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
0a:00.0 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
0a:00.1 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
```

I have unbound `01:00.0` and SnabbBot will use it from now on.

closed 2016-09-21 by eugeneia