snabbco/snabb #924: Ingress drop counter for Intel10G not working for vmdq=true

wingo, 2016-05-19:
Garrr, there is a limitation in #909 -- it works fine for 82599 cards in single-function mode, but in VF mode it doesn't work unless you pass `rxstats=0` in the app config.  This is because in VF mode we need to map virtual statistics to one of the global statistics counters, which are scarce relative to the number of possible VF's.

I propose to default rxstats and txstats to 0 if they are not set for vmdq=true.  WDYT @eugeneia @mwiget @dpino @lukego ?

wingo, 2016-05-19:
This is an instance of the class of bugs also including #749.  Of course the fix here will be more limited.

eugeneia, 2016-05-19:
I think your proposal is sensible. :+1: The limited per-queue statistics register arrays of the 82599 are a constant culprit for me as well...

commit 2b8b453f (2016-05-23): Intel VF interfaces default to statistics counter 0.

Fixes #924.
PR #925 (2016-06-09): Intel 82599 drivers default to RX/TX statistics counter index 0
This PR is based on `next` since it modifies the ingress drop counter code.  Fixes #924.
closed 2016-06-09 by eugeneia