snabbco/snabb #940: snabb.co landing page webmaster?

lukego, 2016-06-10:
Is there anybody who would like to volunteer to create a simple website (landing page) for Snabb that we can host at http://snabb.co/?

I would imagine the requirements to be:

- Make an attractive page that reuses the Snabb logo / color scheme.
- Link to all of the most relevant resources in the Snabb universe (repos, projects, companies, etc).
- Maintain in a simple way e.g. Github pages on a snabbco repo.

Back in #738 we decided that the name "Snabb" would be transferred from my little company over to the project as a whole and this seems like an important step. I will separately create a new page for my professional activities on another domain.

P.S. Could also create `@snabb.co` email aliases for all the maintainers if that seems interesting.

eugeneia, 2016-08-23:
Since nobody has come forward for a while, how about we crowdfund a capable web designer to do it? For instance, I’d consider one of my office mates capable, this would probably be easy work for him.

eugeneia, 2016-11-08:
Resolved in #1050

closed 2016-11-08 by eugeneia