snabbco/snabb #960: virtio/map_from_guest fails to map the host address

domenkozar, 2016-07-01:
# Description

I've seen this one happen quite often in benchmarking, etc.

# Reproduce 

Using `refactor/matrix` branch in [Snabblab](https://github.com/snabblab/snabblab-nixos) one can run following command to reproduce on lugano servers (it's using Snabb `next` branch):

```
$ nix-build -j 8 jobsets/snabb-matrix-packet-ABC.nix --arg snabbAname '"master"' --arg snabbAsrc ~/snabb --arg numTimesRunBenchmark 30 --arg benchmarkNames '[ "dpdk" ]' --arg reportName '"report-by-snabb"' -A benchmarks.l2fwd_snabb=master_dpdk=2-2-0_qemu=2-5-1-num-1
...
lib/virtio/net_device.lua:363: mapping to host address failedcdata<void *>: 0xffffffffffffffff
stack traceback:
        core/main.lua:130: in function <core/main.lua:128>
        [C]: in function 'error'
        lib/virtio/net_device.lua:363: in function 'map_from_guest'
        lib/virtio/virtq_device.lua:39: in function 'get_desc'
        lib/virtio/virtq_device.lua:63: in function 'get_buffers'
        lib/virtio/net_device.lua:128: in function 'receive_packets_from_vm'
        lib/virtio/net_device.lua:114: in function 'poll_vring_receive'
        apps/vhost/vhost_user.lua:96: in function 'method'
        core/app.lua:86: in function 'with_restart'
        core/app.lua:323: in function 'thunk'
        core/histogram.lua:135: in function 'breathe'
        ...
        program/snabbnfv/traffic/traffic.lua:63: in function 'run'
        program/snabbnfv/snabbnfv.lua:17: in function 'run'
        core/main.lua:49: in function <core/main.lua:36>
        [C]: in function 'xpcall'
        core/main.lua:172: in main chunk
        [C]: at 0x00451a90
        [C]: in function 'pcall'
        core/startup.lua:3: in main chunk
        [C]: in function 'require'
        [string "require "core.startup""]:1: in main chunk
```

lukego, 2016-07-02:
My impression is that this issue has appeared with QEMU 2.5.1 and was not present with QEMU 2.4. I would quite like to `git bisect` the QEMU repo between those releases to identify which change causes the issue.

Is there a straightforward way to do this i.e. run the test using a QEMU from a local directory rather than a release tarball?

lukego, 2016-07-02:
Looking over the logs this seems to be related to the guest handing over the Virtio-net device from its kernel driver (loaded at boot time) to the DPDK driver. Both on QEMU 2.4 and QEMU 2.5 there is a suspicious burst of ~500K packets before the DPDK driver renegotiates the device features. Have to determine the source of this burst and why with QEMU 2.5 we receive an illegal pointer value for a packet.

Historically our CI benchmarks for DPDK have not included a hand-over of the Virtio-net device from the Linux kernel to DPDK. Instead we kept it simple by disabling Virtio-net on the kernel so that the device was dedicated to DPDK. This probably explains why we are suddenly seeing this problem since we switched to the new guest image that is bootstrapped with nix.

Here is a quick summary of the relevant bits of the logs:

### QEMU 2.4

First the Snabb process starts and negotiates features with the VM. The feature list is quite extensive, including checksum offload, and this is typical of a Linux kernel driver in the VM (not DPDK).

```
snabbnfv traffic starting (benchmark mode)
Loading program/snabbnfv/test_fixtures/nfvconfig/test_functions/snabbnfv-bench1.port
engine: start app B_NIC
engine: start app B_Virtio
Get features 0x18428001
 VIRTIO_F_ANY_LAYOUT VIRTIO_NET_F_MQ VIRTIO_NET_F_CTRL_VQ VIRTIO_NET_F_MRG_RXBUF VIRTIO_RING_F_INDIRECT_DESC VIRTIO_NET_F_CSUM
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
vhost_user: Caching features (0x18028001) in /tmp/vhost_features_vhost_B.sock
Set features 0x18028001
 VIRTIO_F_ANY_LAYOUT VIRTIO_NET_F_CTRL_VQ VIRTIO_NET_F_MRG_RXBUF VIRTIO_RING_F_INDIRECT_DESC VIRTIO_NET_F_CSUM
```

Then we see a period of idleness punctuated by an unexplained burst of ~500K packets in one second:

```
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
No LuaJIT profiling enabled ($NFV_PROF unset).
No LuaJIT dump enabled ($NFV_DUMP unset).
load: time: 1.00s  fps: 533,934   fpGbps: 0.312 fpb: 50  bpp: 64   sleep: 100 us
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
```

Then we see the Virtio-net features renegotiated in a simpler way, presumably by DPDK, and traffic is processed:

```
vhost_user: Caching features (0x00028000) in /tmp/vhost_features_vhost_B.sock
Set features 0x28000
 VIRTIO_NET_F_CTRL_VQ VIRTIO_NET_F_MRG_RXBUF
rxavail = 0 rxused = 0
vhost_user: Connected and initialized: vhost_B.sock
rxavail = 0 rxused = 0
load: time: 1.00s  fps: 6,889,568 fpGbps: 4.024 fpb: 208 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 6,965,276 fpGbps: 4.068 fpb: 208 bpp: 64   sleep: 0   us
load: time: 1.00s  fps: 6,899,028 fpGbps: 4.029 fpb: 208 bpp: 64   sleep: 0   us
```

### QEMU 2.5

QEMU 2.5 starts in the same way with the Linux kernel apparently initializing the device first:

```
snabbnfv traffic starting (benchmark mode)
Loading program/snabbnfv/test_fixtures/nfvconfig/test_functions/snabbnfv-bench1.port
engine: start app B_NIC
engine: start app B_Virtio
Get features 0x18428001
 VIRTIO_F_ANY_LAYOUT VIRTIO_NET_F_MQ VIRTIO_NET_F_CTRL_VQ VIRTIO_NET_F_MRG_RXBUF VIRTIO_RING_F_INDIRECT_DESC VIRTIO_NET_F_CSUM
Get features 0x18428001
 VIRTIO_F_ANY_LAYOUT VIRTIO_NET_F_MQ VIRTIO_NET_F_CTRL_VQ VIRTIO_NET_F_MRG_RXBUF VIRTIO_RING_F_INDIRECT_DESC VIRTIO_NET_F_CSUM
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
vhost_user: Caching features (0x18008001) in /tmp/vhost_features_vhost_B.sock
Set features 0x18008001
 VIRTIO_F_ANY_LAYOUT VIRTIO_NET_F_MRG_RXBUF VIRTIO_RING_F_INDIRECT_DESC VIRTIO_NET_F_CSUM
```

Then we have an idle period followed by the same burst of ~500K packets but followed immediately by an error:

```
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
No LuaJIT profiling enabled ($NFV_PROF unset).
No LuaJIT dump enabled ($NFV_DUMP unset).
load: time: 1.00s  fps: 497,213   fpGbps: 0.290 fpb: 47  bpp: 64   sleep: 100 us
lib/virtio/net_device.lua:363: mapping to host address failedcdata<void *>: 0xffff88001fa19600
```

So the questions are: what are those packets being sent before DPDK has taken over the NIC? and why the illegal address?

Could be e.g. that somehow the Virtio packet descriptor memory becomes garbage for a period of time.

domenkozar, 2016-07-12:
I'll prioritize this to get qemu bisected. I've counted occurrences of errors and this one is the most frequent.

[on lugano servers](https://hydra.snabb.co/eval/2417):

```
Failed builds in total: 305
Failed builds due to 'mapping to host address failedcdata': 166
Failed builds due to being terminated after 2min: 72
Failed builds due to 'packet payload overflow': 31
Failed builds due to qemu telnet timeout: 0
Failed builds due to 'no server running on /tmp/tmux-0/default': 36
```

[on murren servers using softnic](https://hydra.snabb.co/eval/2380):

```
Failed builds in total: 2932
Failed builds due to 'mapping to host address failedcdata': 1777
Failed builds due to being terminated after 2min: 408
Failed builds due to 'packet payload overflow': 309
Failed builds due to qemu telnet timeout: 3
Failed builds due to 'no server running on /tmp/tmux-0/default': 427
```

PS: I've yet to check if any of these issues were due to lugano-4 having `intel_iommu=pt` set.

lukego, 2016-07-12:
Great overview! How do you do the counting?

domenkozar, 2016-07-12:
@lukego wrote a hacky script: https://github.com/snabblab/snabblab-nixos/blob/master/scripts/hydra_parse_logs.py

domenkozar, 2016-08-11:
Qemu/dpdk custom source support landed in https://github.com/snabblab/snabblab-nixos/pull/62

This issue appears to be fixed on `next`.

```
$ ./scripts/hydra_parse_logs.py --url "https://hydra.snabb.co/eval/2913?full=1"
...
Failed builds in total: 2341
Failed builds due to 'mapping to host address failedcdata': 0
Failed builds due to being terminated after 2min: 0
Failed builds due to 'packet payload overflow': 0
Failed builds due to qemu telnet timeout: 0
Failed builds due to 'no server running on /tmp/tmux-0/default': 2341
Failed builds due to 'lib/virtio/net_device.lua:254: assertion failed': 0
Failed builds with unknown cause: 0: []
```

closed 2016-09-07 by domenkozar