snabbco/snabb #1033: Introduce named programs to Snabb

xray7224, 2016-10-14:
As part of the work to support a NETCONF-managed system through YANG modelling (issue #987) we need to be able to assign names to each Snabb program running on a system. This name will be used by the user in the `snabb config` command to specify which program they wish to get/set/etc. on.

The name should be unique string on the system which can be enumerated along with their corresponding PIDs. This then allows for sending messages between processes using their names.

My proposal for implementing this would be to have each program specify its own unique name via a new function in `core.app`. Setting the name would be optional, this is because most programs, at least in the beginning likely won't use this feature until they have a need for it. The function would look something like this:

```lua
function run (params)
    ...
    engine.set_name("myprogram-1")
end
```

The reason for using setters and getters are because upon setting the name, three things will occur, these are:

1. Sets a "name" attribute on the configuration of the running app which can be retrieved later by a `get_name` function.

2. Validates that the name is unique on the system

3. Creates a directory with the PID in `/var/run/snabb/` and creates a symlink in `/var/run/snabb/by-name/<program name>`

These three things ensure the names remain unique on the system using validation and allows for enumeration of programs by their names as mentioned above. The third job of creating the directory and symlink is done at this stage largely for validation however, the setup done in this step will be used by the Shared Memory functionality in `core.shm`.

The NETCONF-managed system will use, as laid out in #987, `core.shm` to create a ring buffer to pass messages between the config app and the program being configured. The user will specify the app whilst using `snabb config` which will then find the shared memory and communicate a set/add/delete/etc. to the program.

wingo, 2016-10-14:
Looks good in general, note though that the PID directory gets created already.  I.e. anonymous snabb processes have those dirs already.  Maybe it should be "claim_name()" or something, as it is an exclusive resource, and claiming a name could fail.  Dunno :)

lukego, 2016-11-01:
Just curious: How come the PR for this issue is being reviewed on the Igalia repo (Igalia/snabb#518) instead of here on the upstream snabbco repo? This is fine of course but I see a risk that feedback from people here will come late and potentially create more work for you.

lukego, 2016-11-01:
(Sorry for not providing some feedback on the idea already btw. Just been busy. I will try to provide some useful input when the code is PR'd to this repo.)

lukego, 2016-11-01:
(Generally could make sense for somebody to volunteer as the "subsystem maintainer" for Snabb process management. We have a bunch of PRs in this area coming from different applications. Could be really practical to have a branch like `snabbco/process` that is owned by one person who will review and merge such changes before proposing them to an upstream `-next` branch. Having topic-specific subsystem branches is the next logical step in our Git workflow evolution but it seems like these things have to happen at their own pace.)

wingo, 2016-11-01:
I think the review is happening in Igalia/snabb because that's where we expect to merge it -- we'll need it soon.  But yeah, this kind of change is tricky as we don't want to create a situation where a downstream change in Igalia/lwaftr prevents an upstream merge.  There's a tension between getting the right upstream review and having a common work-in-progress space for downstream teams.

We're definitely interested in upstream feedback (hence this issue:) so we'll open a PR against upstream then.

lukego, 2016-11-01:
Feedback is that it seems like you could do the work on a topic branch based on `master` with a PR on `snabbco/snabb` repo for review, and you could still merge that into the downstream lwAFTR branch any time you want. This way you have the benefit of early review of the changes that affect everybody upstream and you also have the flexibility of deciding exactly what to merge and when in your downstream branch.

This is what I am doing with LuaJIT for example where I have a few PRs open on upstream (e.g. LuaJIT/LuaJIT#224) but that is mostly for early visibility and my intention is to merge these changes directly onto my own branches without waiting for any approvals.