snabbco/snabb #1041: Question: TCP Stack using Snabb

rohitjoshi, 2016-10-17:
Is there any implementation of tcp stack using snabb switch?

lukego, 2016-10-17:
There is no native TCP stack for Snabb. However, Antti Kantee has [run the NetBSD network stack on Snabb](https://groups.google.com/forum/#!msg/snabb-devel/2yF5LZ-VS10/DxEifZXuIpkJ) as a proof of concept.

What do you want to do?

rohitjoshi, 2016-10-17:
planning to create [WAF](https://www.owasp.org/index.php/Web_Application_Firewall) using lua based [OWASP](https://www.owasp.org/)  rules.

lukego, 2016-10-18:
That would be neat. No, today we don't have a TCP stack. I would love to find a reasonable excuse to build one :) particularly a special-purpose one that would enable applications you can't implement with off-the-shelf stacks.

Anecdotally, the project I worked on immediately before Snabb was a specialized "man-in-the-middle" TCP stack for optimizing traffic over cellular networks. This acted basically like a proxy but kept TCP state (options, sequence numbers, etc) synchronized between the endpoints. Cool property was that you could pull the power cable and ~99.9% of the connections would continue without breakage. This is currently being marketed as [Sandvine TCP Accelerator](https://www.sandvine.com/products/tcp-accelerator/). There's a great [writeup by Juho Snellman](https://www.snellman.net/blog/archive/2015-08-25-tcp-optimization-in-mobile-networks/) about it. That was a really cool problem to work on :).

blakedotfr, 2017-10-06:
also a great candidate for a multipath TCP proxy...