snabbco/snabb #1050: snabb.co redirect to Github

lukego, 2016-10-21:
I have now transitioned the Snabb domain (http://snabb.co and http://www.snabb.co/) to redirect to our Github project page.  (Could take a little while to propagate.)

This is simply following through on the plan discussed over at https://github.com/snabbco/snabb/issues/738#issuecomment-180727157 i.e. for the Snabb name/trademark to refer to this open source project.

Previously the `snabb.co` website was my own professional  shopfront but I will recreate that later under a different brand.

I suppose the next step is to provide `@snabb.co` email aliases to any maintainers who would like one.

wingo, 2016-10-21:
Neat.  Thanks for being an amazing project technical lead and steward; you have made a really lovely virtual place to work :-)

lukego, 2016-10-21:
Thanks Andy :). 'Tis a pleasure to be hacking with you all!

eugeneia, 2016-11-08:
Closing because the issue is resolved.

closed 2016-11-08 by eugeneia