snabbco/snabb #1056: virtio driver not receiving packets

emmericp, 2016-10-31:
Hi,

I'm trying to use Snabb inside a VM using the virtio module.
I've had two students doing some work with Snabb and they have posted to the mailing list regarding this problem a few weeks ago: https://groups.google.com/forum/#!topic/snabb-devel/E_2EvwGkCoU

I recreated the test setup described the mailing list by them on my private server since our usual testbed sometimes creates weird configurations. But I also can't seem to get the virtio module working there.

I created a simple echo script to test it: https://gist.github.com/emmericp/1214910a09245306a89991865c407e45

The output looks like this:
```
$ src/snabb echo 0000:00:14.0 0000:00:13.0
host_features	2042626023
set features to:	402653216
got features: 	2042626023	402653216
enumerating queues...
queue 0: size: 256
queue 1: size: 256
host_features	2042626023
set features to:	402653216
got features: 	2042626023	402653216
enumerating queues...
queue 0: size: 256
queue 1: size: 256
link report:
                   0 sent on nic1.tx -> nic1.rx (loss rate: 0%)
                   0 sent on nic2.tx -> nic2.rx (loss rate: 0%)
```
which seems fine, except that it should receive packets since I'm sending about 100 packets per second to both links. The packets are visible with tcpdump on the VM tap interface while Snabb inside the VM sees nothing.

The packets show up fine when using the kernel driver. The virtio driver in DPDK also works in the VM.

I added a few debug prints to the driver, e.g.
```
    local idx1, idx2 = self.vring.used.idx, self.last_used_idx
+if debug then print("can_get", idx1, idx2) end
```
in virtq_driver.lua, this always prints 0. So it looks like no packets get written to the ring for some reason. I didn't dig deeper into the driver yet (but it's unfortunately unlikely that I find time to do so in the near future).

Relevant software versions and configurations:
* Snabb master (eb9d141431f3ede56867547e852ec543593b5099) and an old commit (first merge of virtio support)
* QEMU 2.6.0 pve-qemu-kvm_2.6-1
* Hypervisor Linux: 4.4.13-2-pve
* VM Linux: Jessie 3.16.0

The hypervisor is a proxmox machine (which uses weird versions for everything), but the same problem also exists on a Debian with a 3.16 kernel with both qemu 2.1.2 and 2.6.0.

The full command line used to start it is kind of messy as it's generated by proxmox:
```
/usr/bin/kvm -id 10000 -chardev socket,id=qmp,path=/var/run/qemu-server/10000.qmp,server,nowait -mon chardev=qmp,mode=control -pidfile /var/run/qemu-server/10000.pid -smbios type=1,uuid=e2416010-c0da-4e94-ba9e-29e0141afad2 -name test-network -smp 4,sockets=1,cores=4,maxcpus=4 -nodefaults -boot menu=on,strict=on,reboot-timeout=1000 -vga cirrus -vnc unix:/var/run/qemu-server/10000.vnc,x509,password -cpu host,+kvm_pv_unhalt,+kvm_pv_eoi -m 2048 -k en-us -device pci-bridge,id=pci.1,chassis_nr=1,bus=pci.0,addr=0x1e -device pci-bridge,id=pci.2,chassis_nr=2,bus=pci.0,addr=0x1f -device piix3-usb-uhci,id=uhci,bus=pci.0,addr=0x1.0x2 -device usb-tablet,id=tablet,bus=uhci.0,port=1 -device virtio-balloon-pci,id=balloon0,bus=pci.0,addr=0x3 -iscsi initiator-name=iqn.1993-08.org.debian:01:ff59331f831 -drive if=none,id=drive-ide2,media=cdrom,aio=threads -device ide-cd,bus=ide.1,unit=0,drive=drive-ide2,id=ide2,bootindex=200 -device virtio-scsi-pci,id=scsihw0,bus=pci.0,addr=0x5 -drive file=/var/lib/vz/images/10000/vm-10000-disk-1.qcow2,if=none,id=drive-scsi0,cache=writeback,discard=on,format=qcow2,aio=threads,detect-zeroes=unmap -device scsi-hd,bus=scsihw0.0,channel=0,scsi-id=0,lun=0,drive=drive-scsi0,id=scsi0,bootindex=100 -netdev type=tap,id=net0,ifname=tap10000i0,script=/var/lib/qemu-server/pve-bridge,downscript=/var/lib/qemu-server/pve-bridgedown,vhost=on -device virtio-net-pci,mac=EE:D8:3C:1F:48:17,netdev=net0,bus=pci.0,addr=0x12,id=net0,bootindex=300 -netdev type=tap,id=net1,ifname=tap10000i1,script=/var/lib/qemu-server/pve-bridge,downscript=/var/lib/qemu-server/pve-bridgedown,vhost=on -device virtio-net-pci,mac=2A:A1:99:DF:D4:37,netdev=net1,bus=pci.0,addr=0x13,id=net1,bootindex=301 -netdev type=tap,id=net2,ifname=tap10000i2,script=/var/lib/qemu-server/pve-bridge,downscript=/var/lib/qemu-server/pve-bridgedown,vhost=on -device virtio-net-pci,mac=FA:8E:98:13:3E:0F,netdev=net2,bus=pci.0,addr=0x14,id=net2,bootindex=302
```

The relevant part for one of the NICs is: ```-netdev type=tap,id=net1,ifname=tap10000i1,script=/var/lib/qemu-server/pve-bridge,downscript=/var/lib/qemu-server/pve-bridgedown,vhost=on -device virtio-net-pci,mac=2A:A1:99:DF:D4:37,netdev=net1,bus=pci.0,addr=0x13,id=net1,bootindex=301```

Adding ioeventfd=on makes no difference.
I tried to disable the vhost=on option, this crashes qemu with the error message `virtio-net receive queue contains no in buffers` when Snabb is started.

One of the NICs is attached via Open vSwitch, the other via a Linux bridge (just to test if that makes a difference, but it doesn't).

Any help with this is appreciated. I can provide access to a test VM for debugging.

 Paul

eugeneia, 2016-10-31:
@wingo @dpino Can I get you guys to take a peek at this?

wingo, 2016-10-31:
@dpino is on holiday for the next couple weeks I think.  @emmericp do I understand that you are using something other than Snabb NFV on the host to provide networking to the QEMU instance?

emmericp, 2016-10-31:
We use Open vSwitch at the moment for a simple test setup.

But we also tried it with Snabb NFV vhost user recently and got the same problem: it works fine with the kernel virtio driver in the VM but the Snabb virtio app doesn't see any packets in the VM.

wingo, 2016-11-01:
@emmericp strange stuff.  I'm sorry you're hitting these bugs.  We tested this in a limited environment with Snabb NFV on the host and running one of our data planes on the guest (the lwaftr).  Here is the script that we use to start things going: https://github.com/snabbco/snabb/blob/master/src/program/lwaftr/virt/lwaftrctl

Does that help at all? The networking bit looks like

```
        -netdev type=tap,id=net2,ifname=${IFACE},vhost=on,script=${NETSCRIPT} \
            -device virtio-net-pci,netdev=net2,addr=0xa,mac=52:54:00:00:00:03
```

But that would seem to be very similar to what you have.

emmericp, 2016-11-01:
Which version of qemu are you using?

wingo, 2016-11-01:
```
$ qemu-system-x86_64 --version
QEMU emulator version 2.4.0, Copyright (c) 2003-2008 Fabrice Bellard
```

wingo, 2016-11-01:
I think we did test with other qemu versions though.

dpino, 2016-11-04:
@emmericp Sorry for the late reply, I think I have some time now to take a look at this.

We have only tested the virtio-net driver together with SnabbNFV as network hypervisor. I would encourage you that, at least for testing purposes, to use SnabbNFV.

The line that my colleague @wingo pasted from https://github.com/snabbco/snabb/blob/master/src/program/lwaftr/virt/lwaftrctl sets up indeed a tap interface, but that interface is for ssh into the VM. The line that setups the VM with a virtio-net driver is: 

```
  -chardev socket,id=char1,path=${VHU_SOCK1},server \
            -netdev type=vhost-user,id=net0,chardev=char1 \
            -device virtio-net-pci,netdev=net0,addr=0x8,mac=52:54:00:00:00:01 \
```

The vhost-user driver allows QEMU to talk with an user-space program via a socket as IPC (since they're two different programs). On the other hand, Snabb features a VhostUser driver making the communication possible. I guess you got this part right, since it seems you have already tested SnabbNFV successfully.

Once you got into the VM, you should see an eth0 interface with address 0000:00:08.0. The vendor id should be virtio-net (0x1AF4).

```
# ip li sh eth0
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast \
   state UNKNOWN mode DEFAULT group default qlen 1000
   link/ether 52:54:00:00:00:01 brd ff:ff:ff:ff:ff:ff
```

```
# cat /sys/devices/pci0000:00/0000:00:08.0/vendor
0x1af4
```

In the snippet you pasted above, you're using addresses 0x13 and 0x14, but the QEMU setup was:

```
-netdev type=tap,id=net1,ifname=tap10000i1,script=/var/lib/qemu-server/pve-bridge,downscript=/var/lib/qemu-server/pve-bridgedown,vhost=on
-device virtio-net-pci,mac=2A:A1:99:DF:D4:37,netdev=net1,bus=pci.0,addr=0x13,id=net1,bootindex=301 

-netdev type=tap,id=net2,ifname=tap10000i2,script=/var/lib/qemu-server/pve-bridge,downscript=/var/lib/qemu-server/pve-bridgedown,vhost=on
-device virtio-net-pci,mac=FA:8E:98:13:3E:0F,netdev=net2,bus=pci.0,addr=0x14,id=net2,bootindex=302
```

I have never use virtio-net in combination with a tap network backend, only with vhost-user. This might be an issue.

If you want to try the vhost-user network backend, you would need to run SnabbNFV on the host so Snabb can work as an hypervisor of the card:

```
$ sudo ./snabb snabbnfv traffic <pci> <ports.cfg> <socket-path>
```

For instance:

```
$ sudo ./snabb snabbnfv traffic 0000:02:00.0 ports.cfg /tmp/vh1.sock
```

Where `ports.cfg` is a file like:

```
return {
   id = "A",
}
```

The command line running QEMU should point to the same socket path:

```
-chardev socket,id=char1,path=/tmp/vh1.sock,server \
   -netdev type=vhost-user,id=net0,chardev=char1 \
   -device virtio-net-pci,netdev=net0,addr=0x8,mac=52:54:00:00:00:01 \
```

Summarizing:

- Snabb can be used as a networking hypervisor in QEMU. That's done via the vhost-user network backend which is part of QEMU and allows to communicate QEMU with an user space program (Snabb).
- SnabbNFV is the program that allows to run Snabb as a hypervisor for QEMU/KVM.  On its most basic configuration `snabbnfv traffic` appends a VHostUser driver to a network card, making it possible to use the card virtualized from QEMU.
- Network interfaces inside the VM handled by VhostUser appear as virtio-net devices (well probably, that's due to -device virtio-net-pci,...).
- Running Snabb inside a VM using a VirtioNet driver allows Snabb to talk directly with a virtio-net device (0x1AF4), skipping the Linux kernel. 

If I have more time I'd try to run your code example. In the meantime, I hope this comment shed some light.

Lastly, I think that it might be possible the issue was using a QEMU version different than 2.4. I would recommend you to try QEMU 2.4 (and SnabbNFV as hypervisor, of course).

emmericp, 2016-11-04:
Hi,

thanks for the detailed response. Snabb NFV would probably be a work-around. However, using SnabNFV is not what we are looking for at the moment.

We will probably skip the virtio setup for now, it's not yet important for us.
However, it would be nice if this could be fixed. As I've mentioned above: I can provide access to a test VM with two VirtIO interfaces connected to a packet generator that work fine with the kernel driver and the DPDK driver but not with Snabb.

I understand that Open vSwitch will be a huge bottleneck for the small packet IO performance, but we are looking for something that plays well with different hypervisors and configurations for a NFV project for a 5G network backend.

 Paul

dpino, 2016-11-05:
@emmericp I got your example working (actually a simplified version of it that works with one NIC only). Here is a walk-through guide:

- Start SnabbNFV on a NIC:

```bash
$ sudo ./snabb snabbnfv traffic 0000:02:00.0 ports.cfg /tmp/vh1.sock
```

**ports.cfg**

return {
	{
		port_id = "A"
	}
}

My NIC is 0000:02:00.0 which is wired to 0000:82:00.0. Later I will send packets to 0000:82:00.0 with packetblaster and since both NICs are wired, packets will arrive to the VM

- Start QEMU:

**run-qemu.sh**

```lua
#!/usr/bin/env bash

IMAGE=./qemu-virtio.img

sudo qemu-system-x86_64 \
	-kernel ./bzImage -append "earlyprintk root=/dev/vda rw console=tty0" \
	-enable-kvm -drive file=${IMAGE},format=raw,if=virtio -m 1024M \
	-numa node,memdev=mem \
		-object memory-backend-file,id=mem,size=1024M,mem-path=/hugetlbfs,share=on \
	-chardev socket,id=char1,path=/tmp/vh1.sock,server \
		-netdev type=vhost-user,id=net1,chardev=char1 \
		-device virtio-net-pci,netdev=net1,addr=0x8,mac=52:54:00:00:00:01 \
	-serial stdio -curses
```

There are tons of ways of launching a QEMU image. In this case I'm using just one single NIC with vhost-user backend, handled by SnabbNFV. The VM (qemu-virtio.img) already includes snabb source code in its /root folder (other approach is to have a mount point to the host). The VM uses stdio -curses as login system, that means it logs in directly (no telnet, no ssh). The command assumes HugePages are enabled.

Here is a link to the bzImage and qemu-virtio.img I used.

- bzImage: https://people.igalia.com/dpino/virtio/bzImage
- qemu-virtio: https://people.igalia.com/dpino/virtio/qemu-virtio.img

When QEMU is launched it keeps waiting until there is a connection in `/tmp/vh1.sock`. After running SnabbNFV, if things go well a connection should be established and QEMU continues booting until login into the VM.

Once inside the VM, cd to snabb source code and create the echo.lua example (I think the script is already present in qemu-virtio.img). It's slightly different than the original version:

```lua
local VirtioNet = require("apps.virtio_net.virtio_net").VirtioNet

function run (parameters)
   if not (#parameters == 1) then
	  print("Usage: echo <nic1>")
	  main.exit(1)
   end
   local nic1 = parameters[1]
   local c = config.new()
   config.app(c, "nic1", VirtioNet, {pciaddr = nic1})
   config.link(c, "nic1.tx -> nic1.rx")
   engine.configure(c)
   engine.main({duration = 180, report = {showlinks = true}})
end

run(main.parameters)
```

Start running echo.lua:

```bash
$ sudo ./snabb snsh echo.lua 0000:00:08.0
```

I let the script run for 3 minutes to give enough time to setup and run packetblaster in the host.

In the host, use packetblaster to send packets to the VM:

```bash
$ sudo ./snabb packetblaster replay v4v6.pcap 0000:82:00.0
```

File v4v6.pcap is here: https://people.igalia.com/dpino/v4v6.pcap. Other pcap file is of course possible.

If things are going OK, the window running SnabbNFV should print out it's receiving packets now.

```
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
load: time: 1.00s  fps: 0         fpGbps: 0.000 fpb: 0   bpp: -    sleep: 100 us
```

(These numbers should be different than 0 if packets are received by SnabbNFV).

Once the echo.lua script is done it prints out the total number of packets that echoed:

```
link report:
		 117,508,905 sent on nic1.tx -> nic1.rx (loss rate: 0%)
```

And that's it. Please let us know if the same steps worked for you. I used QEMU2.4.

emmericp, 2016-11-09:
Thanks for the detailed explanation!
I've installed qemu-2.4.1 and it's working with SnabbNFV :)

Unfortunately, it still doesn't work with a "normal" virtio backend. But SnabbNFV is close enough for our project for now.