snabbco/snabb #1057: Draft snabb config process

wingo, 2016-10-31:
After the discussion in #987, we realized that we don't need a persistent named binary database in order to be able to handle configuration queries and updates and state queries -- instead we can just have another process running, managed by the same supervisor process (#930), with a back channel to the data plane.  In that way we keep the high performance but avoid version skew between config utils and the data plane, and we avoid schema migration concerns because the persistent configuration representation is just source.  Cool.

To get there though we need to implement that config process, define and implement its backchannel to the data plane, and define and implement a stable lightweight outward-facing protocol that it will speak with such tools as `snabb config query`.

The steps to get there would be:

- [ ] Make separate config process fork off at the beginning, and make sure it's collected when everything stops.

- [ ] Create a back-channel between the config process and the data plane.  I suggest modelling off of the "channel" facility in the lwaftr, though you could also try to do it with links and packets; apparently you can share packets between processes.  Should the data-plane-side thing be an app?  This I don't know.  @lukego might have some ideas here.

- [ ] Design how the back-channel could work with multiple worker processes.  We don't have to implement it now though.

- [ ] Define an initial binary protocol between the config process and the data plane.  For big objects (more than a packet?) we can write those out to tmpfs as separate files.  The initial "language" spoken should just be enough to tell the data plane to reload its configuration.  Queries as always come directly from tmpfs; no need to ask the data plane.  The other operation that we'd want to do would be incremental updates of various kinds, but this can be defined later.  Implement some of this language.

- [ ] Define an external protocol and transport between the config process and clients.  I suggest UNIX domain sockets in the /run/snabb dir and a JSON protocol, though you will need to define what the JSON means.  See the design in #987 for inspiration; perhaps you need transaction counters or something.  The goal here should be to be light-weight and future-proof, which means generic, precise syntax and high-level semantics.

- [ ] What happens when multiple clients are connecting to the process?

- [ ] Implement a "snabb config" util that talks to the config process to give a running data plane an entirely new configuration.  The mid-term goal would be to compile configurations to binary files (Igalia#517) and tell the data plane to reload the conf and keep trucking, but this point would depend on that other bug.

lukego, 2016-11-01:
Great topic and very thought provoking! Ideas -

### Processes

I would like to clarify what the Snabb process tree will look like. Just to make sure that we all mean the same thing when we use terms like worker, supervisor, config process, etc.

Let me make a proposal that does not necessarily represent reality on any specific branch:

- The _Main_ process is the initial one. This process calls the "program" entry point to bring up the application (including starting all related processes.) The Main process is non-realtime: it can run the engine but it is subject to the whims of the kernel scheduler. (Could run an app that answers ARP requests for example.)
- The _Worker_ processes each run the engine on a dedicated CPU core. These processes are started by the "program" logic in the Main process. They receive their configuration via some IPC mechanism (ideally based on a YANG model.)
- The _Supervisor_ is a tiny special-purpose process that runs some cleanup hooks on shutdown: removing stale shm/IPC files, disabling DMA on PCI devices, freeing HugeTLB memory, etc. The number of Supervisor processes is an implementation detail e.g. just one for the Main process or one each for the Workers etc.

In this model I imagine that the _Main_ process could provide all general administrative interfaces e.g. serving `snabb config`, providing debugging interfaces like a REPL, and so on. So we would not need a separate _Config_ process? (Could be that a long-running configuration operation would require spinning off some processing into an asynchronous child process though.)

### Worker config update protocol

I have an opening proposal to shoot down here too :-) How about this:

Over on #1021 I wrote a [mechanism for a worker process to poll for configuration updates](https://github.com/lukego/snabb/blob/1e1f889bc0f1ecc16f5383074a8b13573166b495/src/core/worker.lua#L79-L117). The idea is that polling is cheap (check if a "number of configurations" shm counter has been bumped) and then the actual update is simple (can use the filesystem - syscalls to read from ramdisk should be affordable to process updates.)

### Back-channel

Can you say a bit more about what the requirements are on the back-channel? (what is it for?)

wingo, 2016-11-01:
/cc @kbara @xray7224

lukego, 2016-11-01:
> perhaps there is a ncie simple design lurking somewhere

Just make facilities like config server, repl, etc be apps that run in the engine of the Main process?

> We are shooting for thousands of these incremental updates/s and we need a channel for messages

I see. Likely you need to use a special-purpose mechanism that makes sense to you, and the next person who has a similar requirement can propose a way to generalize it?

wingo, 2016-11-01:
Interesting idea for config server etc to be apps, will look into that.  As far as the backchannel goes, we could have the config app in the main process be responsible for sending messages to the workers, who run their own app I guess, or perhaps the backchannel is a part of the lwaftr app (as it is now, actually).  OK!