snabbco/snabb #1063: The canonical representation of an IPv4 address in Snabb should be host-endian uint32

wingo, 2016-11-07:
Right now the blessed way to represent an IPv4 address in Snabb is the representation returned by the `pton` method of `lib.protocol.ipv4`, which is as a `uint8_t[4]`.  I think this is not the right thing and instead we should represent these addresses as host-endian uint32's (normal Lua numbers).

### Creating a new IPv4 address should never allocate

Because you can represent an IPv4 address as a number without allocation, that's what we should do.  I know that allocation sinking and scalar replacement can elide the Lua FFI box but we shouldn't rely on that.

### It's useful to think of an IPv4 as a number

You can mask off a /N prefix very easily with the bitops module.  You can use binary search over a sorted vector of addresses, possibly with associated values; the range map does this, the lwaftr binding table does this, and DXR does this.  It's nice.

### Host endian is the right endian

Given a packet, you can pull out the bytes as a network-endian address with a simple memory fetch.  However then there's always the question as to what endianness a given address has, given that the scalar advantages from the previous point require host-endian values.  Better to avoid that bug farm and always work with host-endian values; the `htonl` is pretty much free anyway.

### People are going to do this anyway

Say you have a lookup structure keyed by a sorted uint32_t vector of ipv4 addresses.  Well then you're going to be doing `local addr = ffi.cast('uint32_t*', ipv4)[0]` anyway -- and what a gnarly incantation.  Also it's wrong -- you forgot the `lib.ntohl`!

### Consistency with ethernet and ipv6 addresses is less important

One reason things are like they are is that ethernet and ipv6 addresses are represented as byte arrays.  However I think this is less important than the efficiency considerations above.

### What does it mean?

`ipv4:pton` should return a uint32, and we should eventually adapt all code -- if you agree with me of course :)  For now I will just make the representation of IPv4 addresses in yang configurations be a uint32.

plajjan, 2016-11-11:
I agree with all of this except for that last part. At least for YANG models exposed externally, the types defined in http://www.netconfcentral.org/modules/ietf-inet-types should be used - which is dotted quad for IPv4. You need to parse this into uint32.

If you have some internal data structure that is described by a YANG model then I suppose it might be fine leaving it as a uint32.

wingo, 2016-11-11:
This point was about internal representation of IPv4 values.  In yang modules they are of course inet:ipv4-address; it's just a question of how that data type is represented internally.

plajjan, 2016-11-11:
@wingo Ok, good. (wasn't very clear since there's a uint32 in YANG too ;)

wingo, 2016-11-11:
Yeah we are trying to represent internal data in types derived from yang schemas.  It mostly works but you have to add some special cases for certain types; e.g. in yang there is no primitive type for IPv4 addresses as you know, and so the ipv4 typedef is actually as a string (!) whose syntax is restricted by a regular expression (!!!).  So we have overrides internally to efficiently represent IPv4 and IPv6 addresses and CIDR prefixes and ethernet addresses.  Before, we mapped IPv4 to uint8_t[4].  Now we map to uint32_t.

daurnimator, 2016-11-14:
IMO this is a bad choice: I'd prefer to see IPv6 and IPv4 be usable with the same style of code.