snabbco/snabb #1069: Incremental configuration app design

wingo, 2016-11-11:
Sometimes you want to incrementally update a Snabb dataplane while it's running (https://github.com/Igalia/snabb/issues/562).  To implement this, we are planning to use the multiprocess work from Luke (#1021) and run an app in the main process.  This app will communicate with corresponding apps running in the worker processes.  This bug is a design document for the implementation of those apps.  Implementation work can proceed in parallel from #1021, as you can always run both apps in the same process as the data plane instead.

## Terminology

Let us adopt the process terminology from #1021: there is a main process and there is a worker process.  The worker is bound to a CPU core and NUMA node; the main process runs free (and is probably scheduled on the first core of the numa node).  The worker has soft real-time constraints; the main process does not.  Sometimes we call the main process the "manager" process.

The app that runs in the main process is called the "ConfigLeader".  (Other names are welcome.)  Sometimes we'll call this the leader.

The corresponding app that runs in the worker process is the "ConfigFollower".  (Again other names are welcome.)  Likewise sometimes we call this the follower.

## Two protocols

The leader communicates with its followers using a private protocol.  Because the leader and the follower are from the same Snabb version, the details of this protocol are subject to change.  The private protocol's only design constraint is that it should cause the lowest overhead for the data plane.

The leader communicates with the world via a public protocol.  The "snabb config" command-line tool speaks this protocol.  "snabb config get foo /bar" will find the local Snabb instance named "foo", open the UNIX socket that the "foo" instance is listening on, issue a request, then read the response, then close the socket.

## Public protocol

The design constraint on the public protocol is that it be expressive and future-proof.  We also want to enable the leader to talk to more than one "snabb config" at a time.  In particular someone should be able to have a long-lived "snabb config listen" session open, and that shouldn't impede someone else from doing a "snabb config get" to read state.

To this end the public protocol container is very simple:

```
Message = Length Payload
```
Length is a base-10 string of characters indicating the length of the message.  There may be a maximum length restriction.  This requires that "snabb config" build up the whole message as a string and measure its length, but that's OK.  Knowing the length ahead of time allows "snabb config" to use nonblocking operations to slurp up the whole message as a string.  A partial read can be resumed later.  The message can then be parsed without fear of blocking the main process.

The format of Payload is like this:
```
Payload = Request | Response
Request = module-name revision-date? "\n" snabb-yang-rpc
Response = snabb-yang-rpc
```

`snabb-yang-rpc` is the textual data syntax described in https://github.com/Igalia/snabb/blob/yang/src/lib/yang/README.md.  module-name names the YANG module describing the RPCs accepted by the leader.  RPCs supported by the leader will include `get-state`, `get-config`, and similar operations needed to implement `snabb config`.  In this way we can re-use the YANG parser and serializer on both sides, to validate incoming and outgoing data.

## Private protocol

The leader maintains a configuration for the program as a whole.  As it gets requests, it computes the set of changes to worker app graphs that would be needed to apply that configuration.  This process is described in https://github.com/Igalia/snabb/issues/562.  These changes will then be passed through the private protocol to the worker.  No response from the worker is necessary.

In some remote or perhaps not so remote future, all Snabb apps will have associated YANG schemas describing their individual configurations.  In this happy future, the generic way to ship configurations from the leader to a follower is by the binary serialization of YANG data, implemented already in the YANG modules.  Until then however, there is also generic Lua data without a schema.  The private protocol will have to support both kinds of information transfer.

To implement the private protocol, followers will have a little ring buffer from which they read and to which the leader writes.  In this way the leader can be sure to never write more than the follower can handle, and it can know when the follower has applied the information.  Any "get" request will stall the write pipeline, causing the leader to wait for all followers to have applied their changes.

However a message may in general be unbounded in size, whereas a ring buffer is naturally fixed.  Therefore messages which may include variable-sized data may be forced to put that data in the filesystem, and refer to it from the messages in the ring buffer.

## Implementation plan

First we will implement the leader, and incorporate the follower functionality directly there.  (Besides the protocol decode, the follower's functionality is just "engine.apply_configuration_actions".  Then we will implement the private protocol, and test it out in one process.  Then when we manage to incorporate multiple processes, we will put the leader in the main process.  Later when there are multiple workers, we can incorporate that functionality in the leader.

lukego, 2016-11-11:
Sounds sane!

One possibility that I think is worth considering is for `snabb config` to use a private protocol too. This would require `snabb config` to run exactly the same software version as the main process but that is a problem that we already need to solve internally for starting workers anyway.

The upside would be that we can change the interface in the future with impunity. The downside would be that all tooling would need to make requests via `snabb config` rather than implementing the protocol independently.

The mechanism could involve `snabb config` identifying the executable that the main process is running and using exec() to force itself to adopt that same software. Here is pseudo-code for what `snabb config` might do at startup:

```shell
if [ $0 != /var/run/snabb/$mainpid/snabb ]; then
   # Restart this process with the matching software version.
   exec /var/run/snabb/$mainpid/snabb "$@"
fi
```

This would require the main process to have copied its own executable into its shm folder. Seems like we will need to do something equivalent to that in any case for worker spawning though. I am asking stack overflow for a better solution. (I have looked into reconstructing the executable by introspecting our own memory map from `/proc/$pid/maps` but I suspect this is a bit complicated and ELFey with multiple segments and stuff.)

wingo, 2016-11-11:
FWIW I am currently sketching ideas out here: https://github.com/Igalia/snabb/tree/config-leader.  Very early state right now, it only does ping/pong :P

wingo, 2016-11-14:
The question about whether we can eliminate the public protocol is an interesting one.  If we have the binary corresponding to a running snabb instance, that could indeed solve the problem for us.  However on a practical level it's just labelling -- the protocols are different (one is high-level RPC and the other is just an apply-these-actions protocol) so whether the "public" one is really public or not can be decided later.

eugeneia, 2016-11-16:
@lukego I ignored this previously, but I have to ask: why should Leader and Follower be apps instead of programs? That seems to break every control/data plane boundary we have, and seems very complicated for now apparent (to me) reason. Is an instance supposed to be/run multiple Followers?

wingo, 2016-11-16:
How could a follower be a program?  It has to be in the data plane in order to mutate the app graph.  Apps are how we compose different bits of code into a snabb process.  It could just as well be a timer of course, if that's the thing.

eugeneia, 2016-11-16:
I should stop using data/control plane as terms, I think these are overloaded.

Never before did an app mutate the app graph. There is a “management” API (core.engine) that changes the app graph and invokes the “data engine”. So I am thinking the worker that consumes configuration changes (follower) from the main process (leader). Any follower will have exactly one leader? Then the follower process should run the follower program that manages an app graph based on its input from the leader program.

wingo, 2016-11-16:
You mean `core.app`, not `core.engine` I think?

What do you mean by "program"?  Do you mean process, Snabb instance consisting of multiple processes, or general subroutine-like thing?  Best to adopt Luke's process terminology in #1021 I think if we are talking about processes.

For me a follower is a component which should run in a worker process, and a leader is a component that should run in a main process.  I have implemented both as apps, but that's a detail.

lukego, 2016-11-16:
@eugeneia Initially I had imagined a one-size-fits-all `snabb worker <name> <parent-pid>` program that simply executes an app network provided by its parent. Nice and simple and would work fine for snabbnfv. However, @wingo indicated that this is too simplistic for the lwAFTR use case where they need to rapidly process updates to large configurations.

So I modified the `core.worker` module such that the worker executes arbitrary Lua code and so the program spawning the worker can decide exactly what it should do. I have not looked closely at how the config leader/follower apps work yet and I would not be surprised if they are overkill for other programs like snabbnfv with small configurations that seldom change.

eugeneia, 2016-11-17:
@wingo Yes core.app (it is exposed as the global `engine` :-)). By program I mean the top level code executed by the worker, e.g. whatever the Lua expression passed to it runs.

I imagined the leader to spawn and set up communication with N follower workers (running the follower “program” at their top level”), which do something conceptually similar to `snabbnfv`, along the lines of:

```
while true do
   local changes = poll_config_changes(leader_channel)
   if incremental_actions_possible(changes) then
      engine.configure(next_configuration(changes))
   end
   -- The engine doesn’t have to run for a duration, could also be a done callback
   engine.main({duration=1})
end
```

wingo, 2016-11-17:
The loop you give is pretty much what the follower does.  It polls for actions and applies those actions.  It could run via a timer instead.  It happens to run as an app because that's what Luke suggested.

wingo, 2016-11-17:
The follower does have to listen for changes at a high rate though.  We are aiming for a minimum 1000 changes per second without dropping packets, for binding table updates at least.  Only processing updates every second would probably stall the data plane because the batch of updates would be too large.

wingo, 2016-11-17:
I also want to avoid O(N) operations in the workers.  I especially don't want to have to compare binding tables.  For that reason I think there is an advantage for pushing "what changes to apply" logic out of the worker process.

eugeneia, 2016-11-17:
I would recommend to avoid implementing leader/follower as apps, if possible at all. While @lukego’s suggestion is cute, this basically means that the follower app is issuing calls to `engine.configure`, possibly re-configuring (or, you know, stopping) itself, seems overly exiting to me. (On the other hand this gives rise to hot loading new versions as Lua files... :-O)

I don’t see an immediate difference between running the code inside or outside of the breathe loop, unless the reconfiguration needs to happen in between a single breath. The polling could happen in the `done` callback of engine.main, and schedule incremental updates on demand.

> I also want to avoid O(N) operations in the workers. I especially don't want to have to compare binding tables. For that reason I think there is an advantage for pushing "what changes to apply" logic out of the worker process.

No argument here, but I don’t think leader/follower need to be implemented as apps to accomplish that.

wingo, 2016-11-17:
It's not calling `engine.configure`, but rather `engine.apply_config_actions`; see https://github.com/Igalia/snabb/tree/yang/src/apps/config for details.

But yes.  Thing is, for me running something from a timer vs running it from an app is not so dissimilar.  Both are effectively concurrent processes that can be composed into an engine.  How are they different for you?  Do your arguments against apps apply to timers too?

eugeneia, 2016-11-17:
My concern is with apps handling reconfiguration of apps specifically, this recursive behavior seems overly complicated to me. Having the top level code implement follower behavior seems much more intuitive, and simple to me.

E.g., when `follower:pull` calls `apply_config_actions`, how will that affect the rest of the breath? IIRC the breathe order will be changed in the middle of a breath? This wouldn’t be a concern if follower was not implemented as an app and treated `engine` as a black box.

wingo, 2016-11-17:
`Follower:pull()` can cause `breathe_pull_order` and `breathe_push_order`'s bindings to change but it doesn't mutate those tables.  It could cause the breath to start with one pull order but then run a new push order (because it's a pull function and pull functions currently run first).  I see what you're saying though but I'm not going to hit it in my case.

Basically I am looking for another word from Luke here since he suggested implementing as an app, and that's what I'm going to ship tomorrow ;)  Can refactor into something else over time if that's the right thing.

lukego, 2016-11-17:
I don't know what the best solution is but braindump upon request:

Having a completely general follower program `snabb follower <name> <leader>` that derives all of its behavior from leader-supplied configuration (am I an lwaftr? am I an l2vpn? etc) seems optimal to me. Then application developers could reuse that and not bother about multiprocess plumbing. However, I tried this with the initial `snabb worker` API and missed the mark so I can believe that this level of generalization could be premature (or indeed the wrong idea after all.) I would be thrilled if this design turned out to be practical using the leader/follower routines though!

So for now the multiprocess model is that the main and worker processes will run application-specific code and be responsible for their own configuration updates.

This brings us to the leader and follower routines for exporting a config from one process to others. These routines have to be called somehow and it probably does not matter exactly how. I suggested an app only because apps seem like the most well-defined units of composition that we have in Snabb. Could also run it from the code that invokes the engine (via `done` callback) or via timers and to me each approach is more-or-less equivalent but with minor pluses and minuses.

I have not actually looked at the implementation closely yet and I do appreciate the potential for problems for when updating the engine config while the engine is running. This is a minus to using apps that I had not anticipated.

(I do somewhat regret introducing _timers_ in the first place. I see them as basically awkward and redundant with respect to apps. I initially wrote that module due to lack of imagination because every system I had worked on before always used a timer wheel somewhere and it seemed like we should have one too.)