snabbco/snabb #1081: program.snabbnfv.selftest.sh fails in local testing

kbara, 2016-11-28:
Running sudo make test in src/ on a dev machine with real NICs leads to the following error:

```
src]$ cat testlog/program.snabbnfv.selftest.sh 
Defaulting to SNABB_TELNET0=5000
Defaulting to SNABB_TELNET1=5001
Defaulting to SNABB_IPERF_BENCH_CONF=program/snabbnfv/test_fixtures/nfvconfig/test_functions/same_vlan.ports
USING program/snabbnfv/test_fixtures/nfvconfig/test_functions/other_vlan.ports
Defaulting to MAC=52:54:00:00:00:
Defaulting to IP=fe80::5054:ff:fe00:
Defaulting to GUEST_MEM=512
Defaulting to HUGETLBFS=/hugetlbfs
Defaulting to QUEUES=1
Defaulting to QEMU=/nix/var/nix/profiles/default/bin/qemu-system-x86_64
Waiting for VM listening on telnet port 5000 to get ready... [TIMEOUT]
EXITCODE: 1
```

This makes local testing when snabb bot is the wrong tool for the job (for instance, changes that don't target a branch that it covers) unnecessarily painful.

eugeneia, 2016-11-28:
The snabbnfv selftest should probably bail out if it can’t find the QEMU images it depends on, which I assue is the case here?

kbara, 2016-11-30:
Plausibly.

eugeneia, 2016-12-02:
@kbara I can’t really reproduce this. It seems that test_env really finds an image in your case, but the image won’t come up. This is how it looks for me:

```
-bash-4.3$ env | grep SNABB
SNABB_PCI0=0000:01:00.0
SNABB_TEST_FIXTURES=/var/lib/snabb-test-fixtures/
SNABB_PCI_INTEL0=0000:01:00.0
SNABB_PCI_INTEL1=0000:01:00.1
SNABB_KERNEL_PARAMS=init=/nix/var/nix/profiles/system/init
-bash-4.3$ sudo SNABB_PCI0=0000:01:00.0 program/snabbnfv/selftest.sh 
Defaulting to SNABB_TELNET0=5000
Defaulting to SNABB_TELNET1=5001
Defaulting to SNABB_IPERF_BENCH_CONF=program/snabbnfv/test_fixtures/nfvconfig/test_functions/same_vlan.ports
USING program/snabbnfv/test_fixtures/nfvconfig/test_functions/other_vlan.ports
Defaulting to MAC=52:54:00:00:00:
Defaulting to IP=fe80::5054:ff:fe00:
Defaulting to GUEST_MEM=512
Defaulting to HUGETLBFS=/hugetlbfs
Defaulting to QUEUES=1
Defaulting to QEMU=/run/current-system/sw/bin/qemu-system-x86_64
Couldn't find QEMU image: /root/.test_env/qemu0.img
-bash-4.3$ echo $?
43
```

Maybe you have stale test assets in `~/.test_env`, or `SNABB_TEST_FIXTURES`?

kbara, 2016-12-29:
I don't have ~/.test_env or SNABB_TEST_FIXTURES at all, and haven't ever had the testing VMs on my dev laptop.