snabbco/snabb #1085: Relatively high latency with SnabbNFV

emmericp, 2016-12-06:
I'm working on a NFV (research) project in the 5G area and we are looking at the overheads of different virtualization methods and strategies. I've measured the latencies of different virtual switches and found that SnabbNFV performs particularly poor compared to other solutions.

My setup is as follows: SnabbNFV connects a VM to an Intel 82599 port, the VM runs Snabb with a simple echo script that sends back the packets on the same port (the echo script can be found in #1056). I then apply an increasing load on the port and measure the latency.

The result is a median latency of around ~100µs at low rates that slowly increases to ~200µs at ~90% of the max load (i.e. before packet loss).
I expected a latency of around 10µs to 20µs at low rates here.
Running my echo script directly on the 82599 results in a latency in the expected range.

The histogram of the latency does not reveal anything interesting:
[image]
The shape looks like this at all packet rates until packets are dropped.

Other virtual switches perform better by an order of magnitude at low packet rates: plain Open vSwitch in the kernel starts with a latency of ~15µs and increases to ~100µs at higher loads before packets are lost.
Open vSwitch with DPDK and vhost user performs particularly well, it stays consistently below 20µs or so (and it achieves about the same packet rates as SnabbNFV).
In these two cases I've echoed the packets with Open vSwitch in the kernel in VM since the Snabb VirtIO driver doesn't work (#1056).

Can anyone take a guess at what is happening here?
This is not an actual issue for us, I'm well aware that 100µs is an acceptable latency when VMs are involved. I'm just curious why Snabb behaves like it does.

My best guess would that this is a result of trying to avoid a busy wait loop.
However, if this was the case, then the latency should decrease under higher loads.  However, it never drops below 100µs. (Decreasing latency under increasing load is something that can, for example, be observed in the Linux kernel with some drivers and traffic patterns)

lukego, 2016-12-07:
> Can anyone take a guess at what is happening here?

First let's try a bit of "napkin math" to see what the expected latency would be.

Snabb programs typically have ~10 Mpps of processing capacity i.e. spend around 0.1us per packet. One event loop iteration ("breath") typically processes up to 100 packets from each I/O source. So under heavy load a snabbnfv could process 100 packets from the NIC and 100 packets from the VM and then the expected processing time would be ~20us.

If the packets were processed three times (snabbnfv -> vm running snabb -> snabbnfv) then that could triple the latency to ~60us. If the NIC were congested on ingress and/or egress that could also increase the "end to end" latency if there is a delay between packets being processed and hitting the wire. On the other hand if the traffic load is low then each processing step should take less time.

So - if we would estimate that latency should not exceed 60us even under moderate load then your observations seem suspiciously high and worth investigating.

> My best guess would that this is a result of trying to avoid a busy wait loop.

This is an excellent hypothesis. Let's test it with more certainty to begin with.

Snabb does have a heuristic for detecting idleness and asking the kernel to sleep. This can reduce CPU utilization but at the expense of less steady latency. Since you are interested in latency you should disable this and run Snabb in a busy-loop instead.

This can be done on snabbnfv with the [`-b` command line option](https://github.com/snabbco/snabb/blob/master/src/program/snabbnfv/traffic/README#L13-L14). This can be done in your test script inside the VM with the setting `engine.busywait = true`.

Could you try that and see what the impact is?

Thinking ahead, one thing that we could try would be to merge in the "timeline" log support to your test branches. This would produce detailed samples of the event loop including cycle-precise timestamps. Just now some DIY is required to interpret these e.g. make sense of them as CSV files so it is only worth doing if you are up for that. (User-friendly tools are coming down the pipeline, see #1011).

lukego, 2016-12-07:
Side-issue: For me it is invaluable to be able to restart the vswitch without having to restart the VM while experimenting. This "just works" with Snabb if you apply a [small patch to QEMU](https://github.com/SnabbCo/qemu/commit/f393aea2301734647fdf470724433f44702e3fb9.patch). Just mentioning in case you are restarting your VMs every time and finding that frustrating. (Sadly upstream QEMU did not seem interested in this feature the last times I checked but maybe somebody has solve that more recently, if so maybe you can tell me :))

emmericp, 2016-12-08:
Okay, so I ran a few very basic tests with all 4 combinations of busy/non-busy.

**Throughput**

SnabbNFV/Snabb in VM
(B = busy, N = default/non-busy)
B/B: 4.09 Mpps
B/N: 3.23 Mpps
N/B: 4.04 Mpps
N/N: 2.23 Mpps

This is already quite surprising, I didn't expect such a big effect here. Enabling busywait inside the VM seems to be a good idea.

**Latency**
Measured at the three (arbitrarly chosen) packet rates: 0.2 Mpps, 1.4 Mpps, 2.0 Mpps
SnabbNFV/Snabb in VM
(B = busy, N = default/non-busy)
B/B: 22/25/45 µs
B/N: 56/85/200 µs
N/B: 16/25/44 µs
N/N: 95/90/95 µs

(Take these measurement results with a grain of salt: they are only based on a few thousand samples and these values are averages and I had active SSH sessions on the device under test. So the difference between 16µs und 22µs is not significant)

--> The high latency is not SnabbNFV's fault, but an effect of the VirtIO driver in the VM.
Running the same script without VMs with the Intel82599 driver results in a consistently low latency of 10-20µs.

lukego, 2016-12-08:
This sounds more reasonable.

Seems like our "idleness detector" is lousy. First guess is that the VM is seeing false-positives on the idle check because (a) VM event loop is running at a higher frequency than snabbnfv (doing less work) and (b) snabbnfv is making packets available in batches of ~100 and so sometimes the VM will see the ingress queue as empty because the batch is not complete. (In this case it could be that our vhost-user app should poke the virtio ring buffer "packets available index" more frequently to reduce latency and burstiness from the VM perspective.)

If you want another snabbnfv tunable to experiment with you could play with `engine.pull_npackets` which determines the RX burst size from the NIC. The default is ~100 but this may be a little large. If you play with (say) 80 +/- 20 you may see an impact on latency and throughput. I would like to make this auto-tuning but don't know how to formulate the right heuristic yet.

There is no formal command line argument for tuning this but you could use `snsh` to override it during startup. Instead of `snabb snabbnfv traffic ...` you could write e.g.

```
snabb snsh -e 'engine.pull_npackets = 80' \
      -p snabbnfv traffic ...
```