snabbco/snabb #1091: Forward selected traffic to second NIC<Getting Error: Message too long>

mmanoj, 2017-01-05:
Dear All,

I'm new to snabb project and currently studying the possible QoS implementation for selected traffic.I would like to contribute to the project and share my work with community.

In my  POC project I'm doing selectively forward traffic to second NIC. However I'm getting below error while doing it. I'm currently using rawsocket. I use snabwall for this project.However snabbwall firewall section still under development. So I have to implement my own method to achieve the requirement.

Seeking advice to resolve below error.

core/main.lua:26: Message too long
stack traceback:
    core/main.lua:137: in function <core/main.lua:135>
    [C]: in function 'error'
    core/main.lua:26: in function 'assert'
    apps/socket/raw.lua:114: in function 'transmit'
    apps/socket/raw.lua:90: in function 'method'
    core/app.lua:87: in function 'with_restart'
    core/app.lua:335: in function 'thunk'
    core/histogram.lua:98: in function 'breathe'
    core/app.lua:273: in function 'main'
    program/wall/spy/spy.lua:343: in function 'run'
    program/wall/wall.lua:19: in function 'run'
    core/main.lua:56: in function <core/main.lua:43>
    [C]: in function 'xpcall'
    core/main.lua:179: in main chunk
    [C]: at 0x00452230
    [C]: in function 'pcall'
    core/startup.lua:3: in main chunk
    [C]: in function 'require'
    [string "require "core.startup""]:1: in main chunk

========================================================
Following is my wiring

      config.app(c, "nic2", raw.RawSocket, "veth0")
       config.link(c,last_app_name..".north -> qosfw.input")
       last_app_name = "qos"
       -- config.link(c, last_app_name..".south -> qosfw.input")
        config.link(c, "qosfw.output -> nic2.rx")

================================================
Note: As per my study regarding the error and found out it's related to MTU, however I have to manage with MTU of 1500  of my NIC.

Seeking your advice to how to resolve this issue and move forward.

mmanoj, 2017-01-10:
Dear All,

Any update/ advice to resolve this issue. Your advice is highly appreciated.

alexandergall, 2017-01-11:
As you've already diagnosed yourself, you're trying to send a packet on the raw socket that exceeds the MTU of the underlying device. I don't know where that packet is coming from but this looks like an invalid setup, i.e. you're effectively connecting two layer-2 domains with different MTUs. The only proper solution is to make sure that the MTU matches on all interfaces that are part of your app network.

mmanoj, 2017-01-11:
Hi Alex,

Thanks for the advice, I'm using one physical and one virtual interface. Let me try two physical interfaces and the behavior. My app network is simple bridge eth01 get the input and send to eth02. In between I will do some traffic selection/DPI/QoS.

I will keep you updated the progress. Thanks again for the guidelines.

mmanoj, 2017-01-16:
Hi Alex,

Following is my network interfaces:

RJ45:
enp0s31f6 Link encap:Ethernet  HWaddr 50:7b:9d:68:74:7e  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:16 Memory:e1300000-e1320000 

Wireless:
wlp1s0    Link encap:Ethernet  HWaddr e0:94:67:47:63:f5  
          inet addr:192.168.1.102  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::e01:6c63:6d93:f3b2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:62335 errors:0 dropped:0 overruns:0 frame:0
          TX packets:27556 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:77358308 (77.3 MB)  TX bytes:4274770 (4.2 MB)

Both having same MTU of 1500, however still I'm getting the same error. Please advice if any way forward.
-------------------------------------------------

 local c = config.new()
  -- config.app(c, "capture", pcap.PcapReader, input)
   config.app(c, "spray_app", sprayer.Sprayer)
  -- config.app(c, "output_file", pcap.PcapWriter, output)
   config.app(c, "nic1", raw.RawSocket, "wlp1s0",input)
   config.app(c, "nic2", raw.RawSocket, "enp0s31f6",output)

  -- config.link(c, "capture.output -> spray_app.input")
  -- config.link(c, "spray_app.output -> output_file.input")
  
   config.link(c, "nic1.tx -> spray_app.input")
   config.link(c, "spray_app.output -> nic2.rx")

   engine.configure(c)
   engine.main({duration=10, report = {showlinks=true}})
end
-----------------------------------------------------------------------------------------------------------
Thank you,
Manoj M

alexandergall, 2017-01-17:
OK, I actually remember this gotcha now :) The problem is the "TCP segmentation offloading" feature on the ingress interface, which is usually enabled by default and causes the NIC to coalesce multiple TCP segments into a single large segment, which then exceeds the MTU. Try this to fix it:
```
ethtool -K wlp1s0 tso off
ethtool -K enp0s31f6 tso off
```
That should be applied to any interface which is used by Snabb via raw sockets (and should probably be documented).

mmanoj, 2017-01-17:
@alexandergall 

I'm able to change only on 100Mbps intel card only, wireless not allow o change. So I use one physical and one virtual interface (with tso off) but still same error. Any clue ?

mmanoj, 2017-01-17:
@alexandergall 

If any alternative to achieve this task also welcome.

alexandergall, 2017-01-17:
I'm sure that offloading is the issue here (you can confirm that easily by capturing packets with tcpdump). You may have to disable the generic offloading features as well (`gro off`, `gso off`).

mmanoj, 2017-01-18:
@alexandergall 

Thanks for the advice, now no error message ;) but no traffic receiving to NIC2 while capturing traffic in second NIC.