snabbco/snabb #1092: Read duplicated packets from raw sockets

nzinfo, 2017-01-18:
I'm trying to add a reliable UDP transport to Snabb, but I find the Snabb will bump the ethernet packet twice when using raw socket.

My code as the following
[zweig.lua.txt](https://github.com/snabbco/snabb/files/713104/zweig.lua.txt)

Python script to generation udp packet
[t_udp_client.py.txt](https://github.com/snabbco/snabb/files/713107/t_udp_client.py.txt)

When we open the pcap file using wireshark, we will found that packet appeared twice; 
At the same machine, I inspect the network adapter using wireshark, everything goes fine.

Great thanks.

lukego, 2017-01-25:
I reckon that a nice first step would be to isolate whether the duplication happens internally in Snabb or due to some interaction with the operating system. Raw sockets are reasonably funky.

Could you try reproducing the problem while running Snabb like this:

```
strace -f -e open,close,read,write,ioctl,bind snabb ...
```

and making a [gist](http://gist.github.com) of the output?

nzinfo, 2017-02-06:
Sorry for reply later, these days were the Spring Festival in China. : )
the gist is the output, I can't figure out why the bug happen.

https://gist.github.com/nzinfo/22aff96a21343168a6ddbfbbecf68fc6

nzinfo, 2017-02-07:
I found my mistake, I use local lookup adapter ( lo ), packet passed the device twice, one for sending, and one for receiving.  But another question raised, how to identity whether a packet is send / receive ? Great Thanks.

nzinfo, 2017-02-07:
I found there has some flag of sll_pkttype , PACKET_HOST | PACKET_BROADCAST | PACKET_OTHERHOST | PACKET_OUTGOING,  but I have no idea of how to set it. Would you please give me some help ? @lukego

lukego, 2017-02-09:
I am really busy at the moment, sorry! The answers should be in ljsyscall and Linux man pages.