snabbco/snabb #1096: Snabb threading model and synchronization

vmaffione, 2017-01-30:
Hi,
  I'm new to Snabb switch and I have two questions about the high-level architecture.

How is the threading model of Snabb engine like? If I define a complex App engine (with many Apps and many Links), is all the datapath workflow (e.g. push and pull machinery) executed by a single engine thread?

The second question is about synchronization between the Apps: from the documentation it seems that Snabb engine is able to use busy waiting or (configurable) periodic polls on the various links to see if there is work to do. Is it also possible to use any proper notification mechanism (e.g. condition variables or semaphores)?

Thanks!

eugeneia, 2017-02-04:
Hi @vmaffione, 

1. Yes, currently Snabb is a single threaded execution environment (one Snab instance per core). We have plans for making use of multiple cores. One approach we evaluated is to use existing NIC RSS features to “shard” incoming packets across multiple Snabb engines. This design is simple and efficient, but not all types of applications map well to this model. An alternate strategy we have experimented with is to provide “cross-engine links”, i.e. links between apps running in separate engines (Snabb instances). This design is very flexible, but is tricky to implement, as the overhead of cross-engine links must be very small for this to be feasible.

2. The Snabb engine schedules the apps to run. It is configurable how many packets apps will process in one batch, and if the engine should employ a pacing strategy to save cycles at the cost of some initial latency. The apps themselves are “passive”, implemented as a set of callbacks for the engine to execute when sensible (i.e. `push` is called when there are new packets on any input link of the app). An app can’t (read “should not”) block, be it for semaphores or I/O. Generally speaking, the OS level synchronization constructs as mentioned by you are way too expensive to qualify for use in Snabb apps.

vmaffione, 2017-02-04:
Hi @eugeneia ,
  Thanks a lot for your deep explanation, I am comparing different virtual switches solutions.

I'm sure you are using lockless queues to implement links, so probably you could implement cross-engine links in the same way using inter-process shared memory.
Indeed, OS synchronization mechanisms are expensive, so most of the high performance packet I/O solution explicitely avoid them; I just wanted a confirmation that this was the case also for Snabb.

Thanks.

closed 2017-02-04 by vmaffione