snabbco/snabb #1118: How to include custom modules ( .so files) in snabb.

suraj0208, 2017-03-31:
I want to include my own .so file which is compiled from .c file in snabb using lua's required.

How can I do that. when I try to do it, it gives the following error

`/usr/local/lib/lua/5.1/luatimelib.so: undefined symbol: luaL_setfuncs`

lukego, 2017-03-31:
Linking in your own C code at runtime or compile time is generally no problem.

This would usually be done using the [LuaJIT FFI](http://luajit.org/ext_ffi.html) which does not require your shared object to be linked with Lua. Can that work for you, or do you really need the Lua C API for some reason?

In principle you should be able to make both work, but we have always used the FFI in Snabb because it makes life so easy. I don't have an example based on the Lua C API to point to.

suraj0208, 2017-03-31:
Thanks alot . The FFI solved my problem. It would have been great to see shared library example.

lukego, 2017-03-31:
@suraj0208 There is quite a bit of FFI code landing in the next release via #1116 (Snabbwall) i.e. a binding to the nDPI library (interfaced as an external shared object.)

Currently we are relying on the LuaJIT documentation for making people aware of the FFI and how to use it. That may not be optimal but we have limited documentation writing resources of our own at the moment.