snabbco/snabb #1128: Unable to connect VhostUser port to Intel82599 port

vmaffione, 2017-04-12:
Hi,
  I have two dual port Intel 82599, so 4 interfaces, and two cables connecting the two pairs.

````
lspci | grep -i network
01:00.0 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
01:00.1 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
03:00.0 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
03:00.1 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
09:00.0 Ethernet controller: Intel Corporation 82574L Gigabit Network Connection
````

I've written this simple AppEngine (vm2nic.lua), that is meant to forward traffic between a VhostUser port and an 82599 port.
````
module(..., package.seeall)

local vhostuser = require("apps.vhost.vhost_user")
local ixgbe = require("apps.intel.intel_app")

function run (parameters)
   if not (#parameters == 2) then
     print("Usage: vm2nic <unix-socket> <nic-pci-addr>")
      main.exit(1)
   end
   local usock = parameters[1]
   local bdf = parameters[2]

   local c = config.new()
   config.app(c, "vh", vhostuser.VhostUser, {socket_path=usock,is_server=false})
   config.app(c, "nic", ixgbe.Intel82599, {pciaddr=bdf})

   config.link(c, "vh.tx -> nic.rx")
   config.link(c, "nic.tx -> vh.rx")

   engine.configure(c)
-- engine.busywait = true
-- engine.Hz = 20000
   engine.main({report = {showlinks=true, showapps=true}})
end
````
When I run `sudo src/snabb vm2nic /var/run/vm10-10.socket 0000:01:00.1`, snabb correctly connects to the VM (VhostUser port)
`````
Get features 0x18428001
 VIRTIO_F_ANY_LAYOUT VIRTIO_NET_F_MQ VIRTIO_NET_F_CTRL_VQ VIRTIO_NET_F_MRG_RXBUF VIRTIO_RING_F_INDIRECT_DESC VIRTIO_NET_F_CSUM
vhost_user: Read cached features (0x18028001) from /tmp/vhost_features___var__run__vm10-10.socket
Set features 0x18028001
 VIRTIO_F_ANY_LAYOUT VIRTIO_NET_F_CTRL_VQ VIRTIO_NET_F_MRG_RXBUF VIRTIO_RING_F_INDIRECT_DESC VIRTIO_NET_F_CSUM
rxavail = 0 rxused = 0
rxavail = 0 rxused = 0
`````

but when I transmit traffic on the virtio-net interface inside the VM, I don't see any traffic transmitted on the 82599 port identified by 0000:01:00.1 (e.g. I don't see traffic on the other end).

This is confirmed by snabb top, which shows all the packets are being dropped by snabb:
````
 Kfrees/s        freeGbytes/s    breaths/s      
 3198.68         0.19            25100          

 Min breath (us) Average         Maximum        
 1.24            13.82           74.10          

 Links (rx/tx/txdrop in Mpps)    rx      tx      rxGb    txGb    txdrop 
 nic.tx -> vh.rx                 0.00    0.00    0.00    0.00    0.00   
 vh.tx -> nic.rx                 0.00    0.00    0.00    0.00    3.20   
````

What is wrong here? Am I missing something? Should I use apps.intel_mp.intel_mp? If yes, how, and what is the difference?

Thanks

mwiget, 2017-04-12:
That should work as described. I happen to run a very similar tiny snabb app, connecting a pci port to a unix socket / linux interface:

Kfrees/s        freeGbytes/s    breaths/s
0.00            0.00            153000

Min breath (us) Average         Maximum
4.73            6.30            39.59

Links (rx/tx/txdrop in Mpps)    rx      tx      rxGb    txGb    txdrop
int.tx -> nic.rx                0.00    0.00    0.00    0.00    0.00
nic.tx -> int.rx                0.00    0.00    0.00    0.00    0.00

While the MPPS don’t show , I have connectivity.

The important elements of my lua program:

if pciaddr then
      local device_info = pci.device_info(pciaddr)
      if vlan then
         print(string.format("vlan set to %d", vlan))
      end
      if device_info then
         config.app(c, "nic", require(device_info.driver).driver,
         {pciaddr = pciaddr, vmdq = false, macaddr = src_mac, mtu = 9500})
         input, output = "nic.rx", "nic.tx"
      else
         fatal(("Couldn't find device info for PCI or tap device %s"):format(pciaddr))
      end
   end

   if int_interface then
      config.app(c, "int", raw.RawSocket, int_interface)
      config.link(c, output .. " -> int.rx")
      config.link(c, "int.tx -> " .. input)
   end

So two area’s to watch for:
1) are you disabling vmdq as I do to get any mac address pass?
2) did you disable iommu_intel to pt or off? (on wouldn’t work)

Marcel

vmaffione, 2017-04-12:
Ah I had iommu_intel=on.. By putting that off it works!
Thanks a lot!

closed 2017-04-12 by vmaffione