snabbco/snabb #1129: Packets not being received from intel_mp app

raj2569, 2017-04-13:
I am trying to use intel_mp app  but I am not receiving any packets in the tx link. not sure if I am correctly using the app, so wrote a small test program. Basically it takes the packets from tx to input of my test app, and my app discards it after logging it.

```
module(..., package.seeall)
local lib = require("core.lib")
local logger = lib.logger_new({ rate = 32, module = 'testintel' });

testintel = {}

function testintel:new ()
   local o = { packet_counter = 1 }
   return setmetatable(o, {__index = testintel})
end

function testintel:push()
   local i = assert(self.input.input, "input port not found")
   
   while not link.empty(i) do
      self:process_packet(i)
      self.packet_counter = self.packet_counter + 1
   end
end

function testintel:process_packet(i)
   local p = link.receive(i)
   local data, length = p.data, p.length
   logger:log ("received a packet with length " .. length)
   packet.free(p)
end
```

```
module(..., package.seeall)

local test  = require("program.example_test.testintel")
local intel = require("apps.intel_mp.intel_mp")

function run (parameters)
   if not (#parameters == 1) then
      print("Usage: example_spray <pci>")
      main.exit(1)
   end
   local pci_id = parameters[1]

   local c = config.new()
   config.app(c, "if", intel.Intel, {pciaddr=pci_id})
   config.app(c, "test_app", test.testintel)

   config.link(c, "if.tx -> test_app.input")

   engine.configure(c)
   engine.main({duration=1, report = {showlinks=true}})
   -- engine.main({report = {showlinks=true}})
end
```
output shows:
```
./snabb example_test 0000:04:00.0
link report:
                   0 sent on if.tx -> test_app.input (loss rate: 0%)
```

From dmesg I can see that the the driver seems to have released the card and the interface is not visible in ifconfig -a
```
[  277.065612] ixgbe 0000:04:00.1: complete
[  279.121101] ixgbe 0000:04:00.0: complete
```

lspci shows:

```
04:00.0 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
04:00.1 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
```

Any help to figure out what is wrong is much appreciated.

raj

cc: @petebristow

petebristow, 2017-04-13:
@raj2569 the `intel_mp` app uses link names of `input` and `output` rather than `rx` and `tx`. Also if you want to receive packets you need to specify an `rxq` to listen on. The docs mark the field as optional, but that means you don't have to listen on a nic not that you don't have to specify an `rxq` and still be able to listen. You have to specify an `rxq` if you want to receive and a `txq` if you want to send.

https://gist.github.com/anonymous/4fe4c57cc39feec4093b57e2538d985a

Is a complete gist showing the `intel_mp` driver in receive only mode.

raj2569, 2017-04-13:
Thank you, this solved my issue.

closed 2017-04-13 by raj2569