snabbco/snabb #1144: core/main.lua:26: Failed to allocate a huge page for DMA

raj2569, 2017-05-17:
Hello,

To investigate #1138 further I created a simple test program to replicate the issue. 

```
module(..., package.seeall)

local sprayer = require("program.test_zeropkts.sprayer")
local intel    = require("apps.intel_mp.intel_mp")

function run (parameters)
   local c = config.new()
   config.app(c, "south_if", intel.Intel, {pciaddr="04:00.0", rxq = 0, txq = 0})
   config.app(c, "spray_app", sprayer.Sprayer)
   
   config.link(c, "south_if.output -> spray_app.input")
   
   engine.configure(c)
   engine.main({report = {showlinks=true}})
end
```

and

```
module(..., package.seeall)
local ethernet = require("lib.protocol.ethernet")

Sprayer = {}

function Sprayer:new ()
   local o = {
      eth_pkt        = ethernet:new({}),
      packet_counter = 1
   }
   return setmetatable(o, {__index = Sprayer})
end

function Sprayer:push()
   local i = assert(self.input.input, "input port not found")

   if not link.empty(i) then
      self:process_packet(i)
   end
end

function Sprayer:process_packet(i)
   while not link.empty(i) do
      local p = link.receive(i)
      local data, length = p.data, p.length
      if (length > 14) then
         local eth_pkt = self.eth_pkt:new_from_mem(data, length)
      else
         self.packet_counter = self.packet_counter + 1
         print ("got a packet with length " .. length)
      end
   end
end
```

While running the program I get the following messages as soon as the packets from packetblaster starts.

```
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 34 -> 35]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 35 -> 36]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 36 -> 37]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 37 -> 38]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 38 -> 39]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 39 -> 40]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 40 -> 41]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 41 -> 42]
...
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 1250 -> 1251]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 1251 -> 1252]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 1251 -> 1252]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 1251 -> 1252]
core/main.lua:26: Failed to allocate a huge page for DMA
stack traceback:
        core/main.lua:138: in function <core/main.lua:136>
        [C]: in function 'error'
        core/main.lua:26: in function 'assert'
        core/memory.lua:47: in function 'allocate_next_chunk'
        core/memory.lua:36: in function 'dma_alloc'
        core/packet.lua:93: in function 'new_packet'
        core/packet.lua:197: in function 'preallocate_step'
        core/packet.lua:86: in function 'allocate'
        apps/intel_mp/intel_mp.lua:520: in function 'method'
        core/app.lua:128: in function 'with_restart'
        core/app.lua:440: in function 'thunk'
        core/histogram.lua:98: in function 'breathe'
        core/app.lua:379: in function 'main'
        program/test_zeropkts/test_zeropkts.lua:14: in function 'run'
        core/main.lua:56: in function <core/main.lua:43>
        [C]: in function 'xpcall'
        core/main.lua:179: in main chunk
        [C]: at 0x0044f870
        [C]: in function 'pcall'
        core/startup.lua:3: in main chunk
        [C]: in function 'require'
        [string "require "core.startup""]:1: in main chunk
```

The packetblaster running in the same host successfully allocates huge page.

```
sudo ./snabb packetblaster synth -S 1514 0000:04:00.1
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 20 -> 21]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 21 -> 22]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 22 -> 23]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 23 -> 24]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 24 -> 25]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 25 -> 26]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 26 -> 27]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 27 -> 28]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 28 -> 29]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 29 -> 30]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 30 -> 31]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 31 -> 32]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 32 -> 33]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 33 -> 34]
Transmissions (last 1 sec):
```

I tried to reset huge page table with `echo 20 > /proc/sys/vm/nr_hugepages` couple of times, but always results in failed allocation.

lukego, 2017-05-17:
This looks like a memory leak. The process is successfully allocating huge pages, but eventually it asks for too many, and the kernel is not able to provide more than 1251 in total. It's okay for packetblaster because it stops allocating new pages after a while.

Just need to add a `packet.free(p)` into the method `Sprayer:process_packet()` :)

raj2569, 2017-05-17:
oops, left that out  :)

closed 2017-05-17 by raj2569