snabbco/snabb #1175: How to filter or save data from UDP payload

MatthiasKauer, 2017-07-09:
Hi,
I am addressing you with a question because snabb looks very promising for what I am trying to do.
I would like to extract payload data from a PCAP.

The overall app could like something like this: PCAPReader -> PCAPFilter -> MyExtractor -> File

For MyExtractor, I have adapted the Sprayer example as follows:
```
local p = link.receive(i)
local s1, s2 = struct.unpack(">II", ffi.string(p.data, p.length))
```

My questions are:
- Is this the right place for such questions? I apologize if not.
- What is the best way to access payload data? ffi.string and struct.unpack (from lua-struct) could be expensive operations. Are they? Is vstruct better? Is there such functionailty available in snabb already?
- Can I develop my tool without the compile loop somehow?
- If I store the data, say `s2` from above in a global list, how would I know that the PCAP file has ended and I need to write out the results. Or should I implement a form of CSV file data sink where data is appended as the program runs?

Thank you for your help.
Matthias

lukego, 2017-07-18:
Hi Matthias,

> Is this the right place for such questions? I apologize if not.

Yes, this is the right way to ask your question. Sorry about the wait.

> What is the best way to access payload data? ffi.string and struct.unpack (from lua-struct) could be expensive operations. Are they? Is vstruct better? Is there such functionailty available in snabb already?

If you are satisfied with the performance of the struct package, etc, then that is fine. Could be reasonable for this application.

Digression for how it is usually done in realtime applications on live traffic...

Just want to remove headers from the front of a packet? `packet.shiftleft(p, numBytes)` will only cost a few cycles.

Want to construct a new packet by reference the payload of an existing one? `packet.from_pointer(other_packet.data + offset, length)`.

Want to access protocol fields in a structured way? See `lib.protocol` library.

More details in the [docs](http://snabbco.github.io/).

> Can I develop my tool without the compile loop somehow?

Yes. You can write a Lua script and run it like this:

```
$ sudo snabb snsh myscript.lua
```

See `snabb snsh --help` for more information.

> If I store the data, say s2 from above in a global list, how would I know that the PCAP file has ended and I need to write out the results. 

I'm not sure that I understood this one, but I do see a challenge that an app network doesn't necessarily know when it is "finished" because they usually run forever. So could need some hack... :).

MatthiasKauer, 2017-07-20:
Thank you for the explanations.

When you say "Want to access protocol fields in a structured way? See lib.protocol library." what do you mean by "protocol fields". I want to access fields in the payload in a structured way and would like to create my own protocol for that, I suppose. Is the protocol lib still a good starting point?

Finally, you say that the network app doesn't know when it's finished: I had no issues with the app finishing; the only source is the PCAP reader and that is just done at some point, no?