snabbco/snabb #1177: myapp:push is not getting called

raj2569, 2017-07-17:
Hi all,

I am sure this is some silly oversight by me, but can't figure out what could be wrong here, been banging my head for past two days. Just posting it here for some more eyeballs :)

Consider the following two lua files, 
`testpush.lua`
```lua
module(..., package.seeall)
local tap        = require("apps.tap.tap")
local pushtest   = require("program.testpush.pushtest")

function run (parameters)
   local c = config.new()
   config.app(c, "snabbtap",   tap.Tap, "snabbtap")
   config.app(c, "pushtest",   pushtest.PushApp)
   
   config.link(c, "pushtest.output1 -> snabbtap.input")
   
   engine.configure(c)
   engine.main({duration=10, report = {showlinks=true}})
end
```

`pushtest.lua`
```lua
module(..., package.seeall)

PushApp = {}

function PushApp:new ()
   local o = {}
   return setmetatable(o, {__index = self})
end

function PushApp:push()
   assert(self.output.output, "missing output link")
   print ("inside PushApp:push")
end

-- function PushApp:pull()                                                                                                  
--    print ("inside PushApp:pull")                                                                                         
-- end                                                                                                                      
```

On compiling and running, I expect the assertion to fail, as in my engine configuration I am connecting `output1` to `snabbtap` and if that is not happening at least the print to be executed.

Unfortunately neither of them is happening. If I uncomment the `pull()` function, it is working as expected.

dpino, 2017-07-17:
An app's `push` method is called when receiving packets on that app. Your app graph is defined as:

```lua
config.link(c, "pushtest.output1 -> snabbtap.input")
```

Snabb's main loop consists of two steps: an inhale step that pulls packets into the links and an exhale step that pushes packets into the apps. According the graph you've defined above the following app methods will be called: `pushtest:pull` (inhale) and `snabbapp:push` (exhale). And that's why your `PushApp:pull` method gets called but `PushApp:push` not (it's Tap:push which is being called instead).

raj2569, 2017-07-17:
Thanks for the clarification. 

But, how do I configure my app graph if pushtest app is like packetgen? i.e. It has only a single output link, on to which the generated packets needs to be pushed?

dpino, 2017-07-17:
Yes, exactly. If you want to turn your pushtest app into an app that generates packets it has to behave similarly to other packet generator apps (a network driver, loadgen or pcap.PcapReader). All these apps overwrite the `pull`method and there create a packet out of somewhere (the network card, from scratch or fetching it from a .pcap file) and transmit the packet into an output link.

It would look something like this:

```
function PacketGenerator:pull()
   local output = self.output.output
   while not link.full(output) do
      -- Generate packet.
      local pkt = packet.create()
      link.transmit(output, pkt)
   end
end
```

In your graph, the output link has to have the same name as used in the `pull` method, otherwise the link could not be found.

```
config.link(c, "pushtest.output -> snabbtap.input")
```

raj2569, 2017-07-17:
After checking the source of `Source` it became clear that I should be using `pushtest:pull` to generate packets. I misunderstood that the `push` is used to send out packets from the links, while `pull` is used to pull in the packets from links.

raj2569, 2017-07-17:
@dpino Thank you for your insight, just one more question to complete this discussion. What are the general use cases for `push`? I am still not very clear about the difference between pulling the packets _into the links_ and pushing the packets _into the app_

wingo, 2017-07-17:
One pulls packets *into the network*; the network then processes those packets by calling `push` functions of the apps in the network.  First, all apps with pull functions get their pull functions called -- that pulls packets off the NIC and into Snabb.  Then for all links like `a.out -> b.in`, Snabb will call `b:push()` so that `b` can take packets off of its `in` input link and "process" those packets, possibly pushing them out some other link.

closed 2017-07-20 by raj2569