snabbco/snabb #1200: NUMA get_mempolicy failing on Ubuntu 4.10 kernel

wingo, 2017-08-04:
Try this:

```
sudo ./snabb snsh -e 'assert(require("syscall").get_mempolicy())'
```

Works locally on a GuixSD system with Linux 4.10.2 and just one NUMA node, works on a NixOS Linux 3.18.26 machine with two NUMA nodes, doesn't work on Ubuntu 17.04 with Linux 4.10-xx on any number of NUMA nodes.  The error is "invalid argument".  We need to fix this!

mwiget, 2017-08-04:
It also fails on Ubuntu 16.04.3 LTS with Linux 4.10.0-xx kernel.

dpino, 2017-08-04:
I run the test case with `strace` in a NUMA host and a non NUMA host.

NUMA host:

```
get_mempolicy([MPOL_DEFAULT], [000000000000000000], 64, NULL, 0) = 0
```

non-NUMA host:

```
get_mempolicy(0x40c820c8, 0x40c820f8, 64, NULL, 0) = -1 EINVAL (Invalid argument)
```

Apparently the the problem is in the first two parameters: `mode` and `mask`.

dpino, 2017-08-08:
Some updates on the bug.

I took a look a the v4.10 kernel source and apparently the problem is in `maxnode` value. When it gets initialized, it's default value is 64. According to get_mempolicy man page an EINVAL is returned if _"The value specified by maxnode is less than the number of node IDs supported by the system"_. It's this check in the kernel source:

https://github.com/torvalds/linux/blob/v4.10/mm/mempolicy.c#L1477

In my system (Ubuntu 17.04) if I modify ljsyscall and initialize mask.size to 1024, there's no error (`lib/numa.lua` selftest passes).

In the kernel, MAX_NUMNODES gets initialized here:

https://github.com/torvalds/linux/blob/master/include/linux/numa.h

I believe that in the kernel shipped by Ubuntu 17.04 the value of NODES_SHIFT is 10 thus MAX_NUMNODES is 1024.

benagricola, 2017-09-12:
Having the same issue on a CentOS 7 server with the elrepo 4.12.10 kernel, using the `numa.bind_to_cpu()` function. 

As @dpino mentioned, setting mask.size to 1024 by default avoids this, but isn't portable in any fashion.

I assume that the correct fix is to read the value of `MAX_NUMNODES` at runtime and initialise mask.size based on that. There doesn't seem to be an easy, direct way to do this (but my knowledge here is massively lacking).

`numactl` mimics reading `MAX_NUMNODES` by reading `/proc/self/status` and parsing the `Mems_allowed:` field :) - see [here](/numactl/numactl/blob/6e2c6739dffcab9e0d2b9470066070d8d686bf68/libnuma.c#L377)

Each 9 characters in the field represents 32 bits of mask, so the value of `MAX_NUMNODES` is `((strlen($mems_allowed)+1)/9)*32`

I'd hope there's a better way of getting the value of `MAX_NUMNODES` but maybe there isn't...

dpino, 2017-09-17:
@benagricola Thanks for the tip. I don't know either if there's a better way to retrieve MAX_NUMNODES value. I pushed a patch that calculates MAX_NUMNODES as you suggested.

PR #1228 (2017-11-28): Set mask.size equals to MAX_NUMNODES if lower
Fixes #1200.

`get_mempolicy` syscall returns an EINVAL if argument maxnode is lower than the value of MAX_NUMNODES defined in the system:

```
int get_mempolicy(int *mode, unsigned long *nodemask,
                         unsigned long maxnode, void *addr,
                         unsigned long flags);
```

In ljsyscall `S.get_mempolicy` the argument maxnode is mask.size. What the patch does is to check the value of mask.size is lower than MAX_NUMNODES and in that case set mask.size to MAX_NUMNODES.

That fixes the call to `get_mempolicy` in Ubunt [... cut]