snabbco/snabb #1210: mmap: Invalid argument

leolovenet, 2017-08-18:
@lukego  When i use the latest version Snabb on CentOS6(2.6.32-696.3.2.el6.x86_64), I get the [error.txt](https://github.com/snabbco/snabb/files/1233265/error.txt) message, then I traced why，
I found it because `local tmpptr = syscall.mmap(nil, size, "read, write", "shared, hugetlb", fd, 0)` in [memory.lua](https://github.com/snabbco/snabb/blob/master/src/core/memory.lua#L149)， return `nil`, Then I look at the [history](https://github.com/snabbco/snabb/commits/master/src/core/memory.lua) of this file, Found that you rewrite the function `allocate_huge_page ` using Lua for supporting multiprocess, But, this function return `Invalid argument` with flags `shared, hugetlb`,  buf it is OK with flags `shared, anonymous, hugetlb`, I tested. 

At first I suspect that is the kernel version of the problem, So I changed the  [code](http://elixir.free-electrons.com/linux/v2.6.32.71/source/Documentation/vm/map_hugetlb.c) to test it under ubuntu 16.04 (4.4.0-63-generic).  But, It return the same, `mmap: Invalid argument` when flags was `shared, hugetlb`.

So the problem is to use MAP_ANONYMOUS to conflict with your design, What should i do ?

leolovenet, 2017-08-18:
I figured it out,  when i readed the code https://github.com/torvalds/linux/blob/master/tools/testing/selftests/vm/hugepage-mmap.c, Found no `MAP_HUGETLB` flag required, so I remove the `hugetlb` flage too, and It work. 
 :)

memory.lua，Should be like the following:

[149](https://github.com/snabbco/snabb/blob/master/src/core/memory.lua#L149): `local tmpptr = syscall.mmap(nil, size, "read, write", "shared", fd, 0)`
[154](https://github.com/snabbco/snabb/blob/master/src/core/memory.lua#L154): `local ptr = syscall.mmap(virt, size, "read, write", "shared, fixed", fd, 0)`

BTW, I think this is a bug.

dpino, 2017-08-22:
The `hugetlb` flag requests to mmap a region of memory as a hugepage. For some reason requesting a mmap as a hugepage is not working in your system. It seems from the error report you attached that hugepages are enabled in your system. Snabb was at least able to allocate some hugepages and then failed:

```
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 503 -> 504]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 504 -> 505]
[memory: Provisioned a huge page: sysctl vm.nr_hugepages 505 -> 506]
core/main.lua:26: Failed to allocate a huge page for DMA
file not found: core/main.lua: No such file or directory
```

Normally a hugepage size is 2048Kb. Check out how many hugepages are available in your system. This is what my system reports:

```bash
$ cat /proc/meminfo | grep Huge
AnonHugePages:         0 kB
HugePages_Total:    4096
HugePages_Free:     4096
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
```

Unless your system starts with 3 or less hugepages, there shouldn't be any issue. In any case, try increasing the number of hugepages in your system:

```
sysctl -w vm.nr_hugepages=512
```

Another possible cause is that you're using a `shmax` limit that is lower than the default hugepage size. Snabb should automatically handle that if that's the case. See https://github.com/snabbco/snabb/issues/366. May trigger some ideas.

As for the kernel version, I run Snabb on a 3.18.X kernel and I never stumbled into this issue.

leolovenet, 2017-08-23:
@dpino Sorry, this log misled you (I didn't notice it) because I did a lot of tests, so the number of `vm.nr_hugepages` started from 503.

after, I cleaned up everything and rebooted my server, re-tested again. Here is the 
[new log](https://github.com/snabbco/snabb/files/1243900/new.txt).

I know that the value of `vm.nr_hugepages` is growing, that is because this [line ](https://github.com/snabbco/snabb/blob/master/src/core/memory.lua#L86) 
```
lib.writefile("/proc/sys/vm/nr_hugepages", tostring(want))
```
 was successfully executed. The reason why it was repeated execution, is because [here](https://github.com/snabbco/snabb/blob/master/src/core/memory.lua#L63), not return OK.
I have also tested this under CentOS7, Also like this. 

I know the `shmax` limit，but I'm sure it has nothing to do with it.
Here is the output on my system
```
> cat /proc/sys/kernel/shmmax
68719476736
```

according to [https://www.kernel.org/doc/Documentation/vm/hugetlbpage.txt](https://www.kernel.org/doc/Documentation/vm/hugetlbpage.txt) "Using Huge Pages" section,

>  Any files created on /mnt/huge uses huge pages. 

and
> Also, it is important to note that no such mount command is required if
applications are going to use only shmat/shmget system calls or mmap with
MAP_HUGETLB. 

We have used  `ensure_hugetlbfs` function ensure that mounted the hugetlbfs filesystem (on  /var/run/snabb/hugetlbfs),  so the file created by 
```
local tmpfile = "/var/run/snabb/hugetlbfs/alloc."..syscall.getpid()
```
should used huge pages.

So I deleted the `hugetlb` flag in the `memory.lua` file, and It work.

```
149: local tmpptr = syscall.mmap(nil, size, "read, write", "shared", fd, 0)
154: local ptr = syscall.mmap(virt, size, "read, write", "shared, fixed", fd, 0)
```

When my program was running, I found that `HugePages_Free` was reduced.

This may be the operating system, or what other reasons, I do not know.

Hope this information can help to other people.

dpino, 2017-08-29:
In summary what it seems it's happening is that when Snabb tries to allocate a huge page in your system, it cannot. After 3 attempts it fails.

I agree `shmax` is not a problem since  "memory: Enabling huge pages for shm..." message is part of the log.

Try the following minimal test case:

```lua
local syscall = require("syscall")
local memory = require("core.memory")

local resolve_physical = memory.resolve_physical
local ensure_hugetlbfs = memory.ensure_hugetlbfs

local tag = 0x500000000000ULL

function allocate_huge_page (size,  persistent)
   ensure_hugetlbfs()
   local tmpfile = "/var/run/snabb/hugetlbfs/alloc."..syscall.getpid()
   local fd = syscall.open(tmpfile, "creat, rdwr", "RWXU")
   assert(fd, "create hugetlb")
   assert(syscall.ftruncate(fd, size), "ftruncate")
   local tmpptr = syscall.mmap(nil, size, "read, write", "shared, hugetlb", fd, 0)
   assert(tmpptr, "mmap hugetlb")
   assert(syscall.mlock(tmpptr, size))
   local phys = resolve_physical(tmpptr)
   local virt = bit.bor(phys, tag)
   local ptr = syscall.mmap(virt, size, "read, write", "shared, hugetlb, fixed", fd, 0)
   local filename = ("/var/run/snabb/hugetlbfs/%012x.dma"):format(tonumber(phys))
   if persistent then
      assert(syscall.rename(tmpfile, filename))
      shm.mkdir(shm.resolve("group/dma"))
      syscall.symlink(filename, shm.root..'/'..shm.resolve("group/dma/"..lib.basename(filename)))
   else
      assert(syscall.unlink(tmpfile))
   end
   syscall.close(fd)
   return ptr, filename
end

allocate_huge_page (2048 * 1024)
```

I can execute that program in my system without errors:

```bash
$ sudo ./snabb snsh allocate.lua
$ echo $?
0
```

If you're not able to run the program above, that means there's a problem in your system environment. To figure out what's wrong I would run the program above with strace (`sudo strace ./snabb snsh allocate.lua`) and check the actual system call that is failing. Then try to isolate that system call in a more minimal C program and figure out why it's failing.

petebristow, 2017-11-20:
Hi @lukego @dpino @leolovenet,

https://groups.google.com/forum/#!topic/linux.kernel/Hwyme8jQyDM sheds some light on this. Basically `hugetlb` on a file on `hugetlbfs` was considered invalid at the time they minted CentOS 7. It has since become accepted but isn't required.

Does any one have objections to just removing the `hugetlb` / `MAP_HUGETLB` ?

Pete

dpino, 2017-11-21:
So if I understand correctly as long as the file descriptor is from a `hugetlbfs`, there's no need to add the `hugetlb` / `MAP_HUGETLB` flag on a `mmap` (in fact, that created an error on CentOS 7). It's a bit strange, but if that makes Snabb can run on more systems I'm fine with it.

lukego, 2017-11-21:
Sounds basically reasonable to me too. The change is small and the benefit is tangible.

The only concern I have is that **if** a kernel would ever provide us with non-HugeTLB memory for some reason then this would be really bad. DMA of ~4KB+ jumbo frames would cause a buffer overrun in physical memory. So we need to be confident that this won't happen.

Likely this does not happen in practice, since the memory is backed by hugetlbfs, but please consider whether or not you think it is worth checking the [`HUGE` flag of `/proc/kpageflags`](https://github.com/torvalds/linux/blob/master/Documentation/vm/pagemap.txt#L65) to protect against such an error.

I am taking a moment to be thankful that our interface towards the kernel and OS is so narrow and that we have extremely little exposure to this kind of quirk compared with most projects :).

kullanici0606, 2018-07-06:
Any updates in this issue? I also tried the example_replay program from the "Snabb Getting Started Guide" on CentOS 7 i.e 
`sudo src/snabb example_replay src/program/example_replay/input.pcap veth0`
and also the minimal test case and they both failed with the same error.
When I remove hugetlb, as suggested, from mmap, the minimal test case works.

kullanici0606, 2018-07-06:
If I remove the hugetlb parameter in the following lines:

https://github.com/snabbco/snabb/blob/732eae91ff1d9e18570804b42e2ff90534361f9b/src/core/memory.lua#L149-L154

and run `make -j` again, can I assume that Snabb will still use huge pages?

EaseTheWorld, 2018-07-31:
I met same error when running example_replay.
Removing the hugetlb parameter does work.
I checked HugePages_Free increased(app start) and decreased to 0(app stop)
Tested on RHEL 6.9 and 7.4

lukego, 2018-07-31:
Great detective work everybody! This all sounds reasonable to me. Since we are mapping the memory in shared mode on hugetlbfs and with mlock I don't think the kernel really has any option except to map the huge page into our address space.

First one to make a PR out of this rocks :) we can slip the fix quickly into the next release.

kullanici0606, 2018-07-31:
If it is enough to remove parameters, then I made the changes and created a PR : https://github.com/snabbco/snabb/pull/1373
Could you please review it or give feedback what else needs to be done?

commit 73de63bb (2018-07-31): Fixes: #1210 hugetlb parameter causes problem on Centos and since
memory is already mapped on hugetlbfs and with mlock, there is no
need to explicitly use hugetlb parameter
(See comment: https://github.com/snabbco/snabb/issues/1210#issuecomment-409123442)
closed 2018-09-12 by eugeneia