snabbco/snabb #1212: SipHash / ctable badness

wingo, 2017-08-21:
In https://github.com/Igalia/snabb/pull/909, the first patch had a flaky CI failure in `lib.ctable`:

```
src/testlog/lib.ctable:
selftest: ctable
core/main.lua:26: 43

Stack Traceback
===============
(1) Lua function 'handler' at file 'core/main.lua:168' (best guess)
	Local variables:
	 reason = string: "core/main.lua:26: 43"
	 (*temporary) = C function: print
(2) global C function 'error'
(3) Lua global 'assert' at file 'core/main.lua:26'
	Local variables:
	 v = boolean: false
(4) Lua field 'selftest' at file 'lib/ctable.lua:637'
	Local variables:
	 bnot = C function: builtin#65
	 occupancy = number: 2e+06
	 params = table: 0x41586758  {initial_size:5000000, key_type:ctype<unsigned int [1]>, max_occupancy_rate:0.4 (more...)}
	 ctab = table: 0x410842a8  {scale:0.001164153451371, occupancy_lo:0, entries:cdata<struct 1397 *>: 0x2b9f00c00000 (more...)}
	 (for index) = number: 2
	 (for limit) = number: 2
	 (for step) = number: 1
i = number: 2
...
```
Full log here: https://gist.github.com/lwaftr-igalia/a39bdeaa48712b33a40254feb97323b8

(A flaky failure is a failure that only happens sometimes.  Appearance of a flaky failure doesn't necessarily indicate a problem in the PR in which is appears.)

This failure corresponds to a check in which a ctable is populated with the first 2 million integers as keys, and then we check that the max displacement on the table is below a certain number.  Code here: 
https://github.com/snabbco/snabb/blob/master/src/lib/ctable.lua#L604.  When we used the Jenkins hash, this value was 8.  Then when we switched to randomly-seeded SipHash, the value was often 8 or 9 and I thought, well, fine.  Of course this test has a probabilistic aspect to it but I do want to ensure that max displacement doesn't go off the rails -- otherwise the branchless lookup streamer won't work well.  (It takes N keys, hashes them to produce N hashes, streams in the M contiguous max-displacement entries for each key, then does binary search over those M entries.  Total cost N*M for memory, log(M) for search.  M is the max displacement, so if even one entry has a huge displacement for the table, it incurs a cost for all lookups.)

So we changed the assertion to just be, the max displacement should be less than 15.  I didn't do any science or measurement to know what to expect assuming perfect bit mixing of hashes, but oh well.  Anyway in that test the assertion indicates that the max displacement is **43**.  That's terrible!!!!

We need to do more science to know more about max displacements for the siphash hash function.

We could also try with SipHash2,4 (currently we are using SipHash1,2).  Or experiment with different mixing strategies.

Perhaps we should rehash a table if the max displacement exceeds a threshold.

In any case this is a problem that we need to understand better!

takikawa, 2017-09-17:
I looked into this issue a bit last week and I suspect that the displacement issue in this test isn't due to a problem with the hash function (which I've tested and looks to match the reference implementation too), but due to some weird memory corruption bug either in ctable or in LuaJIT.

My reason for suspecting this is first of all that the faulty displacement is *always* `43` and not a random amount based on the seed. Actually, setting the hash seed to be a constant does not change the assertion failures. This also suggests it's not the algorithm's fault.

The other is that if you look at [this line](https://github.com/snabbco/snabb/blob/master/src/lib/ctable.lua#L261) in the `load` method, the `header.max_displacement` field is not set to `43` before the call but gets set to `43` after the call on failing runs. The `hash_seed` field also seems to get written over too. It's non-deterministic though, so I'm not sure what exactly it is about that function call that causes it.

(Note: to inspect what's going on, I've been running a shell snippet like `status=0; while [ $status -eq 0 ]; do sudo ./snabb snsh -t lib.ctable; status=$?; done` and printing the values of fields)

takikawa, 2017-09-19:
Update: I think the real bug here is that the `selftest` uses the FFI in such a way that the buffer underlying the `header` struct ends up getting GCed before it's read.

Here's a simplified example that shows the issue:

```
-- Example showing how Luajit GC works

local ffi = require("ffi")

ffi.cdef[[
  struct s_t {
    uint8_t x;
    uint8_t y;
  };
]]

-- Uncomment to make the two print calls at the end the same
--local prevent_gc

local function read()
   local buf = ffi.new("uint8_t[2]")
   --prevent_gc = buf
   buf[0] = 255; buf[1] = 254;
   return buf
end

local function read_s()
   return ffi.cast("struct s_t*", read())
end

local s = read_s()
print(s.x)
print(s.y)
collectgarbage()
print(s.x)
print(s.y)
```

This test script will print "255\n254" and then some junk data because `buf` gets deallocated. Similarly, the `read_ptr` function at https://github.com/snabbco/snabb/blob/master/src/lib/ctable.lua#L690 lets the buffer in `read` get deallocated. I think this is consistent with the [FFI semantics](http://luajit.org/ext_ffi_semantics.html) but it's kind of hard to tell.