snabbco/snabb #1224: intel_mp compatible devices

benagricola, 2017-09-08:
I've been using the intel_mp (Intel82599) driver with:

```
05:00.0 Ethernet controller: Intel Corporation Ethernet 10G 2P X520 Adapter (rev 01)
05:00.1 Ethernet controller: Intel Corporation Ethernet 10G 2P X520 Adapter (rev 01)
```

for the last week or so. This is an Intel 82599 based card and seems to work fine with the Intel82599 driver.

I added the following to the `byPciId` table to get it working:

```
[0x154d] = { registers = "82599ES", driver = Intel82599, max_q = 16 }
```

Was just wondering if there's a reason this wasn't already in there (aside from whoever wrote intel_mp not having an exhaustive list of Intel Pci ID's, or if there'd be any issue submitting a PR to add it?

petebristow, 2017-10-26:
I didn't have an exhaustive list of IDs nor did I really want to add IDs that I couldn't test. 
Are you going to have this hardware hanging around to verify for breaking changes that might happen?

benagricola, 2018-05-22:
Wow, missed your comment on this completely @petebristow.

FWIW pretty much all of the Snabb stuff we use is using one of these cards (X520 SR2) - I've seen no issues with the intel_mp driver on this for the last 6 or so months and it appears the ID is changed because it's an SR card which is supplied with SFP's pre-installed.