snabbco/snabb #1227: Virtio driver not compliant with latest specification

ichung, 2017-09-16:
The virtio_net driver included in snabb has not been updated to the [Virtio 1.0 specification](http://docs.oasis-open.org/virtio/virtio/v1.0/virtio-v1.0.html).  This means that virtio devices will eventually stop supporting it as they drop legacy driver support.

dpino, 2017-09-17:
The `virtio_net` driver was mostly tested using QEMU v2.4  + SnabbNFV on the host side. Then on the guest side there's no requirements. As far as I recall, other users have reported the `virtio_net` driver was not working when using other hypervisors, or even with QEMU versions higher than 2.6.

My suggestion is, if possible, to use QEMU v2.4 + SnabbNFV on the host side.

dpino, 2018-08-07:
PR https://github.com/snabbco/snabb/pull/1369 fixes `vhost-user` driver for QEMU >v2.8. Likely this was the reason of the error. FWIW, the virtio-net driver is still working as tested in https://github.com/Igalia/snabb/pull/1165.