snabbco/snabb #1253: Snabb at FOSDEM 2018

lukego, 2017-11-21:
Snabb hackers will be gathering at the [FOSDEM 2018](https://fosdem.org/2018/) conference on the weekend of Feb 3-4 2018. **Join us!**

People who are rumoured via Slack (#1252) to be meeting up: @asumu @benagricola @dpino @eugeneia @lukego @petebristow @wingo. (Leave a comment below to add/remove your name here!)

TODO:

- [ ] Confirm [JAM Hotel](http://www.jamhotel.be/) as the suggested hotel choice.
- [ ] Pick a venue for Friday afternoon/evening pre-conference socializing.
- [ ] Confirm which Snabb-related presentations there will be at FOSDEM.

dpino, 2017-11-22:
Confirmed I will attend Fosdem next year.

JAM Hotel looks really cool. Usually I stay at whatever hotel other igalians go (there's a bunch of us attending Fosdem every year so we tend to organize the trip together). I think we haven't chosen a hotel yet, so I suggested to stay at JAM Hotel (and kill two birds with one shot).

As for talks, I proposed an introductory Lighting Talk about Snabb (no notification yet whether was approved or not).

lukego, 2017-11-22:
@dpino Cool see you there :).

@wingo floated the idea that on Friday we could even book a conference room somewhere for an informal miniconf where we can each present what we are working on at the moment? Is anybody up for presenting something? (anyone up for organizing/moderating?)

This could be a nice venue for talks/demos that require familiarity with Snabb. I would be glad to show the RaptorJIT/Studio trace profiling tools that I am hacking.

eugeneia, 2017-11-22:
@lukego That sounds really nice! I could talk a bit about random stuff around Vita. :-)

mwiget, 2017-11-22:
I plan to join too. I guess it is too late to submit a talk, but happy to just tag along and mingle.

justincormack, 2017-11-22:
I will be around at Fosdem (but not until late on Friday).

xray7224, 2017-11-24:
I shall be there too with the other Igalian folks :)

lukego, 2017-11-24:
Groovy :) I am updating the info above as people leave comments.

lukego, 2018-01-01:
Seems like everybody is happy with JAM as the suggested hotel so I've booked in there too. I plan to arrive Friday afternoon and hang out with whoever is around over the evening before the conference. Sounds like we are all up for some informal chats/demos. Have to think about how to bump into each other at the main conference, sounds like that is kind of massive...

eugeneia, 2018-01-05:
I booked in a flat really close to JAM and will be arriving at lunch time. See you guys there. :-)

corsix, 2018-01-08:
I will be around from the afternoon of the 2nd through until the afternoon of the 6th, staying at JAM.