snabbco/snabb #1278: Problem with packet generation by using Intel XL710 (40GbE)

danilocer, 2018-02-09:
Does snabb support the following NIC?
Intel Corporation Ethernet Controller XL710 for 40GbE QSFP+ (rev 02)
(i40e driver is used)

because I'm trying to generate packets with packetblaster but everytime I run it
./snabb packetblaster replay ~/trill64.pcap 0000:02:00.0

it gives me the following error:

filename=/root/trill64.pcap
core/main.lua:26: <PCI> matches no suitable devices.

Stack Traceback
===============
(1) Lua function 'handler' at file 'core/main.lua:168' (best guess)
	Local variables:
	 reason = string: "core/main.lua:26: <PCI> matches no suitable devices."
	 (*temporary) = C function: print
(2) global C function 'error'
(3) Lua global 'assert' at file 'core/main.lua:26'
	Local variables:
	 v = boolean: false
(4) Lua field 'run_loadgen' at file 'program/packetblaster/packetblaster.lua:43'
	Local variables:
	 c = table: 0x40d84af8  {links:table: 0x40d84b80, apps:table: 0x40d84b20}
	 patterns = table: 0x40d84e00  {1:02:00.0}
	 opts = table: 0x40d84bd0  {loop:true}
	 use_loadgen = boolean: true
	 nics = number: 0
(5) Lua field 'run' at file 'program/packetblaster/replay/replay.lua:52'
	Local variables:
	 args = table: 0x40d84e00  {1:02:00.0}
	 c = table: 0x40d84af8  {links:table: 0x40d84b80, apps:table: 0x40d84b20}
	 handlers = table: 0x40d84ba8  {D:function: 0x40d84c30, h:function: 0x40d84ca8, no-loop:function: 0x40d84cc8 (more...)}
	 opts = table: 0x40d84bd0  {loop:true}
	 filename = string: "/root/trill64.pcap"
(6) Lua field 'run' at file 'program/packetblaster/packetblaster.lua:75'
	Local variables:
	 args = table: 0x4071edf8  {1:/root/trill64.pcap, 2:02:00.0}
	 command = string: "replay"
	 modname = string: "program.packetblaster.replay.replay"
(7) Lua function 'main' at file 'core/main.lua:67' (best guess)
	Local variables:
	 program = string: "packetblaster"
	 args = table: 0x4071edf8  {1:/root/trill64.pcap, 2:02:00.0}
(8) global C function 'xpcall'
(9) main chunk of file 'core/main.lua' at line 230
(10)  C function 'require'
(11) global C function 'pcall'
(12) main chunk of file 'core/startup.lua' at line 3
(13) global C function 'require'
(14) main chunk of [string "require "core.startup""] at line 1
	nil

eugeneia, 2018-02-11:
I don’t think it does, see https://github.com/snabbco/snabb/blob/master/src/lib/hardware/pci.lua#L70 for a list of supported cards.

lukego, 2018-02-13:
The XL710 is a sore point at the moment. Everybody wants it, quite a few people are willing to put some work into the development, we have plenty of hardware in the lab, but nobody has taken the first step of adding the support yet.

The feeling seems to be that the hardware interface has been made gratuitously more complex since the 82599/X520 but that is life...