snabbco/snabb #1304: [raptorjit] Traceprof not compatible with raptorjit?

wingo, 2018-03-20:
The traceprof tests fail on Luke's `raptorjit-upstream` branch from https://github.com/snabbco/snabb/pull/1264:

```
$ make && sudo ./snabb snsh -t lib.traceprof.traceprof
make: 'snabb' is up to date.
lib/traceprof/traceprof.lua:53: module 'jit.dump' not found:
	no field package.preload['jit.dump']
	no file './jit/dump.so'
	no file '/usr/local/lib/lua/5.1/jit/dump.so'
	no file '/usr/local/lib/lua/5.1/loadall.so'
	no file './jit.so'
	no file '/usr/local/lib/lua/5.1/jit.so'
	no file '/usr/local/lib/lua/5.1/loadall.so'

Stack Traceback
===============
(1) Lua function 'handler' at file 'core/main.lua:166' (best guess)
	Local variables:
	 (*temporary) = string: "lib/traceprof/traceprof.lua:53: module 'jit.dump' not found:\
\9no field package.preload['jit.dump']\
\9no file './jit/dump.so'\
\9no file '/usr/local/lib/lua/5.1/jit/dump.so'\
\9no file '/usr/local/lib/lua/5.1/loadall.so'\
\9no file './jit.so'\
\9no file '/usr/local/lib/lua/5.1/jit.so'\
\9no file '/usr/local/lib/lua/5.1/loadall.so'"
	 (*temporary) = C function: print
	 (*temporary) = number: 6.92134e-310
(2) global C function 'require'
(3) main chunk of file 'lib/traceprof/traceprof.lua' at line 53
```

I guess the resolution will be to remove traceprof entirely, and rely on Studio or related tools to examine the auditlog instead.

lukego, 2018-03-22:
> I guess the resolution will be to remove traceprof entirely, and rely on Studio or related tools to examine the auditlog instead.

Yes. Traceprof is superseded by the newer auditlog/vmprofile features in RaptorJIT.

wingo, 2018-03-22:
Fixed in https://github.com/Igalia/snabb/pull/1035 and https://github.com/Igalia/snabb/pull/1036