snabbco/snabb #1307: [raptorjit] Occasional segfault running lib.ctable selftest

wingo, 2018-03-21:
With the raptorjit branch from #1306, I get occasional segfaults running the lib.ctable selftest.  They are not deterministic.  Here is one of them that I managed to catch in GDB:

```
Program received signal SIGSEGV, Segmentation fault.
0x0000000000452481 in lj_alloc_free (msp=0x7ffff7fcf010, ptr=<optimized out>) at lj_alloc.c:1266
1266		unlink_chunk(fm, next, nsize);
(gdb) bt
#0  0x0000000000452481 in lj_alloc_free (msp=0x7ffff7fcf010, ptr=<optimized out>) at lj_alloc.c:1266
#1  0x0000000000419bbd in gc_sweep (g=g@entry=0x7ffff7fcf3d8, p=0x7ffff7fe4b28, lim=35, lim@entry=40) at lj_gc.c:383
#2  0x000000000041a835 in gc_onestep (L=L@entry=0x7ffff7fcf378) at lj_gc.c:610
#3  0x000000000041af4d in lj_gc_step (L=L@entry=0x7ffff7fcf378) at lj_gc.c:659
#4  0x000000000045b4c0 in lj_cf_ffi_cast (L=0x7ffff7fcf378) at lib_ffi.c:542
#5  0x00000000004176c5 in lj_BC_FUNCC ()
#6  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
#7  0x00000000004176c5 in lj_BC_FUNCC ()
#8  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
#9  0x00000000004176c5 in lj_BC_FUNCC ()
#10 0x0000000000412dc8 in lua_pcall (L=<optimized out>, nargs=<optimized out>, nresults=<optimized out>, errfunc=<optimized out>) at lj_api.c:1074
#11 0x000000000040d413 in main ()
(gdb) bt full
#0  0x0000000000452481 in lj_alloc_free (msp=0x7ffff7fcf010, ptr=<optimized out>) at lj_alloc.c:1266
        F = 0x7ffff6ced1c8
        B = <optimized out>
        I = 0
        nsize = <optimized out>
        p = 0x7ffff6ced200
        fm = 0x7ffff7fcf010
        psize = 80
        next = 0x7ffff6ced250
#1  0x0000000000419bbd in gc_sweep (g=g@entry=0x7ffff7fcf3d8, p=0x7ffff7fe4b28, lim=35, lim@entry=40) at lj_gc.c:383
        ow = 33
        o = <optimized out>
#2  0x000000000041a835 in gc_onestep (L=L@entry=0x7ffff7fcf378) at lj_gc.c:610
        old = 173319182
        g = 0x7ffff7fcf3d8
#3  0x000000000041af4d in lj_gc_step (L=L@entry=0x7ffff7fcf378) at lj_gc.c:659
        g = 0x7ffff7fcf3d8
        lim = 1600
        ostate = -2
#4  0x000000000045b4c0 in lj_cf_ffi_cast (L=0x7ffff7fcf378) at lib_ffi.c:542
        id = 9
        o = 0x7ffff6c90a50
#5  0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#6  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
        name = 0x7ffff7fd9168 "core.main"
        i = 1
#7  0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#8  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
        name = 0x7ffff7fd8e58 "core.startup"
        i = 1
#9  0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#10 0x0000000000412dc8 in lua_pcall (L=<optimized out>, nargs=<optimized out>, nresults=<optimized out>, errfunc=<optimized out>) at lj_api.c:1074
        g = 0x7ffff7fcf3d8
        oldh = 0 '\000'
        ef = <optimized out>
        status = <optimized out>
#11 0x000000000040d413 in main ()
No symbol table info available.
```

To reproduce, I suggest running `while sudo ./snabb snsh -t lib.ctable; do true; done`.  It takes a nondeterminate amount of runs for it to repro.  I have reproduced this error on multiple machines.

lukego, 2018-03-21:
Awesome with these bug reports! I'm 1-2 days away from being back in the saddle to help chase them down.

wingo, 2018-03-21:
Interestingly, the other times I manage to get the thing to crash, the backtrace is pretty much the same:

```
(gdb) bt
#0  0x0000000000452481 in lj_alloc_free (msp=0x7ffff7fcf010, ptr=<optimized out>) at lj_alloc.c:1266
#1  0x0000000000419bbd in gc_sweep (g=g@entry=0x7ffff7fcf3d8, p=0x7ffff6b90238, lim=25, lim@entry=40) at lj_gc.c:383
#2  0x000000000041a835 in gc_onestep (L=L@entry=0x7ffff7fcf378) at lj_gc.c:610
#3  0x000000000041af4d in lj_gc_step (L=L@entry=0x7ffff7fcf378) at lj_gc.c:659
#4  0x000000000045b4c0 in lj_cf_ffi_cast (L=0x7ffff7fcf378) at lib_ffi.c:542
#5  0x00000000004176c5 in lj_BC_FUNCC ()
#6  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
#7  0x00000000004176c5 in lj_BC_FUNCC ()
#8  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
#9  0x00000000004176c5 in lj_BC_FUNCC ()
#10 0x0000000000412dc8 in lua_pcall (L=<optimized out>, nargs=<optimized out>, nresults=<optimized out>, errfunc=<optimized out>) at lj_api.c:1074
#11 0x000000000040d413 in main ()
(gdb) bt full
#0  0x0000000000452481 in lj_alloc_free (msp=0x7ffff7fcf010, ptr=<optimized out>) at lj_alloc.c:1266
        F = 0x7ffff6ced1c8
        B = <optimized out>
        I = 0
        nsize = <optimized out>
        p = 0x7ffff6ced200
        fm = 0x7ffff7fcf010
        psize = 80
        next = 0x7ffff6ced250
#1  0x0000000000419bbd in gc_sweep (g=g@entry=0x7ffff7fcf3d8, p=0x7ffff6b90238, lim=25, lim@entry=40) at lj_gc.c:383
        ow = 33
        o = <optimized out>
#2  0x000000000041a835 in gc_onestep (L=L@entry=0x7ffff7fcf378) at lj_gc.c:610
        old = 171146182
        g = 0x7ffff7fcf3d8
#3  0x000000000041af4d in lj_gc_step (L=L@entry=0x7ffff7fcf378) at lj_gc.c:659
        g = 0x7ffff7fcf3d8
        lim = 1600
        ostate = -2
#4  0x000000000045b4c0 in lj_cf_ffi_cast (L=0x7ffff7fcf378) at lib_ffi.c:542
        id = 9
        o = 0x7ffff6c90a58
#5  0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#6  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
        name = 0x7ffff7fd9168 "core.main"
        i = 1
#7  0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#8  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
        name = 0x7ffff7fd8e58 "core.startup"
        i = 1
#9  0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#10 0x0000000000412dc8 in lua_pcall (L=<optimized out>, nargs=<optimized out>, nresults=<optimized out>, errfunc=<optimized out>) at lj_api.c:1074
        g = 0x7ffff7fcf3d8
        oldh = 0 '\000'
        ef = <optimized out>
        status = <optimized out>
#11 0x000000000040d413 in main ()
```

This second run was from re-running in the same GDB session; perhaps ASLR doesn't apply to multiple runs within one GDB session?  I just wonder because the addresses are pretty much the same.  Slight difference on the limit in frame 1 but the structure is the same, the outer addresses are the same, and the address on the innermost frame is the same.  I had to re-run 20 or 30 times to get this failure though.  Will try again in separate GDB sessions.

wingo, 2018-03-21:
Nope, seems consistent, even when running in separate GDB instances.

wingo, 2018-03-21:
Replacing lj_alloc.c's

```c
#define LJ_ALLOC_MMAP_PROBE	1
```
with
```c
#define LJ_ALLOC_MMAP32	1
#define LJ_ALLOC_MMAP_PROBE	0
```

does not fix the problem, therefore I conclude that the problem is not related to the block allocator rewrite from 2016.

wingo, 2018-03-21:
Removing the `ffi.gc` call in `lib.ctable` doesn't fix it either :/

wingo, 2018-03-21:
I have a reliable reproducer!  Drop into `test.lua` and run as `snabb snsh test.lua`:

```
local ctable = require('lib.ctable')
local ffi = require('ffi')
local bit = require('bit')

print("selftest: ctable")

local occupancy = 1e5
local params = {
   key_type = ffi.typeof('uint32_t[1]'),
   value_type = ffi.typeof('int32_t[1]'),
}
local ctab = ctable.new(params)

-- Fill with {i} -> { bnot(i) }.
local k = ffi.new('uint32_t[1]');
local v = ffi.new('int32_t[1]');
for i = 1,occupancy do
   k[0] = i
   for j=0,0 do v[j] = bit.bnot(i) end
   ctab:add(k, v)
end

collectgarbage()

print('hey!!!')

-- Save the table out to disk, reload it, and run the same
-- checks.
local tmp = os.tmpname()
do
   local file = io.open(tmp, 'wb')
   local function write(ptr, size)
      local str = ffi.string(ptr, size)
      file:write(ffi.string(ptr, size))
   end
   local stream = {}
   function stream:write_ptr(ptr, type)
      assert(ffi.sizeof(ptr) == ffi.sizeof(type))
      write(ptr, ffi.sizeof(type))
   end
   function stream:write_array(ptr, type, count)
      write(ptr, ffi.sizeof(type) * count)
   end
   ctab:save(stream)
   file:close()
end
do
   local file = io.open(tmp, 'rb')
   -- keep references to avoid GCing too early
   local handle = {}
   local function read(size)
      local buf = ffi.new('uint8_t[?]', size, file:read(size))
      table.insert(handle, buf)
      return buf
   end
   local stream = {}
   function stream:read_ptr(type)
      return ffi.cast(ffi.typeof('$*', type), read(ffi.sizeof(type)))
   end
   function stream:read_array(type, count)
      return ffi.cast(ffi.typeof('$*', type),
                      read(ffi.sizeof(type) * count))
   end
   ctab = ctable.load(stream, params)
   file:close()
end         
os.remove(tmp)

print('collecting...')
collectgarbage()
print('collected.')
```

It will crash at the last `collectgarbage()` call with the backtrace as above.

wingo, 2018-03-21:
Actually excuse me, it doesn't have the same backtrace because this time it's an explicit `collectgarbage` call that initiates the sweep instead of a `ffi.cast` call.  Backtrace:

```
(gdb) bt full
#0  0x0000000000452481 in lj_alloc_free (msp=0x7ffff7fcf010, ptr=<optimized out>) at lj_alloc.c:1266
        F = 0x7ffff6ced1c8
        B = <optimized out>
        I = 0
        nsize = <optimized out>
        p = 0x7ffff6ced200
        fm = 0x7ffff7fcf010
        psize = 80
        next = 0x7ffff6ced250
#1  0x0000000000419bbd in gc_sweep (g=g@entry=0x7ffff7fcf3d8, p=0x7ffff6dcc950, lim=13, lim@entry=40) at lj_gc.c:383
        ow = 33
        o = <optimized out>
#2  0x000000000041a835 in gc_onestep (L=L@entry=0x7ffff7fcf378) at lj_gc.c:610
        old = 17830369
        g = 0x7ffff7fcf3d8
#3  0x000000000041b128 in lj_gc_fullgc (L=0x7ffff7fcf378) at lj_gc.c:717
        g = 0x7ffff7fcf3d8
        ostate = -2
#4  0x000000000041311d in lua_gc (L=L@entry=0x7ffff7fcf378, what=what@entry=2, data=<optimized out>) at lj_api.c:1183
        g = 0x7ffff7fcf3d8
        res = 0
#5  0x0000000000453202 in lj_cf_collectgarbage (L=0x7ffff7fcf378) at lib_base.c:426
        res = <optimized out>
        opt = 2
        data = <optimized out>
#6  0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#7  0x00000000004532e8 in lj_cf_dofile (L=0x7ffff7fcf378) at lib_base.c:406
        fname = <optimized out>
#8  0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#9  0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
        name = 0x7ffff7fd9168 "core.main"
        i = 1
#10 0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#11 0x000000000045897a in lj_cf_package_require (L=0x7ffff7fcf378) at lib_package.c:322
        name = 0x7ffff7fd8e58 "core.startup"
        i = 1
#12 0x00000000004176c5 in lj_BC_FUNCC ()
No symbol table info available.
#13 0x0000000000412dc8 in lua_pcall (L=<optimized out>, nargs=<optimized out>, nresults=<optimized out>, errfunc=<optimized out>) at lj_api.c:1074
        g = 0x7ffff7fcf3d8
        oldh = 0 '\000'
        ef = <optimized out>
        status = <optimized out>
---Type <return> to continue, or q <return> to quit---
#14 0x000000000040d413 in main ()
No symbol table info available.
```

There is a bit of nonderminacy in the test because the ctable's hash function is randomized, but I get the same crash every time.

wingo, 2018-03-21:
Slightly shorter reproducer.
```lua
local ctable = require('lib.ctable')
local ffi = require('ffi')
local bit = require('bit')

local params = {
   key_type = ffi.typeof('uint32_t[1]'),
   value_type = ffi.typeof('int32_t[1]'),
}
print('creating empty ctable')
local ctab = ctable.new(params)

print('first gc')
collectgarbage()

print('writing ctable out to a file')
local tmp = os.tmpname()
local file = io.open(tmp, 'wb')
local function write(ptr, size)
   local str = ffi.string(ptr, size)
   file:write(ffi.string(ptr, size))
end
local stream = {}
function stream:write_ptr(ptr, type)
   assert(ffi.sizeof(ptr) == ffi.sizeof(type))
   write(ptr, ffi.sizeof(type))
end
function stream:write_array(ptr, type, count)
   write(ptr, ffi.sizeof(type) * count)
end
ctab:save(stream)
file:close()
print('reading ctable back in from file')
local file = io.open(tmp, 'rb')
local anchor = {}
local function read(size)
   local buf = ffi.new('uint8_t[?]', size, file:read(size))
   table.insert(anchor, buf)
   return buf
end
local stream = {}
function stream:read_ptr(type)
   return ffi.cast(ffi.typeof('$*', type), read(ffi.sizeof(type)))
end
function stream:read_array(type, count)
   return ffi.cast(ffi.typeof('$*', type),
                   read(ffi.sizeof(type) * count))
end
ctab = ctable.load(stream, params)
file:close()
os.remove(tmp)

print('clearing anchor')
-- If you comment out this line, the test passes.
anchor = nil

print('final collection...')
collectgarbage()
print('success.')
```

I am still a bit perplexed.

wingo, 2018-03-21:
Got a better example that doesn't depend on anything outside LuaJIT!

```lua
local ffi = require('ffi')

print('creating data')
local byte_t = ffi.typeof('uint8_t')
local byte_count = 72
local data = ffi.new('uint8_t[?]', byte_count)

print('making string out of data')
data_str = ffi.string(data, byte_count)

print('loading data from string')
local anchor = {}
local function read(size)
   assert(size == #data_str)
   local buf = ffi.new('uint8_t[?]', #data_str, data_str)
   table.insert(anchor, buf)
   return buf
end
local function read_array(type, count)
   return ffi.cast(ffi.typeof('$*', type),
                   read(ffi.sizeof(type) * count))
end
local data2 = read_array(byte_t, byte_count)

print('clearing anchor')
-- If you comment out this line, the test passes.
anchor = nil

print('final collection...')
collectgarbage()
print('success.')
```

wingo, 2018-03-21:
Well -- I can't get it to repro just running raptorjit.  Have to use snabb snsh currently.

wingo, 2018-03-21:
This repro case is nice because there will be no JIT compilation at all.  Here's as short as I think I can get it:

```lua
local ffi = require('ffi')

-- 72 bytes.
local data_str =
   '012345678901234567890123456789012345678901234567890123456789012345678901'

print('loading data from string')
local anchor = {}
local function read_bytes()
   local buf = ffi.new('uint8_t[?]', #data_str, data_str)
   table.insert(anchor, buf)
   return ffi.cast('uint8_t*', buf)
end
local data2 = read_bytes()

print('clearing anchor')
-- If you comment out this line, the test passes.
anchor = nil

print('running gc...')
collectgarbage()

print('success.')
```

wingo, 2018-03-21:
LOL, even smaller:

```lua
local ffi = require('ffi')

-- 72 bytes.
local data_str =
   '012345678901234567890123456789012345678901234567890123456789012345678901'

local buf = ffi.new('uint8_t[?]', #data_str, data_str)
ffi.cast('uint8_t*', buf)
-- If you comment out this line, the test passes.
buf = nil

print('running gc...')
collectgarbage()
print('success.')
```