snabbco/snabb #1312: Snabb lwAFTR is not starting, because there is bad argument #1 to 'require' (string expected, got nil) in file program/lwaftr/setup.lua in line 233

sebastianczech, 2018-03-26:
I am trying to start snabb lwAFTR using commnad:

```
# sudo ./snabb lwaftr run --conf program/lwaftr/tests/data/icmp_on_fail.conf --v4 0000:03:00.0 --v6 0000:0b:00.0
```

Currenty on machine I have 2 ethernet controllers, which I used to start lwAFTR:

```
# lspci | grep -i ether
03:00.0 Ethernet controller: VMware VMXNET3 Ethernet Controller (rev 01)
0b:00.0 Ethernet controller: VMware VMXNET3 Ethernet Controller (rev 01)
```

Unfortunately during starting I am receving error:

```
program/lwaftr/setup.lua:233: bad argument #1 to 'require' (string expected, got nil)

Stack Traceback
===============
(1) Lua function 'handler' at file 'core/main.lua:168' (best guess)
        Local variables:
         reason = string: "program/lwaftr/setup.lua:233: bad argument #1 to 'require' (string expected, got nil)"
         (*temporary) = C function: print
(2) global C function 'require'
(3) Lua upvalue 'f' at file 'program/lwaftr/setup.lua:233'
        Local variables:
         c = table: 0x4122c530  {links:table: 0x4122c5b8, apps:table: 0x4122c558}
         conf = table: 0x41220358  {softwire_config:table: 0x41220768}
         v4_nic_name = string: "inetNic"
         v6_nic_name = string: "b4sideNic"
         ring_buffer_size = nil
         v4_pci = string: "0000:03:00.0"
         id = number: 0
         queue = table: 0x41228e90  {external_interface:table: 0x41229038, internal_interface:table: 0x41229c38 (more...)}
         v6_pci = string: "0000:0b:00.0"
         v4_info = table: 0x4122c900  {pciaddress:03:00.0, vendor:0x15ad, usable:no, device:0x07b0}
         v6_info = table: 0x4122caa0  {pciaddress:0b:00.0, vendor:0x15ad, usable:no, device:0x07b0}
         (*temporary) = Lua function 'app' (defined at line 27 of chunk core/config.lua)
         (*temporary) = table: 0x4122c530  {links:table: 0x4122c5b8, apps:table: 0x4122c558}
         (*temporary) = string: "inetNic"
(4) Lua field 'setup_fn' at file 'program/lwaftr/setup.lua:597'
        Local variables:
         conf = table: 0x4225a650  {softwire_config:table: 0x4225a9a8}
         worker_app_graphs = table: 0x401e2cd8  {}
         (for generator) = C function: next
         (for state) = table: 0x401e2d00  {0000:03:00.0/0:table: 0x41220358}
         (for control) = number: nan
         worker_id = string: "0000:03:00.0/0"
         worker_config = table: 0x41220358  {softwire_config:table: 0x41220768}
         app_graph = table: 0x4122c530  {links:table: 0x4122c5b8, apps:table: 0x4122c558}
(5) Lua method 'set_initial_configuration' at file 'lib/ptree/ptree.lua:145'
        Local variables:
         self = table: 0x41a7afd8  {rpc_handler:function: 0x401e2c00, setup_fn:function: 0x41a7a828, support:table: 0x40fc1f18 (more...)}
         configuration = table: 0x4225a650  {softwire_config:table: 0x4225a9a8}
(6) Lua field 'ptree_manager' at file 'lib/ptree/ptree.lua:97'
        Local variables:
         conf = table: 0x41a7a870  {setup_fn:function: 0x41a7a828, worker_default_scheduling:table: 0x405c0a48 (more...)}
         conf = table: 0x41a7a960  {setup_fn:function: 0x41a7a828, socket_file_name:config-leader-socket, worker_default_scheduling:table: 0x405c0a48 (more...)}
         ret = table: 0x41a7afd8  {rpc_handler:function: 0x401e2c00, setup_fn:function: 0x41a7a828, support:table: 0x40fc1f18 (more...)}
(7) Lua field 'run' at file 'program/lwaftr/run/run.lua:178'
        Local variables:
         args = table: 0x4042d350  {1:--conf, 2:program/lwaftr/tests/data/icmp_on_fail.conf, 3:--v4, 4:0000:03:00.0 (more...)}
         opts = table: 0x41a16d30  {verbosity:0}
         scheduling = table: 0x405c0a48  {ingress_drop_monitor:flush}
         conf_file = string: "program/lwaftr/tests/data/icmp_on_fail.conf"
         v4 = string: "0000:03:00.0"
         v6 = string: "0000:0b:00.0"
         conf = table: 0x4225a650  {softwire_config:table: 0x4225a9a8}
         setup_fn = Lua function 'setup' (defined at line 153 of chunk program/lwaftr/run/run.lua)
         manager_opts = table: 0x41a7a7c8  {worker_default_scheduling:table: 0x405c0a48}
(8) Lua field 'run' at file 'program/lwaftr/lwaftr.lua:25'
        Local variables:
         args = table: 0x4042d350  {1:--conf, 2:program/lwaftr/tests/data/icmp_on_fail.conf, 3:--v4, 4:0000:03:00.0 (more...)}
         command = string: "run"
         modname = string: "program.lwaftr.run.run"
(9) Lua function 'main' at file 'core/main.lua:67' (best guess)
        Local variables:
         program = string: "lwaftr"
         args = table: 0x4042d350  {1:--conf, 2:program/lwaftr/tests/data/icmp_on_fail.conf, 3:--v4, 4:0000:03:00.0 (more...)}
(10) global C function 'xpcall'
(11) main chunk of file 'core/main.lua' at line 236
(12)  C function 'require'
(13) global C function 'pcall'
(14) main chunk of file 'core/startup.lua' at line 3
(15) global C function 'require'
(16) main chunk of [string "require "core.startup""] at line 1
        nil
```

Am I using correct command to start lwAFTR ? What could the cause of the problem in this situation ?

wingo, 2018-04-03:
Hi!  This is a terrible error message :P  Thanks for reporting.

Unfortunately it does correspond to an error situation -- Snabb does not implement drivers for these devices:

```
# lspci | grep -i ether
03:00.0 Ethernet controller: VMware VMXNET3 Ethernet Controller (rev 01)
0b:00.0 Ethernet controller: VMware VMXNET3 Ethernet Controller (rev 01)
```

Currently the lwAFTR targets only those devices that have native Snabb drivers, mainly the Intel 82599 / i210 / i350 cards.  For evaluation purposes, we should probably extend it to support tap interfaces, which would allow you to evaluate the lwAFTR, though not at high performance.

In the long term if you are interested in running the lwAFTR or some other Snabb network function on VMware devices, the right thing to do is to write a Snabb driver.  It's not so bad -- about 1000 lines or so of code, probably a bit less considering it's a virtual interface already.
It doesn't know how to use Tap inter
I suggest for the time being that you use the TAP drivers

sebastianczech, 2018-04-03:
Thank you for the response. 

I have also tried to start lwAFTR after changing in VMware network interfaces to:

```
# lspci | grep -i ethernet
02:01.0 Ethernet controller: Intel Corporation 82545EM Gigabit Ethernet Controller (Copper) (rev 01)
02:02.0 Ethernet controller: Intel Corporation 82545EM Gigabit Ethernet Controller (Copper) (rev 01)
```

but I doesn't help. In this case it is e1000 driver:

```
# lshw -class network | grep -i driver
       configuration: autonegotiation=on broadcast=yes driver=e1000 driverversion=7.3.21-k8-NAPI duplex=full ip=10.0.0.85 latency=0 link=yes mingnt=255 multicast=yes port=twisted pair speed=1Gbit/s
       configuration: autonegotiation=on broadcast=yes driver=e1000 driverversion=7.3.21-k8-NAPI duplex=full latency=0 link=no mingnt=255 multicast=yes port=twisted pair speed=1Gbit/s
```

dpino, 2018-06-01:
Tap support on the lwAFTR has recently landed. What this means is that the lwAFTR can run now on any Linux kernel network interface, but it won't be performant which makes it unusable in any practical sense. You can try it out by pulling https://github.com/Igalia/snabb/ or wait until Snabb's July release when likely these changes will get merged.

closed 2023-03-12 by sebastianczech