snabbco/snabb #1365: intel_mp not filtering multicast pause frames

alexandergall, 2018-06-29:
I have observed that the `intel_mp` app delivers pause frames sent to the well-known multicast address `01:80:c2:00:00:01` (https://en.wikipedia.org/wiki/Ethernet_flow_control) like regular frames. Pause frames sent to the unicast MAC address of the NIC are handled correctly by the hardware.

In my test case, the problem was solved by disabling TX flow-control on the neighboring system (`ethtool -A <dev> tx off`).

eugeneia, 2018-07-02:
I believe there are registers in the NICs we support that govern this behavior. If we agree on an excpected behavior (i.e., not forwarding these, even if in promiscous mode / OR: add VMDq-less non-promiscous mode support) I could take a shot at this in https://github.com/snabbco/snabb/pull/1353 or a follow-up.

alexandergall, 2018-07-02:
The statement about pause frames sent the unicast address is probably wrong. The issue is whether the NIC is supposed to pass MAC control frames to the host or not, which is controlled by the `DPF` bit of the `MFLCN` register. This bit should be set by the driver by default when FC is enabled.