snabbco/snabb #1391: Snabb lwAFTR not starting. apps/intel_mp/intel_mp.lua:642: attempt to call field 'RTTDQSEL' (a nil value)

ok1cyc, 2018-10-26:
When starting lwAFTR with
` ./snabb lwaftr run --cpu 3   --conf ../conf/lwaftr.conf --name lwaftr`
I'm getting following error:
```
 ../conf/lwaftr.conf: loading compiled configuration from ../conf/lwaftr.o
 ../conf/lwaftr.conf: compiled configuration is up to date.
 Binding data-plane PID 1298 to CPU 3.
 Bound main process to NUMA node: 0 (CPU 0)
 hugetlb mmap failed (Cannot allocate memory), falling back.
 hugetlb mmap failed (Cannot allocate memory), falling back.
 [mounting /var/run/snabb/hugetlbfs]
 apps/intel_mp/intel_mp.lua:642: attempt to call field 'RTTDQSEL' (a nil value)
 line not found
 
 Stack Traceback
 ===============
 (1) Lua field 'RTTDQSEL' at file 'core/main.lua:168'
         Local variables:
          reason = string: "apps/intel_mp/intel_mp.lua:642: attempt to call field 'RTTDQSEL' (a nil value)"
          (*temporary) = C function: print
 (2) Lua method 'init_tx_q' at file 'apps/intel_mp/intel_mp.lua:642'
         Local variables:
          self = table: 0x4251d6f8  {shm_root:/intel-mp/01:00.0/, pciaddress:0000:01:00.0, rate_limit:0, vmdq:true (more...)}
          (*temporary) = nil
          (*temporary) = number: 0
          (*temporary) = number: 32768
          (*temporary) = number: 32768
          (*temporary) = TDLEN[0000e008]:00008000  {longname:Tx Descriptor Ring Length, offset:57352, name:TDLEN, ptr:cdata<unsigned int *>: 0x7f5df5afa008 (more...)}
          (*temporary) = string: "attempt to call field 'RTTDQSEL' (a nil value)"
 (3) Lua method 'new' at file 'apps/intel_mp/intel_mp.lua:408'
         Local variables:
          self = table: 0x40f88530  {rss_tab:function: 0x40f89a78, transmit:function: 0x40f89c60, rss_tab_build:function: 0x40f89a98 (more...)}
          conf = table: 0x40f8e248  {rate_limit:0, vmdq:true, pciaddr:0000:01:00.0, poolnum:0, ring_buffer_size:2048 (more...)}
          self = table: 0x4251d6f8  {shm_root:/intel-mp/01:00.0/, pciaddress:0000:01:00.0, rate_limit:0, vmdq:true (more...)}
          vendor = string: "0x8086"
          device = string: "0x1533"
          byid = table: 0x40f89340  {driver:table: 0x40f88be8, registers:i210, max_q:4}
 (4) Lua function 'ops' at file 'core/app.lua:342' (best guess)
         Local variables:
          name = string: "nic"
          class = table: 0x40f88530  {rss_tab:function: 0x40f89a78, transmit:function: 0x40f89c60, rss_tab_build:function: 0x40f89a98 (more...)}
          arg = table: 0x40f8e248  {rate_limit:0, vmdq:true, pciaddr:0000:01:00.0, poolnum:0, ring_buffer_size:2048 (more...)}
 (5) Lua field 'apply_config_actions' at file 'core/app.lua:369'
         Local variables:
          actions = table: 0x4012d668  {1:table: 0x40268898, 2:table: 0x4026fa68, 3:table: 0x40261ae8, 4:table: 0x41296148 (more...)}
          ops = table: 0x4012e200  {unlink_output:function: 0x4012e248, stop_app:function: 0x4012e3c8, free_link:function: 0x4012e2e0 (more...)}
          remove_link_from_array = Lua function 'remove' (defined at line 289 of chunk core/app.lua)
          (for generator) = C function: builtin#6
          (for state) = table: 0x4012d668  {1:table: 0x40268898, 2:table: 0x4026fa68, 3:table: 0x40261ae8, 4:table: 0x41296148 (more...)}
          (for control) = number: 10
          _ = number: 10
          action = table: 0x40e2ced0  {1:start_app, 2:table: 0x40f924c0}
          name = string: "start_app"
          args = table: 0x40f924c0  {1:nic, 2:table: 0x40f88530, 3:table: 0x40f8e248}
 (6) Lua method 'commit_pending_actions' at file 'lib/ptree/worker.lua:77'
         Local variables:
          self = table: 0x4025fd30  {no_report:false, channel:table: 0x4025fe30, duration:inf, period:0.001 (more...)}
          to_apply = table: 0x4012d668  {1:table: 0x40268898, 2:table: 0x4026fa68, 3:table: 0x40261ae8, 4:table: 0x41296148 (more...)}
          should_flush = boolean: true
 (7) Lua method 'handle_actions_from_manager' at file 'lib/ptree/worker.lua:89'
         Local variables:
          self = table: 0x4025fd30  {no_report:false, channel:table: 0x4025fe30, duration:inf, period:0.001 (more...)}
          channel = table: 0x4025fe30  {ring_buffer:cdata<struct 1509 *>: 0x7f5e2e16f000}
          (for index) = number: 1
          (for limit) = number: 4
          (for step) = number: 1
          i = number: 1
          len = number: 4
          action = table: 0x4012d5e8  {1:commit, 2:table: 0x4012d628}
 (8) Lua field 'main' at file 'lib/ptree/worker.lua:104'
         Local variables:
          self = table: 0x4025fd30  {no_report:false, channel:table: 0x4025fe30, duration:inf, period:0.001 (more...)}
          stop = number: inf
          next_time = number: 129.571
 (9) main chunk of [string "require('lib.scheduling').apply({["cpu"]=3,["..."] at line 2
 (10) Lua function 'main' at file 'core/main.lua:58' (best guess)
         Local variables:
          expr = string: "require('lib.scheduling').apply({[\"cpu\"]=3,[\"busywait\"]=true,[\"ingress_drop_monitor\"]=\"flush\"})\
 require('lib.ptree.worker').main()"
          f = Lua function '?' (defined at line 0 of chunk "require('lib.scheduling').apply({["cpu"]=3,["..."])
 (11) global C function 'xpcall'
 (12) main chunk of file 'core/main.lua' at line 242
 (13)  C function 'require'
 (14) global C function 'pcall'
 (15) main chunk of file 'core/startup.lua' at line 3
 (16) global C function 'require'
 (17) main chunk of [string "require "core.startup""] at line 1
         nil
 Error while running fiber: core/shm.lua:37: shm open error (1298/apps/fragmenterv4/out-ipv4-frag.counter):No such file or directory
```
Currently using:
```
root@linux:/opt/snabb/src# lshw -c net
  *-network UNCLAIMED
       description: Ethernet controller
       product: I210 Gigabit Network Connection
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:01:00.0
       version: 03
       width: 32 bits
       clock: 33MHz
       capabilities: pm msi msix pciexpress bus_master cap_list
       configuration: latency=0
       resources: memory:91300000-913fffff memory:91400000-91403fff memory:91200000-912fffff
```
As understood i210 should be supported by intel_mp. Could the issue link to VMDq not supported on i210? Any help appreciated. Thanks.

dpino, 2018-11-06:
The lwAFTR uses vMDQ by default. The reason is that according to the Intel 82599 datasheet, VLAN tag addition and removing by hardware is available only if vMDQ is enabled.

If vMDQ is not supported by i210 that is the likely cause for the fail (RTTDQSEL register not available in i210 but used when vMDQ is enabled).

Perhaps you can work around the issue by disabling vMDQ in https://github.com/snabbco/snabb/blob/master/src/program/lwaftr/setup.lua#L217 and https://github.com/snabbco/snabb/blob/master/src/program/lwaftr/setup.lua#L227. If that fixes the issue, we can take a deeper look and come up with a better patch.

cc @takikawa

ok1cyc, 2018-11-14:
Thanks Diego for the advice. I disabled vMDQ as you suggested but hit the other issue:

> root@linux:/opt/snabb/src#  ./snabb lwaftr run -v --name lwaftr --conf ../conf/lwaftr.conf --on-a-stick 0000:01:00.0 --cpu 3
> ../conf/lwaftr.conf: loading compiled configuration from ../conf/lwaftr.o
> ../conf/lwaftr.conf: compiled configuration is out of date; recompiling.
> ../conf/lwaftr.conf: loading source configuration
> ../conf/lwaftr.conf: wrote compiled configuration ../conf/lwaftr.o
> Bound main process to NUMA node: 0 (CPU 0)
> Binding data-plane PID 7524 to CPU 3.
> hugetlb mmap failed (Cannot allocate memory), falling back.
> hugetlb mmap failed (Cannot allocate memory), falling back.
> apps/intel_mp/intel_mp.lua:964: attempt to index field 'PFVFSPOOF' (a nil value)
> line not found
> 
> Stack Traceback
> ===============
> (1) Lua metamethod '__index' at file 'core/main.lua:168'
>         Local variables:
>          reason = string: "apps/intel_mp/intel_mp.lua:964: attempt to index field 'PFVFSPOOF' (a nil value)"
>          (*temporary) = C function: print
> (2) Lua method 'set_transmit_MAC' at file 'apps/intel_mp/intel_mp.lua:964'
>         Local variables:
>          self = table: 0x406089c0  {shm_root:/intel-mp/01:00.0/, pciaddress:0000:01:00.0, rate_limit:0, vmdq:false (more...)}
>          poolnum = number: 1
>          (*temporary) = nil
>          (*temporary) = number: 0
>          (*temporary) = number: 0.125
>          (*temporary) = number: 279.374
>          (*temporary) = RAH[01][0000540c]:80080291  {longname:Receive Address High, offset:21516, name:RAH[01], ptr:cdata<unsigned int *>: 0x7f92c7ea440c (more...)}
>          (*temporary) = string: "attempt to index field 'PFVFSPOOF' (a nil value)"
> (3) Lua method 'set_MAC' at file 'apps/intel_mp/intel_mp.lua:926'
>         Local variables:
>          self = table: 0x406089c0  {shm_root:/intel-mp/01:00.0/, pciaddress:0000:01:00.0, rate_limit:0, vmdq:false (more...)}
> (4) Lua method 'new' at file 'apps/intel_mp/intel_mp.lua:410'
>         Local variables:
>          self = table: 0x40725fc8  {rss_tab:function: 0x40727510, transmit:function: 0x407276f8, rss_tab_build:function: 0x40727530 (more...)}
>          conf = table: 0x4072cae0  {rate_limit:0, vmdq:false, pciaddr:0000:01:00.0, poolnum:1, ring_buffer_size:2048 (more...)}
>          self = table: 0x406089c0  {shm_root:/intel-mp/01:00.0/, pciaddress:0000:01:00.0, rate_limit:0, vmdq:false (more...)}
>          vendor = string: "0x8086"
>          device = string: "0x1533"
>          byid = table: 0x40726dd8  {driver:table: 0x40726680, registers:i210, max_q:4}
> (5) Lua function 'ops' at file 'core/app.lua:348' (best guess)
>         Local variables:
>          name = string: "b4sideNic"
>          class = table: 0x40725fc8  {rss_tab:function: 0x40727510, transmit:function: 0x407276f8, rss_tab_build:function: 0x40727530 (more...)}
>          arg = table: 0x4072cae0  {rate_limit:0, vmdq:false, pciaddr:0000:01:00.0, poolnum:1, ring_buffer_size:2048 (more...)}
> (6) Lua field 'apply_config_actions' at file 'core/app.lua:375'
>         Local variables:
>          actions = table: 0x411c47c8  {1:table: 0x4041ca20, 2:table: 0x40424120, 3:table: 0x40d56a78, 4:table: 0x40d66b80 (more...)}
>          ops = table: 0x411c4998  {unlink_output:function: 0x411c49e0, stop_app:function: 0x411c5190, free_link:function: 0x411c4a78 (more...)}
>          remove_link_from_array = Lua function 'remove' (defined at line 295 of chunk core/app.lua)
>          (for generator) = C function: builtin#6
>          (for state) = table: 0x411c47c8  {1:table: 0x4041ca20, 2:table: 0x40424120, 3:table: 0x40d56a78, 4:table: 0x40d66b80 (more...)}
>          (for control) = number: 9
>          _ = number: 9
>          action = table: 0x40f4d970  {1:start_app, 2:table: 0x40f35858}
>          name = string: "start_app"
>          args = table: 0x40f35858  {1:b4sideNic, 2:table: 0x40725fc8, 3:table: 0x4072cae0}
> (7) Lua method 'commit_pending_actions' at file 'lib/ptree/worker.lua:77'
>         Local variables:
>          self = table: 0x404140e0  {no_report:false, channel:table: 0x404141e0, duration:inf, period:0.001 (more...)}
>          to_apply = table: 0x411c47c8  {1:table: 0x4041ca20, 2:table: 0x40424120, 3:table: 0x40d56a78, 4:table: 0x40d66b80 (more...)}
>          should_flush = boolean: true
> (8) Lua method 'handle_actions_from_manager' at file 'lib/ptree/worker.lua:89'
>         Local variables:
>          self = table: 0x404140e0  {no_report:false, channel:table: 0x404141e0, duration:inf, period:0.001 (more...)}
>          channel = table: 0x404141e0  {ring_buffer:cdata<struct 1512 *>: 0x7f931fcd6000}
>          (for index) = number: 3
>          (for limit) = number: 4
>          (for step) = number: 1
>          i = number: 3
>          len = number: 4
>          action = table: 0x411c4748  {1:commit, 2:table: 0x411c4788}
> (9) Lua field 'main' at file 'lib/ptree/worker.lua:104'
>         Local variables:
>          self = table: 0x404140e0  {no_report:false, channel:table: 0x404141e0, duration:inf, period:0.001 (more...)}
>          stop = number: inf
>          next_time = number: 1570.39
> (10) main chunk of [string "require('lib.scheduling').apply({["cpu"]=3,["..."] at line 2
> (11) Lua function 'main' at file 'core/main.lua:58' (best guess)
>         Local variables:
>          expr = string: "require('lib.scheduling').apply({[\"cpu\"]=3,[\"busywait\"]=true,[\"ingress_drop_monitor\"]=\"flush\"})\
[... cut ...]
>          f = Lua function '?' (defined at line 0 of chunk "require('lib.scheduling').apply({["cpu"]=3,["..."])
> (12) global C function 'xpcall'
> (13) main chunk of file 'core/main.lua' at line 242
> (14)  C function 'require'
> (15) global C function 'pcall'
> (16) main chunk of file 'core/startup.lua' at line 3
> (17) global C function 'require'
> (18) main chunk of [string "require "core.startup""] at line 1
>         nil
> Error while running fiber: core/shm.lua:37: shm open error (7524/apps/fragmenterv4/out-ipv4-frag.counter):No such file or directory
>

takikawa, 2018-12-11:
Just a small update on this: I'm working on a set of patches that might fix this issue. Basically it would let the lwAFTR run without enabling VMDq in certain cases, which should make it possible to run on i210 NICs.