snabbco/snabb #1400: How to run selftests ?

mingodad, 2018-12-19:
Hello !
I've just finished the first release of snabb-ljs https://github.com/mingodad/snabb-ljs for testing purposes and I could not find if there is a way to execute all the selftests to check that my conversion is working as expected.

What it mean ?
Basically it's the snabb based on luajit converted to ljsjit a lua with C/C++/Java/Javascript syntax see https://github.com/mingodad/ljs . 

Also I do not have any equipment to test it, if someone could try it and report any problems I would appreciate the help.

Cheers !

dpino, 2018-12-19:
To run all selftests, simply type:

```bash
$ sudo make test
```

Tests that depend on hardware (NICs) will be skipped.

mingodad, 2018-12-21:
Hello again !
Now snabb-ljs passes all selftests that snabb pass on my machine.
Thanks !