snabbco/snabb #1412: FOSDEM 2019

lukego, 2019-01-30:
[FOSDEM 2019 is this weekend](https://fosdem.org/2019/)!

This issue is to help keep track of who is attending and where we can find each other. Please leave comments if you're coming and if you know where you will be! I'll update the summary.

Attending:
- [JAM Hotel](https://goo.gl/maps/s679C3gBJRp): @eugeneia @lukego @petebristow.
- Other: @wingo @dpino.

Talks:
- @lukego [How a ConnectX device driver works](https://fosdem.org/2019/schedule/event/how_connectx_device_driver_works/) Saturday 13:20 in Software Defined Networking devroom.

Friday evening:
- ???

Saturday evening:
- Networking/SDN crowd meeting at [Manneken Pis](https://goo.gl/maps/UcmUeDfuyg22) for dinner. Everybody welcome. Time? Guessing ~8pm.

cc @mwiget @justincormack @emmericp.

mwiget, 2019-01-30:
Something came up unfortunately, can't make it this time.

emmericp, 2019-01-30:
I'll be giving a talk at Saturday 13:00 with one of my students: https://fosdem.org/2019/schedule/event/writing_network_drivers_in_high_level_languages/

See [https://github.com/ixy-languages/ixy-languages](https://github.com/ixy-languages/ixy-languages), might be of interest to some of you?

justincormack, 2019-01-30:
I am arriving fairly late on Friday, and already commited for dinner on Sat but let me know where you will be and will see how to meet up...

dpino, 2019-01-31:
I'm arriving on Friday evening. I'm staying at Alma Hotel (near Grand Place), together with a few more igalians (although I think Andy is staying somewhere else). See you on Saturday at the SDN room!

lukego, 2019-01-31:
(What hardware interface will Ixy target next? i40e? fm10k? ConnectX? AVF? other? That's what I'd love to hear your thoughts about over a beer @emmericp.)

lukego, 2019-01-31:
Updated the text above based on the notion that at least @lukego @petebristow and @eugeneia will meet up at JAM Hotel in the evening e.g. for pizza and a few beers like last year. Everybody welcome!

emmericp, 2019-02-04:
Recordings are available:

Luke's talk: https://fosdem.org/2019/schedule/event/how_connectx_device_driver_works/
Our talk: https://fosdem.org/2019/schedule/event/writing_network_drivers_in_high_level_languages/