snabbco/snabb #1426: Error while running snabb ipfix probe command

nishantambastha, 2019-04-11:
Hello,

I'm trying to generate ipfix streams using 'snabb ipfix probe' 

`src/snabb ipfix probe -m 08:xx:27:xx:f2:08 -a 10.0.0.1 -c 10.0.0.2 -p 8080 --active-timeout 90 --transport udp --idle-timeout 90 -i /home/n/ipfix/bigFlows.pcap -o raw --cpu 1
`
However I'm getting below error :

> core/main.lua:26: unknown input type
> file not found: core/main.lua: No such file or directory
> file not found: program/ipfix/probe/probe.lua: No such file or directory
> file not found: core/main.lua: No such file or directory

I gave input file as .pcap, not sure why getting unknown input type error. If I have missed something in my command, pls let me know.

Cheers!

takikawa, 2019-04-11:
@nishantambastha Hi there, I think the issue here is just in the syntax of the command.

In particular, the `-i` flag takes the type of input to use to interpret the `<input>` position of the command. In the case of a pcap file it should be `pcap` rather than a file path.

For example, you can try a command like this instead:

```
./snabb ipfix probe -m 08:xx:27:xx:f2:08 -a 10.0.0.1 -c 10.0.0.2 -p 8080 --active-timeout 90 --transport udp --idle-timeout 90 -i pcap -o raw --cpu 1 /home/n/ipfix/bigFlows.pcap eth0
```

Replacing `eth0` with a suitable interface for your machine. (and you'll want to run this from the `snabb/src` directory)

takikawa, 2019-04-12:
BTW: looks like there's also a bug in the command, so you may need to apply the fix in #1427 if you don't also use the `-D` flag.