snabbco/snabb #1428: Race against auditlog being committed

eugeneia, 2019-04-22:
There is a race between when Snabb is started and when we open RaptorJIT’s audit log. When we exceed RaptorJIT’s temporary buffer before committing the audit log to file backed memory via `jit.auditlog(shm.path("audit.log"))`, Snabb will crash with a message such as: `core/app.lua:39: auditlog could not be opened (already open?)`

It is possible to mitigate this by increasing the size of the temporary buffer for the audit log (see below), but I do wonder what is the correct fix? Should we try to open the audit log earlier than the first call to `engine.main()`?

As an aside, the resulting error message could probably be more descriptive of the underlying issue. Cc @lukego 

```diff
modified   lib/luajit/src/lj_auditlog.c
@@ -13,7 +13,7 @@
 #include "lj_debuginfo.h"
 
 /* Maximum data to buffer in memory before file is opened. */
-#define MAX_MEM_BUFFER 10*1024*1024
+#define MAX_MEM_BUFFER 20*1024*1024
 /* State for initial in-memory stream. */
 static char *membuffer;
 static size_t membuffersize;

```

lukego, 2019-04-28:
Hm! Yes it seems like we should open the auditlog earlier. Do we need an `engine.init()` method or the like? Something that we call as soon as we have decided that the process will be running the engine and having a `shm` directory but e.g. before configuration loading. Meanwhile bumping the size seems like a reasonable workaround to me.

lukego, 2019-04-28:
Relatedly: I think the size of the auditlog can be shrunk quite a bit. We're logging raw C structs and some like `jit_State` are much bigger than they need to be. So a little refactoring in the RaptorJIT VM might be able to shrink things a bit. But yes really Snabb needs to be opening the log with only a bounded amount of data logged. Maybe also warning rather than erroring if the log can't be opened?

closed 2019-11-08 by eugeneia