snabbco/snabb #1440: Snabb lwAFTR not starting - Error message varies

Jaspils, 2019-08-13:
Not sure where to ask this, so correct me if I'm in the wrong place.

I'm quite new to this, so I hope my problem is pretty easy to tackle.

I'm experimenting with a lw4o6 demo and trying to get Snabb lwAFTR running on a basic Linux VM (Debian 10). I have one VMXNET3 interface connected to a management VLAN, and one Intel i350 NIC as passthrough to my network running lw4o6, meant to use as an on-a-stick setup. 

`0b:00.0 Ethernet controller: VMware VMXNET3 Ethernet Controller (rev 01)`
`13:00.0 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)`

Running lwAFTR I used the following command:
`snabb lwaftr run --conf lwaftr.conf --on-a-stick 13:00.0 --name lwaftr --cpu 3`
Resulting in the following error:
```
lwaftr.conf: loading compiled configuration from lwaftr.o
lwaftr.conf: compiled configuration is up to date.
WARNING: Buggy kernel '4.19.0-5-amd64'. Not binding CPU to NUMA node.
WARNING: Buggy kernel '4.19.0-5-amd64'. Not binding CPU to NUMA node.
Bound main process to NUMA node: 0 (CPU 0)
Binding data-plane PID 878 to CPU 3.
WARNING: Buggy kernel '4.19.0-5-amd64'. Not binding CPU to NUMA node.
hugetlb mmap failed (Cannot allocate memory), falling back.
core/main.lua:26: Invalid argument
line not found

Stack Traceback
===============
(1) Lua function 'handler' at file 'core/main.lua:172' (best guess)
        Local variables:
         (*temporary) = string: "core/main.lua:26: Invalid argument"
         (*temporary) = C function: print
         (*temporary) = number: 6.94067e-310
(2) global C function 'error'
(3) Lua global 'assert' at file 'core/main.lua:26'
        Local variables:
         (*temporary) = nil
(4) Lua field 'map_pci_memory_unlocked' at file 'lib/hardware/pci.lua:161'
        Local variables:
         (*temporary) = string: "13:00.0"
         (*temporary) = number: 0
         (*temporary) = boolean: false
         (*temporary) = string: "/sys/bus/pci/devices/0000:13:00.0/resource0"
         (*temporary) = nil
(5) Lua method 'new' at file 'apps/intel_mp/intel_mp.lua:396'
        Local variables:
         (*temporary) = table: 0x7fc43b83d070  {rss_tab:function: 0x7fc43b83f108, transmit:function: 0x7fc43b83ece0, rss_tab_build:function: 0x7fc43b83f138 (more...)}
         (*temporary) = table: 0x7fc43b843438  {rate_limit:0, vmdq:true, pciaddr:13:00.0, poolnum:0, ring_buffer_size:2048 (more...)}
         (*temporary) = table: 0x7fc410145730  {shm_root:/intel-mp/13:00.0/, pciaddress:13:00.0, rate_limit:0, vmdq:true (more...)}
         (*temporary) = string: "0x8086"
         (*temporary) = string: "0x1521"
         (*temporary) = table: 0x7fc43b83e0e0  {driver:table: 0x7fc43b83df08, registers:i350, max_q:8}
(6) Lua function 'ops' at file 'core/app.lua:378' (best guess)
        Local variables:
         (*temporary) = string: "nic"
         (*temporary) = table: 0x7fc43b83d070  {rss_tab:function: 0x7fc43b83f108, transmit:function: 0x7fc43b83ece0, rss_tab_build:function: 0x7fc43b83f138 (more...)}
         (*temporary) = table: 0x7fc43b843438  {rate_limit:0, vmdq:true, pciaddr:13:00.0, poolnum:0, ring_buffer_size:2048 (more...)}
(7) Lua field 'apply_config_actions' at file 'core/app.lua:405'
        Local variables:
          = table: 0x7fc43b7d7398  {1:table: 0x7fc43c7a2de0, 2:table: 0x7fc43b8be918, 3:table: 0x7fc43b88bf20 (more...)}
         (*temporary) = table: 0x7fc43b7d8080  {unlink_output:function: 0x7fc43b7d80f8, stop_app:function: 0x7fc43b7d82d8 (more...)}
         (*temporary) = Lua function 'remove' (defined at line 325 of chunk core/app.lua)
         (*temporary) = C function: builtin#6
         (*temporary) = table: 0x7fc43b7d7398  {1:table: 0x7fc43c7a2de0, 2:table: 0x7fc43b8be918, 3:table: 0x7fc43b88bf20 (more...)}
         (*temporary) = number: 10
         (*temporary) = number: 10
         (*temporary) = table: 0x7fc43b859280  {1:start_app, 2:table: 0x7fc43b843020}
         (*temporary) = string: "start_app"
         (*temporary) = table: 0x7fc43b843020  {1:nic, 2:table: 0x7fc43b83d070, 3:table: 0x7fc43b843438}
(8) Lua method 'commit_pending_actions' at file 'lib/ptree/worker.lua:77'
        Local variables:
         (*temporary) = table: 0x7fc43c7a2ce0  {no_report:false, channel:table: 0x7fc43c7a2e48, duration:inf, period:0.001 (more...)}
         (*temporary) = table: 0x7fc43b7d7398  {1:table: 0x7fc43c7a2de0, 2:table: 0x7fc43b8be918, 3:table: 0x7fc43b88bf20 (more...)}
         (*temporary) = boolean: true
(9) Lua method 'handle_actions_from_manager' at file 'lib/ptree/worker.lua:89'
        Local variables:
         (*temporary) = table: 0x7fc43c7a2ce0  {no_report:false, channel:table: 0x7fc43c7a2e48, duration:inf, period:0.001 (more...)}
         (*temporary) = table: 0x7fc43c7a2e48  {ring_buffer:cdata<struct 1519 *>: 0x7fc43c695000}
         (*temporary) = number: 1
         (*temporary) = number: 4
         (*temporary) = number: 1
         (*temporary) = number: 1
         (*temporary) = number: 4
         (*temporary) = table: 0x7fc43b7d7000  {1:commit, 2:table: 0x7fc43b7d7338}
(10) Lua field 'main' at file 'lib/ptree/worker.lua:112'
        Local variables:
         (*temporary) = table: 0x7fc43c7a2ce0  {no_report:false, channel:table: 0x7fc43c7a2e48, duration:inf, period:0.001 (more...)}
         (*temporary) = table: 0x7fc43cd60970  {close:function: builtin#151, select:function: builtin#152, stop:function: builtin#154 (more...)}
         (*temporary) = number: inf
         (*temporary) = number: 5915.1
(11) main chunk of [string "require('lib.scheduling').apply({["cpu"]=3,["..."] at line 2
(12) Lua function 'main' at file 'core/main.lua:59' (best guess)
        Local variables:
         (*temporary) = string: "require('lib.scheduling').apply({[\"cpu\"]=3,[\"busywait\"]=true,[\"ingress_drop_monitor\"]=\"flush\"})\
require('lib.ptree.worker').main()"
         (*temporary) = Lua function '?' (defined at line 0 of chunk "require('lib.scheduling').apply({["cpu"]=3,["..."])
(13) global C function 'xpcall'
(14) main chunk of file 'core/main.lua' at line 246
(15)  C function 'require'
(16) global C function 'pcall'
(17) main chunk of file 'core/startup.lua' at line 3
(18) global C function 'require'
(19) main chunk of [string "require "core.startup""] at line 1
        nil
Error while running fiber: core/shm.lua:37: shm open error (878/apps/ndp/next-hop-macaddr-v6.counter):No such file or directory
```

The error in the last line varies every time I run. Few examples:
- 786/apps/reassemblerv4/dtime.counter
- 751/apps/reassemblerv4/drop-ipv4-frag-random-evicted.counter
- 793/apps/fragmenterv4/out-ipv4-frag-not.counter
- 799/apps/reassemblerv4/dtime.counter
- 805/apps/reassemblerv4/drop-ipv4-frag-invalid-reassembly.counter

Any feedback is welcome

eugeneia, 2019-08-14:
Could you try with the `max-next` branch? There have recently been some fixes related to newer kernels and DMA memory mapping (https://github.com/snabbco/snabb/pull/1436). In any case it looks like your VM has trouble allocating or mapping hugetlb memory for Snabb.

I might also be off here and @dpino knows more.

Jaspils, 2019-08-14:
Thanks @eugeneia for your reply!

I have tried what you mentioned and tested it with the max-next branch. It gave me a different error:

```
lwaftr.conf: loading source configuration
lwaftr.conf: wrote compiled configuration lwaftr.o
lib/cltable.lua:38: attempt to index field 'keys' (a nil value)

Stack Traceback
===============
(1) Lua metamethod '__index' at file 'core/main.lua:172'
        Local variables:
         (*temporary) = string: "lib/cltable.lua:38: attempt to index field 'keys' (a nil value)"
         (*temporary) = C function: print
         (*temporary) = number: 6.92346e-310
(2) Lua field 'pairs' at file 'lib/cltable.lua:38'
        Local variables:
         cltable = table: 0x7f73177d06c0  {0:table: 0x7f73177a2858}
         (*temporary) = nil
         (*temporary) = Lua function '(anonymous)' (defined at line 24 of chunk core/main.lua)
         (*temporary) = nil
         (*temporary) = boolean: true
         (*temporary) = string: "attempt to index field 'keys' (a nil value)"
(3) Lua upvalue '' at file 'program/lwaftr/run/run.lua:42'
        Local variables:
         (*temporary) = table: 0x7f73177f21a0  {softwire_config:table: 0x7f73177f2488}
         (*temporary) = string: "13:00.0"
         (*temporary) = string: "13:00.0"
         (*temporary) = string: "13:00.0"
         (*temporary) = table: 0x7f73177a2a28  {queue:table: 0x7f73177d06c0}
(4) Lua field 'run' at file 'program/lwaftr/run/run.lua:148'
        Local variables:
         (*temporary) = table: 0x7f73185dac68  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
         (*temporary) = table: 0x7f731767f190  {on-a-stick:13:00.0, name:lwaftr, verbosity:0}
         (*temporary) = table: 0x7f731767f210  {ingress_drop_monitor:flush}
         (*temporary) = string: "lwaftr.conf"
         (*temporary) = string: "13:00.0"
         (*temporary) = string: "13:00.0"
         (*temporary) = table: 0x7f73177f21a0  {softwire_config:table: 0x7f73177f2488}
(5) Lua field 'run' at file 'program/lwaftr/lwaftr.lua:25'
        Local variables:
         (*temporary) = table: 0x7f73185dac68  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
         (*temporary) = string: "run"
         (*temporary) = string: "program.lwaftr.run.run"
(6) Lua function 'main' at file 'core/main.lua:70' (best guess)
        Local variables:
         (*temporary) = string: "lwaftr"
         (*temporary) = table: 0x7f73185dac68  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
(7) global C function 'xpcall'
(8) main chunk of file 'core/main.lua' at line 247
(9)  C function 'require'
(10) global C function 'pcall'
(11) main chunk of file 'core/startup.lua' at line 3
(12) global C function 'require'
(13) main chunk of [string "require "core.startup""] at line 1
        nil
```

Not sure what to do with this.

eugeneia, 2019-08-14:
Yeah, ok, you hit an unrelated regression introduced in `max-next`. This is useful information to me, not so much for you. ;-) Could you try https://github.com/snabbco/snabb/pull/1436 in isolation just to confirm whether or not what you are experiencing is https://github.com/snabbco/snabb/issues/1286?

Jaspils, 2019-08-14:
I edited the intel_mp.lua as you suggested
```
diff --git a/src/apps/intel_mp/intel_mp.lua b/src/apps/intel_mp/intel_mp.lua
index 273e6baf..78469ab7 100644
--- a/src/apps/intel_mp/intel_mp.lua
+++ b/src/apps/intel_mp/intel_mp.lua
@@ -435,6 +435,8 @@ function Intel:new (conf)
       -- Other processes wait for the shared lock before memory mapping the and
       -- loading registers.
       self.fd:flock("sh")
+       assert(shm.exists("group/dma/pci/"..self.pciaddress),
+               "NIC is already in use: "..self.pciaddress)
       self.base = pci.map_pci_memory(self.fd)
       self:load_registers(byid.registers)
    end
```
Resulting in the following error:
```
core/shm.lua:37: shm open error (12293/engine/breaths.counter):No such file or directory

Stack Traceback
===============
(1) Lua function 'handler' at file 'core/main.lua:172' (best guess)
        Local variables:
         (*temporary) = string: "core/shm.lua:37: shm open error (12293/engine/breaths.counter):No such file or directory"
         (*temporary) = C function: print
         (*temporary) = number: 6.93275e-310
(2) global C function 'error'
(3) Lua field 'create' at file 'core/shm.lua:37'
        Local variables:
         (*temporary) = string: "engine/breaths.counter"
         (*temporary) = boolean: false
         (*temporary) = boolean: true
         (*temporary) = string: "12293/engine/breaths.counter"
         (*temporary) = string: "read, write"
         (*temporary) = number: 8
         (*temporary) = nil
         (*temporary) = nil
(4) Lua field 'create' at file 'core/counter.lua:52'
        Local variables:
         (*temporary) = string: "engine/breaths.counter"
         (*temporary) = nil
         (*temporary) = number: 1
         (*temporary) = table: 0x7f9ee19c04b0  {}
(5) main chunk of file 'core/app.lua' at line 47
(6) global C function 'require'
(7) Lua global 'initialize' at file 'core/main.lua:162'
        Local variables:
         (*temporary) = _G (global table)
(8) Lua function 'main' at file 'core/main.lua:49' (best guess)
(9) global C function 'xpcall'
(10) main chunk of file 'core/main.lua' at line 247
(11)  C function 'require'
(12) global C function 'pcall'
(13) main chunk of file 'core/startup.lua' at line 3
(14) global C function 'require'
(15) main chunk of [string "require "core.startup""] at line 1
        nil
```

Jaspils, 2019-08-15:
> Thanks @eugeneia for your reply!
>
> I have tried what you mentioned and tested it with the max-next branch. It gave me a different error:
>
> ```
> lwaftr.conf: loading source configuration
> lwaftr.conf: [...]

Update: New day, fresh boot, tried again with the edited intel_mp.lua as mentioned. But now I get back to the same error as before. So my latest message might not be accurate.

eugeneia, 2019-08-15:
https://github.com/snabbco/snabb/issues/1440#issuecomment-521289570
This doesn’t look applied right. You can check out the PR head like so:

```
$ git fetch origin pull/1436/head:pci-unbind-before-map2
$ git checkout pci-unbind-before-map2
```

Jaspils, 2019-08-15:
I'm not sure what I did wrong in this [#1440 (comment)](https://github.com/snabbco/snabb/issues/1440#issuecomment-521289570) but as mentioned in this other [#1440 (comment)](https://github.com/snabbco/snabb/issues/1440#issuecomment-521547440) I'm now back at this:
```
sudo ./snabb lwaftr run --conf lwaftr.conf --on-a-stick 0b:00.0 --name lwaftr --cpu 1
lwaftr.conf: loading source configuration
lwaftr.conf: wrote compiled configuration lwaftr.o
lib/cltable.lua:38: attempt to index field 'keys' (a nil value)

Stack Traceback
===============
(1) Lua metamethod '__index' at file 'core/main.lua:172'
        Local variables:
         (*temporary) = string: "lib/cltable.lua:38: attempt to index field 'keys' (a nil value)"
         (*temporary) = C function: print
         (*temporary) = number: 6.93198e-310
(2) Lua field 'pairs' at file 'lib/cltable.lua:38'
        Local variables:
         cltable = table: 0x7f9b47b6d060  {0:table: 0x7f9b47b73820}
         (*temporary) = nil
         (*temporary) = Lua function '(anonymous)' (defined at line 24 of chunk core/main.lua)
         (*temporary) = nil
         (*temporary) = boolean: true
         (*temporary) = string: "attempt to index field 'keys' (a nil value)"
(3) Lua upvalue '' at file 'program/lwaftr/run/run.lua:42'
        Local variables:
         (*temporary) = table: 0x7f9b47bc28e0  {softwire_config:table: 0x7f9b47bc2bc8}
         (*temporary) = string: "0b:00.0"
         (*temporary) = string: "0b:00.0"
         (*temporary) = string: "0b:00.0"
         (*temporary) = table: 0x7f9b47b739f0  {queue:table: 0x7f9b47b6d060}
(4) Lua field 'run' at file 'program/lwaftr/run/run.lua:148'
        Local variables:
         (*temporary) = table: 0x7f9b489a3a50  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:0b:00.0, 5:--name, 6:lwaftr (more...)}
         (*temporary) = table: 0x7f9b47a47960  {on-a-stick:0b:00.0, name:lwaftr, verbosity:0}
         (*temporary) = table: 0x7f9b47a479e0  {ingress_drop_monitor:flush}
         (*temporary) = string: "lwaftr.conf"
         (*temporary) = string: "0b:00.0"
         (*temporary) = string: "0b:00.0"
         (*temporary) = table: 0x7f9b47bc28e0  {softwire_config:table: 0x7f9b47bc2bc8}
(5) Lua field 'run' at file 'program/lwaftr/lwaftr.lua:25'
        Local variables:
         (*temporary) = table: 0x7f9b489a3a50  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:0b:00.0, 5:--name, 6:lwaftr (more...)}
         (*temporary) = string: "run"
         (*temporary) = string: "program.lwaftr.run.run"
(6) Lua function 'main' at file 'core/main.lua:70' (best guess)
        Local variables:
         (*temporary) = string: "lwaftr"
         (*temporary) = table: 0x7f9b489a3a50  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:0b:00.0, 5:--name, 6:lwaftr (more...)}
(7) global C function 'xpcall'
(8) main chunk of file 'core/main.lua' at line 247
(9)  C function 'require'
(10) global C function 'pcall'
(11) main chunk of file 'core/startup.lua' at line 3
(12) global C function 'require'
(13) main chunk of [string "require "core.startup""] at line 1
        nil
```

I also tested this :

> [#1440 (comment)](https://github.com/snabbco/snabb/issues/1440#issuecomment-521289570)
> This doesn’t look applied right. You can check out the PR head like so:
> 
> ```
> $ git fetch origin pull/1436/head:pci-unbind-before-map2
> $ git checkout pci-unbind-before-map2
> ```

But came out with the same error.

Jaspils, 2019-08-15:
Just went ahead to test it on CentOS for a change. Maybe this is usefull, maybe not.

CentOS, same config, master-branch:
```
./snabb lwaftr run --conf lwaftr.conf --on-a-stick 13:00.0 --name lwaftr --cpu 1                     
lwaftr.conf: loading source configuration
lwaftr.conf: wrote compiled configuration lwaftr.o
syscall/linux/syscalls.lua:513: incompatible nodemask sizes

Stack Traceback
===============
(1) Lua function 'handler' at file 'core/main.lua:172' (best guess)
        Local variables:
         (*temporary) = string: "syscall/linux/syscalls.lua:513: incompatible nodemask sizes"
         (*temporary) = C function: print
         (*temporary) = number: 6.91636e-310
(2) upvalue C function 'assert'
(3) Lua field 'migrate_pages' at file 'syscall/linux/syscalls.lua:513'
        Local variables:
         (*temporary) = number: 0
(4) Lua field 'bind_to_numa_node' at file 'lib/numa.lua:194'
        Local variables:
         = number: 0
         (*temporary) = nil
         (*temporary) = string: "3.10.0-957.27.2.el7.x86_64"
(5) Lua method 'bind_to_numa_node' at file 'lib/cpuset.lua:87'
        Local variables:
         (*temporary) = table: 0x7f51abbfb528  {by_node:table: 0x7f51abbfb570}
         (*temporary) = table: 0x7f51aa267758  {1:0}
(6) Lua method 'start' at file 'lib/ptree/ptree.lua:174'
        Local variables:
         (*temporary) = table: 0x7f51aaf39ad0  {setup_fn:function: 0x7f51aaf391f0, peers:table: 0x7f51aaf44d80, current_in_place_dependencies:table: 0x7f51aa319b58 (more...)}
(7) Lua field 'ptree_manager' at file 'lib/ptree/ptree.lua:112'
        Local variables:
         b = table: 0x7f51aaf392b0  {setup_fn:function: 0x7f51aaf391f0, worker_default_scheduling:table: 0x7f51abe01b80 (more...)}
         (*temporary) = table: 0x7f51aaf392f8  {setup_fn:function: 0x7f51aaf391f0, worker_default_scheduling:table: 0x7f51abe01b80 (more...)}
         (*temporary) = table: 0x7f51aaf39ad0  {setup_fn:function: 0x7f51aaf391f0, peers:table: 0x7f51aaf44d80, current_in_place_dependencies:table: 0x7f51aa319b58 (more...)}
(8) Lua field 'run' at file 'program/lwaftr/run/run.lua:187'
        Local variables:
         (*temporary) = table: 0x7f51abd05ca0  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
         (*temporary) = table: 0x7f51abe01b00  {on-a-stick:13:00.0, name:lwaftr, verbosity:0}
         (*temporary) = table: 0x7f51abe01b80  {ingress_drop_monitor:flush}
         (*temporary) = string: "lwaftr.conf"
         (*temporary) = string: "13:00.0"
         (*temporary) = string: "13:00.0"
         (*temporary) = table: 0x7f51aaf67c60  {softwire_config:table: 0x7f51aaf67db8}
         (*temporary) = Lua function 'setup' (defined at line 154 of chunk program/lwaftr/run/run.lua)
         (*temporary) = table: 0x7f51aaf39138  {worker_default_scheduling:table: 0x7f51abe01b80}
(9) Lua field 'run' at file 'program/lwaftr/lwaftr.lua:25'
        Local variables:
         (*temporary) = table: 0x7f51abd05ca0  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
         (*temporary) = string: "run"
         (*temporary) = string: "program.lwaftr.run.run"
(10) Lua function 'main' at file 'core/main.lua:70' (best guess)
        Local variables:
         (*temporary) = string: "lwaftr"
         (*temporary) = table: 0x7f51abd05ca0  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
(11) global C function 'xpcall'
(12) main chunk of file 'core/main.lua' at line 246
(13)  C function 'require'
(14) global C function 'pcall'
(15) main chunk of file 'core/startup.lua' at line 3
(16) global C function 'require'
(17) main chunk of [string "require "core.startup""] at line 1
        nil
```
CentOS, same config, max-next branch:
```
./snabb lwaftr run --conf lwaftr.conf --on-a-stick 13:00.0 --name lwaftr --cpu 1                     
lwaftr.conf: loading source configuration
[... 27 lines cut ...]
         (*temporary) = table: 0x7fb307668568  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
         (*temporary) = table: 0x7fb30670c9d0  {on-a-stick:13:00.0, name:lwaftr, verbosity:0}
         (*temporary) = table: 0x7fb30670ca50  {ingress_drop_monitor:flush}
         (*temporary) = string: "lwaftr.conf"
         (*temporary) = string: "13:00.0"
         (*temporary) = string: "13:00.0"
         (*temporary) = table: 0x7fb3068afec8  {softwire_config:table: 0x7fb3068ae688}
(5) Lua field 'run' at file 'program/lwaftr/lwaftr.lua:25'
        Local variables:
         (*temporary) = table: 0x7fb307668568  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
         (*temporary) = string: "run"
         (*temporary) = string: "program.lwaftr.run.run"
(6) Lua function 'main' at file 'core/main.lua:70' (best guess)
        Local variables:
         (*temporary) = string: "lwaftr"
         (*temporary) = table: 0x7fb307668568  {1:--conf, 2:lwaftr.conf, 3:--on-a-stick, 4:13:00.0, 5:--name, 6:lwaftr (more...)}
(7) global C function 'xpcall'
(8) main chunk of file 'core/main.lua' at line 247
(9)  C function 'require'
(10) global C function 'pcall'
(11) main chunk of file 'core/startup.lua' at line 3
(12) global C function 'require'
(13) main chunk of [string "require "core.startup""] at line 1
        nil
```

eugeneia, 2019-08-19:
@Jaspils I just merged a fix for the regression in https://github.com/snabbco/snabb/issues/1440#issuecomment-521217762 to `max-next`.

Running `snabb lwaftr run --conf lwaftr-test.conf --on-a-stick 01:00.0 --name lwaftr --cpu 1` with a configuration like below seemed to work for me.

```
softwire-config {
  instance {
    device "01:00.0";
    queue {
      id 0;
      external-interface {
        ip 10.10.10.10;
        mac 12:12:12:12:12:12;
        // vlan-tag 42;
        next-hop {
          mac 68:68:68:68:68:68;
        }
      }
      internal-interface {
        ip 8:9:a:b:c:d:e:f;
        mac 22:22:22:22:22:22;
        // vlan-tag 64;
        next-hop {
          mac 44:44:44:44:44:44;
        }
      }
    }
  }

  // IPv4 interface.
  external-interface {
    allow-incoming-icmp true;
    // Limit ICMP error transmit rate to PACKETS/PERIOD.
    error-rate-limiting {
      packets 600000;
      period 2;
    }
    // Generate ICMP errors at all?
    generate-icmp-errors true;
    // Basic parameters.
    mtu 1460;
    // Where to go next.  Either one will suffice; if you specify the IP,
    // the next-hop MAC will be determined by ARP.
    // Control the size of the fragment reassembly buffer.
    reassembly {
      max-fragments-per-packet 40;
      max-packets 20000;
    }
  }
  // The same thing as for the external interface, but on the IPv6 side
  // and with IPv6 addresses.
  internal-interface {
    allow-incoming-icmp true;
    error-rate-limiting {
      packets 600000;
      period 2;
    }
    generate-icmp-errors true;
    // One more interesting thing -- here we control whether to support
    // routing traffic between two B4s.
    hairpinning true;
    mtu 1500;
    reassembly {
      max-fragments-per-packet 40;
      max-packets 20000;
    }
  }
  // Now the binding table!
  binding-table {
    softwire {
      ipv4 178.79.150.233;
      psid 22788;
      b4-ipv6 127:11:12:13:14:15:16:128;
      br-address 8:9:a:b:c:d:e:f;
      port-set {
        psid-length 16;
      }
    }
    softwire {
      ipv4 178.79.150.233;
      psid 2700;
      b4-ipv6 127:11:12:13:14:15:16:128;
      br-address 8:9:a:b:c:d:e:f;
      port-set {
        psid-length 16;
      }
    }
    softwire {
      ipv4 178.79.150.15;
      psid 1;
      b4-ipv6 127:22:33:44:55:66:77:128;
      br-address 8:9:a:b:c:d:e:f;
      port-set {
        psid-length 4;
      }
    }
    softwire {
      ipv4 178.79.150.2;
      psid 7850;
      b4-ipv6 127:24:35:46:57:68:79:128;
      br-address 1e:1:1:1:1:1:1:af;
      port-set {
        psid-length 16;
      }
    }
  }
}
```

Jaspils, 2019-08-19:
@eugeneia you're a hero. Tried it with your most recent max-next and it seems to be starting now! I'd still have to experiment with some B4 and see how far I can get. Thanks for your help.

For your interest, still got some warnings but I think my problem can be marked as 'solved' :

```
sudo ./snabb lwaftr run --conf lwaftr.conf --on-a-stick 0b:00.0 --name lwaftr -
-cpu 1 -v                                                                                                                         
lwaftr.conf: loading compiled configuration from lwaftr.o                                                                         
lwaftr.conf: compiled configuration is up to date.                                                                                
Migrating instance '13:00.0' to '0b:00.0'                                                                                         
WARNING: Buggy kernel '4.15.0-58-generic'. Not binding CPU to NUMA node.                                                          
WARNING: Buggy kernel '4.15.0-58-generic'. Not binding CPU to NUMA node.                                                          
Bound main process to NUMA node: 0 (CPU 0)                                                                                        
Binding data-plane PID 5089 to CPU 1.                                                                                             
WARNING: Buggy kernel '4.15.0-58-generic'. Not binding CPU to NUMA node.                                                          
hugetlb mmap failed (Cannot allocate memory), falling back.                                                                       
Time (s),v4 MPPS,v4 Gbps,v4 MPPS,v4 Gbps,v6 MPPS,v6 Gbps,v6 MPPS,v6 Gbps                                                          
1.002390,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000                                                  
2.000394,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000                                                  
2.998575,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000
```

closed 2019-08-19 by Jaspils

eugeneia, 2019-08-19:
FWIW I think the fix for your problem was really #1436, which happens to be included in `max-next`.