snabbco/snabb #1447: Zero-copy in most NICs

gorgonical, 2019-10-30:
I couldn't find anything in the docs or the code that confirmed this question, so I figured I would ask. I would have asked in the Slack but the invite link is broken.

Does Snabb provide zero-copy semantics for NICs? I know that the memory module provides DMA memory for the NIC to access and the usage in some parts of the code such as:

```src/apps/intel_mp/intel_mp.lua:555:   memory.dma_alloc(ffi.sizeof(rxdesc_ring_t)))
src/apps/intel_mp/intel_mp.lua:625:   memory.dma_alloc(ffi.sizeof(txdesc_ring_t)))```

indicate that the DMA memory is being used for the NIC queues, but I wanted to be sure.

eugeneia, 2019-11-03:
Yes, that is what we do. I.e., let the hardware write to and read from our `struct packet` buffers in DMA memory directly.

https://github.com/lukego/blog/issues/13 might give slightly more detail from the OS perspective.

eugeneia, 2019-11-12:
The invite link should work again btw. :-)