snabbco/snabb #1458: Why will TLS shootdowns cause packet loss?

NanXiao, 2020-04-24:
Hi,

Greetings from me!

When reading [Watch for TLB shootdowns](https://github.com/snabbco/snabb/blob/master/src/doc/performance-tuning.md#watch-for-tlb-shootdowns), I came across following statements:

> Occasional packet loss at ingress has been observed while TLB shootdowns happened on the system.

Could some guys help to explain why TLS shootdowns cause packet loss? Thanks very much in advance!

Best Regards
Nan Xiao

mrichtarsky, 2020-06-11:
The correct term is TLB for Translation Lookaside Buffer.

TLB shootdowns are usually handled via interrupts and can be time-consuming (relatively to other low-level operations). Therefore the performance of the system is impacted and there can be packet loss since not all packets can be processed in time. 

There are some interesting details in this paper:
https://yoav.net.technion.ac.il/files/2016/05/DiDi-PACT-2011.pdf

NanXiao, 2020-06-11:
@mrichtarsky Thanks very much for your explanation!

closed 2020-06-11 by NanXiao