snabbco/snabb #1528: Snabb not working in a stateless way

norbertn503, 2025-02-13:
So I have the following setup using VMware virtual machines running Debian Linux. 

[image]

And this is working. I can successfully ping the Server from the Client. Since the lwAFTR works in a stateless way. I should be able to reach the lwB4 IPv4 address from the Server, however when I try to ping  it the packets arrive at lwAFTR, but they are not encapsulated and sent to lwB4.

I saw in the tutorial that it is possible that lwAFTR forwards packets originating from the outside interface  [https://github.com/snabbco/snabb/tree/master/src/program/lwaftr/doc/tutorial](url)

Do I need to add something to my lwAFTR config to force this behavior, or should I config something on the Linux that runs snabb?

On the lwAFTR  I did the following configurations:
```
echo 1 > /proc/sys/net/ipv4/ip_forward
echo 1 > /proc/sys/net/ipv6/conf/all/forwarding

#B4 facing interface
ip link set eth1 up

#Server facing interface
ip link set eth2 up
```

And I have this config for lwAFTR:

```
 softwire-config {
  instance {
    device "eth1";
    external-device "eth2";
    queue {
      id 0;
      // IPv6 interface
      internal-interface {
        ip "2001:db8:0:1::1";     // Snabb's internal IP
        mac "00:0c:29:35:be:62";  // lwAFTR eth1 MAC
        next-hop {
          ip "2001:db8:0:1::2";   // lwB4 eth2 IPv6
          // or: mac "";
        }
      }

      // IPv4 interface
      external-interface {
        ip "192.0.2.1";          // Snabb's external IP
        mac "00:0c:29:35:be:6c"; // lwAFTR eth2 MAC
        next-hop {
          ip "192.0.2.2";        // IPv4 Server IP
        }
      }

    }
  }

  // IPv6 interface
  internal-interface {
    mtu 1540; // MTU of the IPv6 interface (note: IPv6 encapsulation overhead!)
    flow-label 42; // Flow label used for IPv6 encapsulation
    allow-incoming-icmp true;
    generate-icmp-errors true;
  }

  // IPv4 interface.
  external-interface {
    mtu 1500; // MTU of the IPv4 interface
    generate-icmp-errors true;
    allow-incoming-icmp true;
  }

  binding-table {
    softwire {
      ipv4 203.0.113.1;        // lwB4 public and external IP
      psid 1;
      b4-ipv6  2001:db8:0:1::2;
      br-address 2001:db8:2::2;  // to me, I see it as the tunnel end point, its                                                                                                                                                            an IP on lwAFTR side
      port-set {
        psid-length 6;
      }
    }
 }
}
```

eugeneia, 2025-02-17:
Do the snabb/lwaftr statistics counters yield some clues as to where the packets get dropped? IIRC we should count ICMP separately and that's where I would look first to see what's going on. I think `./snabb config get-state my-lwaftr-instance /` was the query to get those.

norbertn503, 2025-02-18:
@eugeneia thank you, for your tip. 

It counted the droped packets as `drop-in-by-rfc7596-icmpv4-packets` . 

According to RFC 7596 8.1 section
```
Check the ICMP Type field.

   2.  If the ICMP Type field is set to 0 or 8 (echo reply or request),
       then the lwAFTR MUST take the value of the ICMP Identifier field
       as the source port and use this value to look up the binding
       table for an encapsulation destination.  If a match is found, the
       lwAFTR forwards the ICMP packet to the IPv6 address stored in the
       entry; otherwise, it MUST discard the packet.

```

So, when I simply used the ping command, the ICMP identifier filed was not set to any source port.

Then I used `hping3 203.0.113.1 -p 1500` which sends a TCP packet to port 1500 and snabb was able to forward it to the lwB4 IPv4 address.

Even though it's not ICMP packet that was forwarded, it is still good enough for me. I can see that snabb works in a stateless way.

eugeneia, 2025-02-19:
Awesome!

closed 2025-02-19 by eugeneia